eclipse-mraa/mraa #350: "canoninal mode" for uart (on Intel Edison)

barlesh, 2015-11-03:
I try to send & receive data to/from a modem using mraa's uart lib.
Sending raw data not including 0x0A (10 Dec, ascii's "\n") works perfectly, but when adding 0x0A everything is massed up. 
While trying to receive a packet from my modem, if a 0x10 isn't sent at the end of the buffer, mraa_uart_data_available() doesn't return (just like mraa_uart_read() ).
sounds like canonical mode at termios lib for serial connections.
anyone has any idea if that is the problem and if so how can I set non-canonical (raw) mode?
tnx 
bar

alext-mkrs, 2015-11-03:
As you can see in [uart.c](https://github.com/intel-iot-devkit/mraa/blob/master/src/uart/uart.c#L226-L229) we actually do set it into raw mode.

Can you provide an example of the code that doesn't work for you, elaborate on "everything is messed up" and clarify what modem are you talking about?

Sanderhuisman, 2015-11-09:
I have the same problem with my edison: when I don't include '\n' in the transmitted string, dataAvailable() will not return. I am using the code as follows:

```c++
mraa::Uart *dev;
	try {
		dev = new mraa::Uart("/dev/ttyMFD1");
	} catch (std::exception& e) {
		std::cout << "Error while setting up raw UART" << std::endl;
		std::terminate();
	}

	if (dev->setBaudRate(9600) != MRAA_SUCCESS) {
		std::cout << "Error setting parity on UART" << std::endl;
	}
	/*
	 if (dev->setMode(8, MRAA_UART_PARITY_NONE, 1) != MRAA_SUCCESS) {
	 std::cout << "Error setting parity on UART" << std::endl;
	 }
	 if (dev->setFlowcontrol(false, false) != MRAA_SUCCESS) {
	 std::cout << "Error setting flow control UART" << std::endl;
	 }
	 */

	const uint8_t MAX_BUFFERLENGTH = 255;

	while (1) {
		if (dev->dataAvailable(1000)) {			// wait
			char receivedBuffer[MAX_BUFFERLENGTH + 1];

			uint8_t receivedBytes = dev->read(receivedBuffer, MAX_BUFFERLENGTH);
			receivedBuffer[receivedBytes] = NULL;

			printf("received %d %s\n", receivedBytes, receivedBuffer);
		}
	}
```

barlesh, 2015-11-09:
here is my code..... 
actually I used termios to solve my problem (other version of this program), but maybe you can find something new....

#include "port5_trans.h"

void init(){
		f = 0; 	
		init_modem();
		sleep(1);
		send_garbage(PACKET_SIZE);
}

void main(){
	byte buffer[PACKET_SIZE] = { 0 };	
	byte i=0,j=0, m_len=0, flag=0, len=0, X=0;
	byte payload_size = PACKET_SIZE-2, header_size =2; 
	printf("starting.....\n");
	init();
	//init_file(src);
	show_buffer(buffer, sizeof(buffer));
	len = create_debug_packet(buffer, 2);
	
	while(!f){
		printf("sending massage of length %d, msg is:", len);
		show_buffer(buffer, len);
		printf("\n");
		
		sendToModem((char*)buffer, len);
		
		i++;
		if(i==10) i=0;
		buffer[2] = i;
		sleep(4);
	}//while(f)
	
	deinit_modem();

}//main

void show_buffer(byte *buffer, short len){
	int i=0, n=0;
	//printf("len is:%d\n", len);
	for(i;i<len; i++){
		n = buffer[i];
		n = n & 0xFF;
		printf("(%d)%d_",i, n);
		}
	printf("\n");

}

int create_debug_packet(byte * buffer, byte type){
	int i = 2;
	unsigned short c = 1;
	buffer[0] = type;
	buffer[1] = 0 ;
	for(i=2; i<PACKET_SIZE; i++)
		{buffer[i] = c; c++; if(c==10) c++;}
	return i;
}

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
////////////////////////////////////////	Modem related functions		//////////////////////////////////////
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
int init_modem(){
 	//mraa_init();
	 if( (uart = mraa_uart_init(0)) == NULL   ){
	   		//writeToFile(src, "error INITIATING UART\n");
  			exit(UART_ERROR);
	 }
	 if ( mraa_uart_set_baudrate(uart, UART_BOUD_RATE)!=MRAA_SUCCESS) {
  		//writeToFile(src, "error seting baudrate\n");
  		exit(UART_ERROR);
  		}		//set uart boud rate [bps]
  if ( mraa_uart_set_mode(uart, 8,MRAA_UART_PARITY_NONE , 1)!=MRAA_SUCCESS) {
  //writeToFile(src, "error seting mode\n");
	}
	mraa_uart_set_flowcontrol(uart, 0, 0);
	
	//writeToFile(src, "init modem\n");
	return 1;	
}

int is_incoming_modem(){
	int x=0;
	if ( (x = mraa_uart_data_available(uart,0) ) < 0   ) {
	  		//writeToFile(src, "error reading data from uart\n");
  			exit(UART_ERROR);
	}	
	return x;
}

int get_modem_frame(char* uart_buffer){
	return mraa_uart_read(uart, uart_buffer, PACKET_SIZE);
}

void sendToModem(char* buffer, int len){
	int ans=0, i;
	//ans = mraa_uart_write( uart , buffer, len);
	for(i=0;i<len;i++){
		mraa_uart_write( uart , &(buffer[i]), 1);
	}
	printf("try to write to uart %d [byte].\twrote %d [byte]\n", len, ans);
	//mraa_uart_write( uart, buff, 1);
}

void deinit_modem(){
	printf("destruct uart\n");
	mraa_uart_stop(uart);
}

void send_garbage(int len){
	byte buffer[len];
	byte i=0;
	for(i;i<len; i++) 
		buffer[i] = i;
	printf("sending %d bytes of garbage\n", len);
	sendToModem((char*)buffer,len);

}

alext-mkrs, 2015-11-09:
Thank you both, I'll analyze that and get back to you.

alext-mkrs, 2015-11-14:
@Sanderhuisman looks like I can't reproduce this. I've tried your program in a setup where I have Galileo with a dummy sketch sending a single string "test" (without any additional line endings) at one end and Edison with your program at another end. Works fine for me, receiving 4 bytes:

```
root@edison1:~# ./mraa-350-repro 
received 4 test
received 4 test
received 4 test
^C
root@edison1:~# 
```

And I've double-checked the Edison end is not in the canonical mode when the program is running:

```
root@edison1:~# stty -F /dev/ttyMFD1 -a
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V;
flush = ^O; min = 1; time = 0;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo echoe echok -echonl -noflsh -xcase -tostop -echoprt echoctl echoke
```

@barlesh, I can't run your code directly, so if you could diminish that to the shortest possible reproducer, that would help.

Sanderhuisman, 2015-11-15:
I ran the same command on my Edison and get this respond:
```
root@edison:~# stty -F /dev/ttyMFD1 -a
speed 9600 baud;stty: /dev/ttyMFD1
 line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q;
stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V; flush = ^O; min = 1; time = 0;
-parenb -parodd cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr icrnl ixon -ixoff -iuclc -ixany -imaxbel -iutf8
opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
isig icanon iexten echo echoe echok -echonl -noflsh -xcase -tostop -echoprt echoctl echoke
root@edison:~#
```
My Edison currently runs with Yocto version 159, libmraa0 version 0.7.3 and UPM 0.3.2

I tested with a USB-serial adapter which is hooked up with my laptop. I tried with Putty as well as a program I made in Visual Basic and I keep having the problem that when I don't include `\n` the data will not be received.

The program I made can be found below:

```vb.net
Imports System.IO.Ports

Class edisonControl
    Dim SerialPort As Array
    Dim PortStatus As Boolean = False
    Dim PortInUse As Boolean = False

    Private Sub edisonControl_Load(sender As Object, e As EventArgs) Handles MyBase.Load
        SerialPort = IO.Ports.SerialPort.GetPortNames()
        PortComboBox.Items.AddRange(SerialPort)
    End Sub

    Private Sub QuadRemote_Closing(sender As Object, e As System.ComponentModel.CancelEventArgs) Handles MyBase.Closing
        While (PortInUse = True)
        End While

        SerialPort1.Close()
    End Sub

    Private Sub ConnectButton_Click(sender As Object, e As EventArgs) Handles ConnectButton.Click
        If PortComboBox.Text = Nothing Or BaudComboBox.Text = Nothing Then
            If PortComboBox.Text = Nothing Then
                Call MsgBox("Please select a device", 0, "A Message")
            Else
                Call MsgBox("Please select a baudrate", 0, "A Message")
            End If
        Else
            If PortStatus = False Then
                SerialPort1.PortName = PortComboBox.Text
                SerialPort1.BaudRate = BaudComboBox.Text

                SerialPort1.DataBits = 8
                SerialPort1.Parity = Parity.None
                SerialPort1.StopBits = StopBits.One
                SerialPort1.Handshake = Handshake.None
                SerialPort1.Encoding = System.Text.Encoding.Default

                PortStatus = True
                ConnectButton.Text = "Disconnect"

                SerialPort1.Open()
            Else
                While (PortInUse = True)

                End While

                SerialPort1.Close()

                PortStatus = False
                ConnectButton.Text = "Connect"
            End If
        End If
    End Sub

    Private Sub PortComboBox_Click(sender As Object, e As EventArgs) Handles PortComboBox.Click
        PortComboBox.Items.Clear()
        SerialPort = IO.Ports.SerialPort.GetPortNames()
        PortComboBox.Items.AddRange(SerialPort)
    End Sub

    Private Sub Button1_Click(sender As Object, e As EventArgs) Handles Button1.Click
        If SerialPort1.IsOpen Then

            Dim dataArray(5) As Byte
            'Dim dataArray(4) As Byte
            dataArray(0) = Convert.ToSByte("#"c)
            dataArray(1) = Convert.ToSByte("t"c)
            dataArray(2) = Convert.ToSByte("e"c)
            dataArray(3) = Convert.ToSByte("s"c)
            dataArray(4) = Convert.ToSByte("t"c)
            dataArray(5) = 10
            SerialPort1.Write(dataArray, 0, dataArray.Length)
        End If
    End Sub
End Class
```

alext-mkrs, 2015-11-15:
Ok, I see what may be the reason. You have a bit old version of the library - 0.7.3.

I have fixed an UART init bug some time ago, but after that version. So just upgrade using opkg by running "opkg update && opkg install mraa" as root on the board and you will get version 0.8.1, which should work fine.

Please give it a try and let me know how it goes after the update.

Sanderhuisman, 2015-11-15:
Thank you so much! I had to update opkg first and then I was able to update MRAA. After the update, the UART works as it should! Thank you;)

alext-mkrs, 2015-11-15:
Glad to read that and you're most welcome :)

@barlesh, what about you?

alext-mkrs, 2015-11-22:
Closing for no activity and solution provided. If you need any further help - fee free to add a comment.

closed 2015-11-22 by alext-mkrs