eclipse-mraa/mraa #353: “Invalid GPIO pin specified” when sudo in Edison Debian

geotheory, 2015-11-04:
Efforts to connect to GPIO pin fails when using both Python and C methods detailed in the [Sparkfun tutorial](https://learn.sparkfun.com/tutorials/installing-libmraa-on-ubilinux-for-edison) for Ubilinux.  Our setup is however a jailed 7.9 Wheezy running on top of Yocto via `chroot`, which may be part of the problem.  Python returns error:

```sudo python blink.py
Traceback (most recent call last):
  File "blink.py", line 5, in <module>
    x = mraa.Gpio(31)
  File "/usr/local/lib/python2.7/site-packages/mraa.py", line 466, in __init__
    this = _mraa.new_Gpio(*args)
ValueError: Invalid GPIO pin specified```

Any idea how to identify the problem here?  Help very much appreciated!

tingleby, 2015-11-04:
IRC, stackoverflow and now here :)
Could you check the output of `ls /sys/class/gpio/` inside the chroot

geotheory, 2015-11-04:
Yes!  It's an important project..  That gives me:

    ls /sys/class/gpio/
    export   gpio125  gpio127  gpio129  gpio130  gpio132  gpio134  gpiochip0
    gpio124  gpio126  gpio128  gpio13   gpio131  gpio133  gpio44   unexport

arfoll, 2015-11-04:
What baseboard are you using and make sure you're giving it enough power.
If on arduino baseboard use external psu not just USB. Also what Edison
image version are you using and which mraa version.

Could you try from yocto directly first? Some device node may not have been
exported correctly in your chroot.

alext-mkrs, 2015-11-04:
I'd also recommed you to run the script under strace, the information captured might be quite useful for debugging.

```
sudo strace -ff -s 256 python blink.py
```

geotheory, 2015-11-04:
@arfoll it's running on a Sparkfun Base Block together with MicroSD, GPIO and battery blocks.  Battery block is running plus 2 USB cables for serial connection with screen, so I'd imagine power is ok?

The reason for using `chroot` is partly i think that Yocto doesn't support apt-get.  I'm not sure how feasible this installation is without it..

geotheory, 2015-11-04:
@alext-mkrs thanks for the suggestion.  Here's the [strace report](https://gist.github.com/geotheory/2471a28147c9eaf39a71).

arfoll, 2015-11-04:
What's the mraa version? Also the default YP image for edison uses opkg instead of apt-get but what are your reasons for using apt-get? Which packages can you not get installed on YP? Tbh that strace report just confirms gpio31 can't get open leading to an error. Can you get this running without the chroot on the latest official edison image?

geotheory, 2015-11-05:
I cloned `https://github.com/intel-iot-devkit/mraa` so must be v0.8.1.  Not sure why opkg wasn't implemented - the project started life some years ago so perhaps it predates it, as it did Ubilinux back then - hence the Yocto/Debian chroot arrangement.  Migration to full Ubilinux is an aspiration, but this is the priority.  I haven't even tried to replicate this mraa installation on Yocto - partly due to [horror stories](https://communities.intel.com/message/221564#221564) of 100s of lost hours..

Do you know if opkg is likely to support mraa's dependancies - libpcre3-dev/make/cmake/python-dev/swig - plus _their_ dependancies?

arfoll, 2015-11-05:
I think you're mixing a few things up. I would never recommend ubilinux.
Secondly opkg can handle anything apt-get can (at least at this simplistic
level). But that discussion is best left to the intel maker forum.

You don't need to build anything to get mraa on the std yp images, its on
there just update it via opkg if you want.

I don't know how you plan on using pin13 since the gpio block doesn't have
that exposes, you'll need to use the docs/Edison.md pin conversion table to
work out which io maps to what there are enums to help you out for Edison
look in types.hpp they'll match your board markings

geotheory, 2015-11-05:
Probably am mixing it up - Yocto is fairly new to me.  Sounds like I need to try opkg.  Not sure why the references to pin13 though - my code is 31 (which I gather is gpio44).

arfoll, 2015-11-05:
Ah right yes my bad that should work yes. @srware, @tingleby could you give
this a quick spin if you have a miniboard?

geotheory, 2015-11-05:
On Yocto side I get the following error with `import mraa` in Python:

    Failed to write to export
    Intel Edison Failed to initialise Arduino board TriState,                         check i2c devices! FATAL
    Failed to write to export
    Intel Edison Failed to initialise Arduino board TriState,                         check i2c devices! FATAL

`opkg update` gets a couple of packages then stops with:

    Collected errors:
     * opkg_conf_parse_file: /etc/opkg/base-feeds.conf:1: Ignoring invalid line: `rc/gz all http://repo.opkg.net/edison/repo/all'

geotheory, 2015-11-05:
@arfoll On Yocto side I get the following error with `import mraa` in Python:

    Failed to write to export
    Intel Edison Failed to initialise Arduino board TriState,                         check i2c devices! FATAL
    Failed to write to export
    Intel Edison Failed to initialise Arduino board TriState,                         check i2c devices! FATAL

`opkg update` gets a couple of packages then stops with:

    Collected errors:
     * opkg_conf_parse_file: /etc/opkg/base-feeds.conf:1: Ignoring invalid line: `rc/gz all http://repo.opkg.net/edison/repo/all'

alext-mkrs, 2015-11-05:
Nice, thanks for reporting back.

closed 2015-11-05 by alext-mkrs

geotheory, 2015-11-06:
@arfoll I'd be very interested to hear your reservations about Ubilinux.  We'l be considering longer-term strategic options at some point, and whether to migrate to full Yocto vs Ubilinux/other will be a key question.  [This reddit](https://www.reddit.com/r/IntelEdison/comments/2zqcx9/advantage_of_using_distros_other_than_yocto/) might be a suitable place for discussion.  Views from others also much appreciated..

arfoll, 2015-11-06:
I'm actually currently on vacation in Scotland but I'll happily comment in
there when I get back on Tuesday.

arfoll, 2015-11-09:
@geotheory was going to comment on the reddit post but it's locked/closed so couldn't reply.

geotheory, 2015-11-09:
Oh yes. [Here you go.](https://www.reddit.com/r/IntelEdison/comments/3s77l0/yocto_vs_ubilinux_for_intel_edison/)