eclipse-mraa/mraa #357: mraa not properly installing on Edison running node 0.12.27+

mikaye, 2015-11-09:
After upgrading my edison to node 0.12.27, mraa@0.8.1 fails to npm install. I double checked that npm install works fine for mraa@0.8.0.

tingleby, 2015-11-09:
How does it fail,
Could you provide the output from NPM perhaps.

mikaye, 2015-11-09:
Console -
make: *** [Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/tmp/node_modules/mraa
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Linux 3.10.17-poky-edison+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "mraa"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! mraa@0.8.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@0.8.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/root/tmp/npm-debug.log

npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'install', 'mraa' ]
2 info using npm@2.11.3
3 info using node@v0.12.7
4 verbose install initial load of /home/root/tmp/package.json
5 verbose installManyTop reading scoped package data from /home/root/tmp/node_modules/mraa/package.json
6 info package.json mraa@0.8.1 No repository field.
7 verbose readDependencies loading dependencies from /home/root/tmp/package.json
8 verbose install mraa is installed as 0.8.1
9 silly cache add args [ 'mraa', null ]
10 verbose cache add spec mraa
11 silly cache add parsed spec { raw: 'mraa',
11 silly cache add   scope: null,
11 silly cache add   name: 'mraa',
11 silly cache add   rawSpec: '',
11 silly cache add   spec: '*',
11 silly cache add   type: 'range' }
12 silly addNamed mraa@*
13 verbose addNamed "*" is a valid semver range for mraa
14 silly addNameRange { name: 'mraa', range: '*', hasData: false }
15 silly mapToRegistry name mraa
16 silly mapToRegistry using default registry
17 silly mapToRegistry registry https://registry.npmjs.org/
18 silly mapToRegistry uri https://registry.npmjs.org/mraa
19 verbose addNameRange registry:https://registry.npmjs.org/mraa not in flight; fetching
20 verbose request uri https://registry.npmjs.org/mraa
21 verbose request no auth needed
22 info attempt registry request try #1 at 17:36:48
23 verbose request id d0700a5389408431
24 verbose etag "A8U7CZY0M62W8UVVJADPI2K4J"
25 http request GET https://registry.npmjs.org/mraa
26 http 304 https://registry.npmjs.org/mraa
27 silly get cb [ 304,
27 silly get   { date: 'Mon, 09 Nov 2015 17:36:48 GMT',
27 silly get     via: '1.1 varnish',
[... 437 lines cut ...]
300 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "mraa"
301 error node v0.12.7
302 error npm  v2.11.3
303 error code ELIFECYCLE
304 error mraa@0.8.1 install: `node-gyp rebuild`
304 error Exit status 1
305 error Failed at the mraa@0.8.1 install script 'node-gyp rebuild'.
305 error This is most likely a problem with the mraa package,
305 error not with npm itself.
305 error Tell the author that this fails on your system:
305 error     node-gyp rebuild
305 error You can get their info via:
305 error     npm owner ls mraa
305 error There is likely additional logging output above.
306 verbose exit [ 1, true ]
307 verbose unbuild node_modules/mraa
308 info preuninstall mraa@0.8.1
309 info uninstall mraa@0.8.1
310 verbose unbuild rmStuff mraa@0.8.1 from /home/root/tmp/node_modules
311 info postuninstall mraa@0.8.1
312 silly gentlyRm /home/root/tmp/node_modules/mraa is being purged from base /home/root/tmp
313 verbose gentlyRm don't care about contents; nuking /home/root/tmp/node_modules/mraa
314 silly vacuum-fs purging /home/root/tmp/node_modules/mraa
315 silly vacuum-fs removing /home/root/tmp/node_modules
316 silly vacuum-fs finished vacuuming up to /home/root/tmp

alext-mkrs, 2015-11-09:
Looks like there's nothing specific in this log unfortunately aside from the fact that node-gyp rebuild failed. It mentions npm-debug.log, could you please post that one, if it has any additional information?

mikaye, 2015-11-09:
The output from npm-debug.log is below the console output. The only other information is that it also failed when installing from npm node 5.0

FlippieCoetser, 2015-11-09:
I also had endless trouble to update node. Ended up with a re flash of Yocto and just updated MRAA to 8.1. Would be great if there is a good walkthrough of both updating Node and MRAA on the latest Yocto image on Edison...

arfoll, 2015-11-11:
Tbh node.js is seriously starting to get on my nerved, the v8 API keeps changing, v5.0.0 breaks more stuff.

I'll likely drop support for everything before v4 as that's the lastest LTS and we're starting to drag a bunch of untested code in leading to such bugs. Any objections?

arfoll, 2015-11-11:
So I just managed to build this with node.js 0.12.7 using Arch. Can you try rebuild using cmake from master? I believe the latest fix for node v5.0.0  I just put in maybe have fixed this as well....

alext-mkrs, 2015-11-11:
> Any objections?

FWIW, I'd vote for that, recently there's a real spike in Node-related problems. I just wonder how it's going to work for Edison and Galileo, for which there are no official 5.0 packages. And as far as Galileo is concerned, building even 0.12.x for it proved to be a problem as all the libraries there are quite old.

mikaye, 2015-11-11:
I'm for ignoring 5.0 until it stabilizes. I had to upgrade to 0.12.7 because 0.10.38 didn't support Promise. I followed these steps ( https://communities.intel.com/thread/80570?sr=stream ) to update it with the exception of step 1.

arfoll, 2015-11-11:
isn't 5.0 stable? At least 4.x seems to be the 'LTS' so not sure why you'd chose 0.12.? This might be a very stupid question I'm really not that well versed in the versioning system of node. Tbh my test was quite quick, I'll do a better one tmrw and try make a travis test so that we can automate the builidng on multiple node.js versions as this has been a common source of pain.

mikaye, 2015-11-11:
Immediately after I started using 5, I was unable to debug with WebStorm IDE. Their support tech told me it was a problem with 5. I then noticed other problems appearing on the web with 5. Today, WebStorm released a new version that addresses the problem they had with 5. I'm sure these are just teething problems and I'll return to 5.x as soon as I decide it's worth the while. 4 probably would have been fine as well.

alext-mkrs, 2015-11-12:
@arfoll, while not directly related, do you think we could setup building python3 bindings as well, alongside python2 ones? That was requested a couple of times and if we setup something for multiple node versions, multiple pythons would come in handy too :smiley:

xbolshe, 2015-11-14:
V8 version translation creates a problem like this. See https://github.com/intel-iot-devkit/mraa/pull/363

rwaldron, 2015-11-14:
> FWIW, I'd vote for that, recently there's a real spike in Node-related problems. I just wonder how it's going to work for Edison and Galileo, for which there are no official 5.0 packages.

FWIW, this is has been felt throughout the community for any project that depends on, or is, a compiled module. The good news is that all of this insane churn is over with; and there is no bad news. I recommend sticking with Node 4.2.1 and up, while dropping everything prior. I would also not rush on Node 5.0, mainly because there there isn't any reason to, which also means that it's totally ok to tell users that this project is on the LTS train. My recommendation is based on the advice [here](https://nodejs.org/en/blog/release/v5.0.0/): 

> The general rule for deciding which version of Node.js to use is:
>
> - Stay on or upgrade to Node.js v4.2.x if you need stability and have a complex production environment, e.g. you are a medium or large enterprise.
> - Upgrade to Node.js v5.x if you have the ability to upgrade versions quickly and easily without distributing your environment and want to play with the latest features as they arrive.

Based on the discussion and general (shared) aversion to churn, I think this project should be allowed to follow the first rule.

xbolshe, 2015-11-14:
On my Galileo boards both NodeJS 4.2.1/5.0.0 versions work with MRAA 0.8.1, sqlite3, etc.
Do you know limitations of MRAA to work on NodeJS 4.x or 5.x only?

rwaldron, 2015-11-15:
> Do you know limitations of MRAA to work on NodeJS 4.x or 5.x only?

Different versions of V8 ship with different versions of Node, but that's not really the point: I was hoping to bolster support for @arfoll's proposal to drop support for pre-4.x versions.

alext-mkrs, 2015-11-15:
Thanks @rwaldron, that's definitely useful input.

arfoll, 2015-11-16:
@rwaldron thanks for opinion and advice on which version we should be on.

@xbolshe there are no limitations codewise about node.js 4.x or 5.x just that we can't test everything and we need a default to recommend. If people want to go outside of that it's fine and we encourage it but those users should expect the unexpected :) Thanks for your PR for converting versions to hex, I do wonder if that's the right approach considering the comments from the SWIG devs. It's good to know future v8 versions have this issue fixed.

alext-mkrs, 2016-01-16:
the discussing went beyond the original issue, but just to let everyone know - @arfoll fixed the 0.12.x-related problem with commit c2681dc78c5 and I've just tested it to work ok on node.js 0.12.7 (the version I have in my repo).

Not sure if we'd want to close this issue, @arfoll?

alext-mkrs, 2016-02-06:
@arfoll, ping

I would vote for closing this one, as the original problem is long solved and if we'd want to tackle the side thread of making Node 4.x or 5.x the only officially supported Node version, we'd rather open a new issue/do it outside of issue tracker at all.

What do you think?

arfoll, 2016-02-07:
Agreed. Actually travis now builds for node 0.10/0.12/4 & 5 so this type of issue is hopefully gone. Still need to implement an npm check though.

closed 2016-02-07 by arfoll

mikaye, 2016-02-07:
I agree as well. It should be closed.

Mike