eclipse-mraa/mraa #369: mraa uart issue

peidong, 2015-11-23:
@alext-mkrs There is an issue about the mraa uart part.
We tried to use Edison break board to receive 37-byte (including a header) package Uno via UART continuously, the code is shown in this file: 
[drone_include.h](https://github.com/peidong/drone/blob/master/Edison/main/include/drone_include.h#L496-L546)

The baud rate is 115200 in both Edison and UNO. And we use an space ' ' as a header of each package to identify the beginning of each package. And here is the problem:
As long as Edison receives 55-56 correct packages, it seems that the buffer is empty. Because for a very long time (e.g. about 50ms), the mraa_uart_data_available returns 0. As far as I know, the buffer size of Edison is 64 bytes. It should take a very short time to refill the buffer if it is empty. By the way, the mraa version is up-to-date(0.8.1).

alext-mkrs, 2015-11-23:
Do you mean that after receiving 55-56 those packages each 37 byte long you see a sort of pause for those 50ms?

How the data sending piece looks like, are you sure that it sends the data continuously and it's not the sending side problem?

What else is running on your Edison in parallel to that program?

Generally that looks to me, at the first glance, as either an OS scheduling artifact or just a non-continuous stream from the sender.

If you could narrow your code down to sender and receiver parts, which I could use to reproduce the issue - that would help.

alext-mkrs, 2015-11-25:
@peidong, is this still relevant?

peidong, 2015-11-25:
@alext-mkrs , We set up a delay on Uno for about 17000us, and it seems better. But still waits for 3ms after several packages.
The uart sender part is [uart_code](https://github.com/peidong/drone/blob/master/Edison/main/include/mpu9250/mpu_uno_read/mpu_uno_read.ino)

We first want to use i2c to connect the gyro sensor with edison, but there is also a freeze bug randomly. Thanks!

peidong, 2015-11-25:
@alext-mkrs , and also, can we use one thread to read from UART and use another thread to write to UART? Will it conflict because they act at the same time?

alext-mkrs, 2015-11-28:
Sorry, that's not that much helpful - a ~1000-line file and I unfortunately don't have that much time to go through it in detail. What I meant was for you to provide a minimal program that reproduces the behavior you're talking about.

I'll see if I can set something like this up and then reproduce the issue.

UART is a physically single device, reading and writing from/to it fromtwo different threads will not make any good, I believe.

alext-mkrs, 2015-11-29:
Yesterday and today I was working on reproducing this and in short I don't think this is an mraa problem. This is generally something why you shouldn't use general-purpose OSes for time-sensitive workloads in userspace and why they've invented RTOSes and bare-metal applications (or MCUs, to that matter) or at least kernel-level workloads (like device drivers) for that.

I've wrote two simple pieces for both ends, sender and receiver. [Sender](https://gist.github.com/alext-mkrs/4891289bcb62668f79a5) simply repeats the same 37-byte message into the serial port. [Receiver](https://gist.github.com/alext-mkrs/7a30f6e130c8f5b7df84) tries to receive that and then gather some statistics. I've also left some commented out `printf()` statements, so that you can play with that if you want to.

Initially I've implemented the receiver exactly the way you did - i.e. first a `mraa_uart_data_available()`, then one-byte read and then 36-byte read. That exhibited similar behavior to what you describe and I think that's an interference between UART HW buffer, kernel buffer, driver, `read()` function and standard jitter within the userland, which causes it to behave that way. Linux userspace programs (or pretty much any non-RTOS ones) are just not guaranteed to be able to serve time-sensitive workloads (which such stream processing is). In your program I see you also do some data processing with the same loop - which should make it even worse.

I've optimized the approach and the result is in the gist I've linked above. That shows much less lost data packets (those 37-byte ones, see exact stats below), because we're now receiving them in a byte-by-byte mode (just like the UART sends and receives them) +  there is no extraneous processing.

I'd suggest you to take my gist as a starting point and implement your program in a similar way. Offload all the calculations and processing to a separate thread and let the receiver run in a thread on its own, doing only data recive tasks - if you want to keep it in userspace and insist on small amount of lost frames. That is different from what you've asked earlier - we are separating data receive (I/O operation, low computation intensivity, high sensitivity to delays) and data processing (vice versa).

Here are the results from running sender (on Galileo Gen2) and receiver (on Edison). You can see how the packet loss rate slightly increases with baud rate increase (because of the jitter and reasons I've described above), but it's still quite low, just several packets per 100K sent ones.

With baud rate of 921600, packet loss rate average is about 0.169%
```
root@edison1:~# ./mraa-369-receiver-optimized
full messages received: 99842
read failures: 0
non-flag symbols received: 158
root@edison1:~# ./mraa-369-receiver-optimized 
full messages received: 99837
read failures: 0
non-flag symbols received: 163
root@edison1:~# ./mraa-369-receiver-optimized 
full messages received: 99814
read failures: 0
non-flag symbols received: 186
```

With baud rate of 115200, packet loss rate average is about 0.051%
```
root@edison1:~# ./mraa-369-receiver-optimized 
full messages received: 99929
read failures: 0
non-flag symbols received: 71
root@edison1:~# ./mraa-369-receiver-optimized 
full messages received: 99962
read failures: 0
non-flag symbols received: 38
root@edison1:~# ./mraa-369-receiver-optimized 
full messages received: 99956
read failures: 0
non-flag symbols received: 44
```

closed 2015-11-29 by alext-mkrs