eclipse-mraa/mraa #375: Mraa autocomplete Python

BrechtW, 2015-12-02:
Is it possible to get autocompletion in Python an IDE?

I tried PyCham and iPython without luck.

arfoll, 2015-12-02:
@tingleby didn't you try this?

tingleby, 2015-12-03:
No, I haven't. But something worth looking at.

alext-mkrs, 2015-12-03:
Looks like this might be just a matter of [properly configuring PyCharm](http://stackoverflow.com/questions/5049842/autocomplete-in-pycharm-for-python-compiled-extensions?rq=1), @BrechtW, check this out.

BrechtW, 2015-12-04:
I tried to remove and add the python interpreter. I copied the mraa folder in c:\Python27\Lib\site-packages and did Project Interpreter>Gear Icon>More>Show path for selected interpreter>Refresh. The autocomplete still doesn't work. I still get "No module named mraa".

alext-mkrs, 2015-12-04:
Ah, looks like you're doing this on Windows, that's not going to work. Python binding consists of a simple .py wrapper, which imports a compiled binary (shared object, .so), which is compiled out of SWIG-generated code and is OS-specific, with Windows not being supported (mraa uses Linux-specific mechanisms). And autocomplete needs to actually import it to work, AFAIU.

BrechtW, 2015-12-04:
Is it possible to make a sort of list of functions and properties and let PyCharm derive autocomplete of this?

alext-mkrs, 2015-12-07:
It would look like it's not possible. Both in PyCharm and IPython autocompletion works by actually importing Python modules and indexing/analyzing them, not by using some additional dictionary file or something like this. So when you have such a "Windows IDE - Linux-only module" situation it's beyond of their capabilities.

Just as a wild idea you could try running indexing in PyCharm on Linux and then moving whatever cache it generates to Windows, however that's really a rude hack and I don't believe it's going to work.

alext-mkrs, 2016-01-01:
Unless anyone has any other ideas, I think we just have to state syntax autocompletion is not supported for Windows-based IDEs due to (a) the fact that mraa is a Linux-only library and (b) the way these IDEs work with autocompletion - by trying to actually import the module and retrospect it.

@arfoll, @tingleby, what do you think?

alext-mkrs, 2016-05-18:
As mentioned previously, the way autocomplete works is simply not appropriate for a situation when you have a Linux-compiled Python module and Windows-hosted IDE, so I don't think we can make it work. Closing this one.

closed 2016-05-18 by alext-mkrs