eclipse-mraa/mraa #379: SPI Maximum transfer Length

leonyuhanov, 2015-12-07:
I noticed in /src/spi/spi.c there is a line defining the maximum single transfer length of an SPI buffer:

#define SPI_MAX_LENGTH 4096

Is there a reason for this particular limit?
Can this limit be increased during run-time or does it have to be changed and then recompiled?
Is there any impact to it being increased to 4 times its size?

I also noticed that SPI transfers of larger than 4096 bytes cause a "Seg Fault" when run in NodeJS on a Raspberry Pi A+. Works fine if i split the packet into segments and transmit in 4096 byte blocks...

arfoll, 2015-12-07:
Yes, there is, that's the maximum message size allowed via spidev using the linux kernel. However we don't actually use that define so it's useless in the code :/

It may be wise to enforce that limit strictly.

leonyuhanov, 2015-12-07:
Perfect thanks. Im happy splittng my message into blocks

closed 2015-12-07 by leonyuhanov