eclipse-mraa/mraa #387: ATAL ERROR: v8::HandleScope::CreateHandle() with nodejs (4.2.3) and button.isr()

Torsten85, 2015-12-23:
Hey there,
I'm totally new to the hole microprocessor stuff so there might be a good chance that I'm missing something.

I've installed mraa 0.9.0 on my edison and I'm using it with the [sparkfun OLED block] (https://learn.sparkfun.com/tutorials/sparkfun-blocks-for-intel-edison---oled-block-?_ga=1.205389012.595082432.1450255418).

Currently I'm trying to read the button state of that block by following the [isr.js](https://github.com/intel-iot-devkit/mraa/blob/master/examples/javascript/isr.js) example.

When I run that program and press the corresponding button, the program crashes with:
`FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope`

What am I missing? The code is really simple and should work:

```javascript
var m = require('mraa');

function h(args) {
  console.log("HELLO!!!!", args);
}

x = new m.Gpio(m.INTEL_EDISON_GP49);
x.isr(m.EDGE_BOTH, h);

(function loop() {
  setTimeout(loop, 1000);
})();
```

Thanks in advance.

arfoll, 2015-12-23:
At a quick glance can't spot anything obvious. Which version of node is this on and how did you install mraa (npm/opkg/src?) Can you also check mraa version being included with a getVersion() call. It may be worth trying a downgrade real quick as well.

I'll give this a quick double check when I get to a computer later today.

Torsten85, 2015-12-23:
I tried Node 5.3 and 4.2.3 and I've installed mraa via npm. Also I've tried mraa version 8.1 and 9.

arfoll, 2015-12-23:
So does this happen when the button is pressed? i.e. when the isr/h() runs?

arfoll, 2015-12-23:
Actually can you try to add `Nan::EscapableHandleScope scope;` on before line 167 in api/mraa/gpio.hpp? It looks like that error is scope related and that may just fix it.

Torsten85, 2015-12-23:
Yes the crash happens the moment the button is pressed. The callback function is never called.

I've tried to add the line of code you mentioned, but that brings up errors while building:

```
 In file included from ../src/mraajsJAVASCRIPT_wrap.cxx:1674:0:
../api/mraa/gpio.hpp: In static member function ‘static void mraa::Gpio::v8isr(uv_work_t*, int)’:
../api/mraa/gpio.hpp:167:9: error: ‘Nan’ has not been declared
../api/mraa/gpio.hpp:167:35: error: expected ‘;’ before ‘scope’
```

CurtisReedy, 2016-01-19:
I have run into the same issue as Torsten85 using node version 4.2.4 while installing mraa version 0.9.0 from npm. I tried adding "Nan::EscapableHandleScope scope;" between lines 166 and 167 and it built, but threw the same error. Ultimately I had to revert back to node version 0.10.36 (note that I am using ublinux and version 0.10.36 would not npm install mraa until I reflashed my edison), I am not sure what was causing the error but everything is working very smoothly with mraa version 0.9.0 and node version 0.x.x. If this error is possible to fix or someone has already found a fix I would love to hear about it, thanks.

closed 2016-02-29 by arfoll

mihok, 2016-03-07:
I'm also seeing this error and not sure why.

node v5.6.0
npm v3.9.0
mraa v0.9.4

```
"use strict";

var m = require('mraa');
console.log('MRAA Version: ' + m.getVersion());

var counter   = 0;
var interrupt = 0;
var led_state = 0;
var argument  = 'myargument';

/*
function callback(arg) {
  interrupt++;
  console.log(arg);
}
*/

var callback = function() {
  interrupt++;
  console.log(arguments);
};

var myled    = new m.Gpio(6);
myled.dir(m.DIR_OUT);

var mybutton = new m.Gpio(2);
mybutton.dir(m.DIR_IN);
mybutton.isr(m.EDGE_BOTH, callback);

function mainloop() {
  myled.write(led_state = led_state ? 0 : 1);
  counter++;
  console.log('counter = ' + counter + ' -- interrupt done = ' + interrupt);
  setTimeout(mainloop, 1000);
}

mainloop();
```

I get the same error:
```
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope
```

arfoll, 2016-03-07:
@alext-mkrs is it possible that we're missing a `scope.Escape(self)` in there?

alext-mkrs, 2016-03-07:
I think we don't as I've considered that when preparing a patch, but let me double-check tomorrow when I'm in front of my dev machine.

mihok, 2016-03-08:
Ah thanks @arfoll and @alext-mkrs, looks like mraa 0.9.5 solves the issue!! Thanks! :+1:

arfoll, 2016-03-08:
@alext-mkrs Ooops I totally missed that it was on 0.9.4 and not 0.9.5.

alext-mkrs, 2016-03-08:
Glad we've clarified that :)

commit 350a6489 (2016-02-29): gpio.hpp: added HandleScope creation for Node 4+

Closes #387.