eclipse-mraa/mraa #391: Unexpected behaviour of "mraa_spi_transfer_buf( )" - C

agaelema, 2015-12-31:
I'm developing a lib to read/write some SPI flash memories like the W25x40/Winbond.

Some unexpected behavior has occurred when using the "mraa_spi_transfer_buf( )" function for transferring certain amount of data sequentially.

For 20 bytes or less the code works well, but for more the code crash randomly, beeing necessary reboot the Edison.

I'm using the last version of MRAA on the Eclipse.

[Copy of SPI_-_Flash_mem_large_array.txt](https://github.com/intel-iot-devkit/mraa/files/75788/Copy.of.SPI_-_Flash_mem_large_array.txt)

alext-mkrs, 2016-01-01:
Could you please post the exact crashes you observe?

There were several fixes for SPI for similar symptoms recently, which I think were not yet rolled up to an official image release, see the full list below. If you're up to that, you can build your own image with these added to see if that helps.

https://github.com/01org/edison-linux/commit/3964570409edc05d16d38a61644906747d4ecb5c
https://github.com/01org/edison-linux/commit/c527c66918be70f90255bf85d6435c2615bbd6bb
https://github.com/01org/edison-linux/commit/2d7dacd26d06afec9fd4d7a5df73bc4f77571195

agaelema, 2016-01-01:
Thanks for the feedback.

The crash is a little random. I Tried to send/receive one array with 2000 elements. Sometimes the code finish, sometimes it crash during the transfer. After this crash , eclipse shows an error when I try to execute the code ("error during file upload"), being necessary reboot the system.

It's possible to enable a debug log to verify what's happening?

I tried other function, "mraa_spi_write_buf( )", but the same occur.

I think that the first link listed is related with this crash.

One more question, it's possible to use the last version of MRAA without need to build and install a new image?

Thanks.

alext-mkrs, 2016-01-04:
The library logs errors to the OS' journal, you can check that with `journalctl --no-pager -xn40`. I don't think it's accessible from the Eclipse, you'll need to log onto the board using SSH.

As far as updating mraa versions - sure, by default on official images you have an opkg repo with all mraa and UPM releases configured. Just run "opkg update && opkg install mraa" to update it.

If that's indeed that kernel problem, then we unfortunately can do nothing on the mraa level - that's a kernel thing we have no influence onto. But as I mentioned, you can build your own image with that patch. And I think sooner or later there might be another official image release hopefully with this fix incorporated.

Let's see OS logs to try to confirm this hypothesis.

alext-mkrs, 2016-01-13:
@agaelema, so what's in the system logs?

alext-mkrs, 2016-01-17:
@agaelema is this still actual for you or you'll just wait for the kernel update/update it on your own?

agaelema, 2016-01-17:
Hi @alext-mkrs,

The problem keeps happening.

This is the log using the command "journalctl --no-pager -xn40".

--------------------------------------------------------------------------------------
root@edison:~# journalctl --no-pager -xn40
-- Logs begin at Sat 2000-01-01 00:00:11 UTC, end at Sun 2016-01-17 21:14:56 UTC. --
Jan 17 21:14:56 edison kernel:  [<c15c1130>] ? spidev_write+0xb0/0xb0
Jan 17 21:14:56 edison kernel:  [<c15c1380>] ? spidev_probe+0x190/0x190
Jan 17 21:14:56 edison kernel:  [<c1336686>] do_vfs_ioctl+0x306/0x550
Jan 17 21:14:56 edison kernel:  [<c1480cba>] ? inode_has_perm.isra.32.constprop.65+0x3a/0x50
Jan 17 21:14:56 edison kernel:  [<c1480d57>] ? file_has_perm+0x87/0x90
Jan 17 21:14:56 edison kernel:  [<c148113b>] ? selinux_file_ioctl+0x4b/0xe0
Jan 17 21:14:56 edison kernel:  [<c1336930>] SyS_ioctl+0x60/0x90
Jan 17 21:14:56 edison kernel:  [<c18df928>] syscall_call+0x7/0xb
Jan 17 21:14:56 edison kernel: INFO: task SPI_-_Flash_mem:457 blocked for more than 120 seconds.
Jan 17 21:14:56 edison kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jan 17 21:14:56 edison kernel: SPI_-_Flash_mem D f54e1b40  6348   457      1 0x00000004
Jan 17 21:14:56 edison kernel:  f518fe40 00000086 00000000 f54e1b40 f5cb525c c1d04000 c18dc63d f73fdc00
Jan 17 21:14:56 edison kernel:  c15c0f2b f6c83d30 f73fdc00 f51bfa60 f518fdfc c1270ddb 00000001 f518fe18
Jan 17 21:14:56 edison kernel:  c18e2c75 f639cdcc f54e1b40 00000002 f639cd80 f73fa894 f518fe24 c1270ddb
Jan 17 21:14:56 edison kernel: Call Trace:
Jan 17 21:14:56 edison kernel:  [<c18dc63d>] ? mutex_unlock+0xd/0x10
Jan 17 21:14:56 edison kernel:  [<c15c0f2b>] ? spidev_open+0x6b/0x110
Jan 17 21:14:56 edison kernel:  [<c1270ddb>] ? get_parent_ip+0xb/0x40
Jan 17 21:14:56 edison kernel:  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0
Jan 17 21:14:56 edison kernel:  [<c1270ddb>] ? get_parent_ip+0xb/0x40
Jan 17 21:14:56 edison kernel:  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0
Jan 17 21:14:56 edison kernel:  [<c18e2b6d>] ? add_preempt_count+0x7d/0xf0
Jan 17 21:14:56 edison kernel:  [<c18de57d>] schedule_preempt_disabled+0x2d/0x70
Jan 17 21:14:56 edison kernel:  [<c18dc75b>] __mutex_lock_slowpath+0x11b/0x360
Jan 17 21:14:56 edison kernel:  [<c1333173>] ? do_last+0x153/0xd50
Jan 17 21:14:56 edison kernel:  [<c18dc9c3>] mutex_lock+0x23/0x30
Jan 17 21:14:56 edison kernel:  [<c15c13fe>] spidev_ioctl+0x7e/0x6f0
Jan 17 21:14:56 edison kernel:  [<c1333e11>] ? path_openat+0xa1/0x3f0
Jan 17 21:14:56 edison kernel:  [<c14f50c6>] ? pca953x_write_single.isra.3+0x56/0x90
Jan 17 21:14:56 edison kernel:  [<c147d957>] ? avc_has_perm_flags+0xd7/0x170
Jan 17 21:14:56 edison kernel:  [<c13347db>] ? do_filp_open+0x2b/0x90
Jan 17 21:14:56 edison kernel:  [<c15c1380>] ? spidev_probe+0x190/0x190
Jan 17 21:14:56 edison kernel:  [<c1336686>] do_vfs_ioctl+0x306/0x550
Jan 17 21:14:56 edison kernel:  [<c1480cba>] ? inode_has_perm.isra.32.constprop.65+0x3a/0x50
Jan 17 21:14:56 edison kernel:  [<c1480d57>] ? file_has_perm+0x87/0x90
Jan 17 21:14:56 edison kernel:  [<c132ff01>] ? lookup_fast+0x1d1/0x2a0
Jan 17 21:14:56 edison kernel:  [<c13301a4>] ? putname+0x24/0x40
Jan 17 21:14:56 edison kernel:  [<c148113b>] ? selinux_file_ioctl+0x4b/0xe0
Jan 17 21:14:56 edison kernel:  [<c1336930>] SyS_ioctl+0x60/0x90
Jan 17 21:14:56 edison kernel:  [<c18df928>] syscall_call+0x7/0xb
--------------------------------------------------------------------------------------

alext-mkrs, 2016-01-18:
Thanks. Yes, this looks very much like those SPI bugs fixed in the kernel, so unfortunately we can do nothing about them in the library. As I mentioned, you'll need to wait for the official kernel release with those incorporated or build one youself.

closed 2016-01-18 by alext-mkrs

agaelema, 2016-01-18:
@alext-mkrs Thanks for the answer. There is a tutorial of how to build my version, I'm begginer in linux.

alext-mkrs, 2016-01-19:
There's so called BSP building guide, which has all the build process ste by step. If you have further questions on the process - feel free to start a thread in Intel Edison community on communities.intel.com