eclipse-mraa/mraa #393: How compile mraa on Edison

massca, 2016-01-03:
Hi, I need to compile on Edison mraa (and upm) to work with node.js 0.12.x

I followed mraa github build instructions. I starded installing Edison OS 2.1 and run configure_edison.
According to build doc I checked/installed the prerequisites:
SWIG 3.0.5: already installed
python 2.7: already installed
CMake 2.8.8: already installed

# install git
opkg install http://repo.opkg.net/edison/repo/core2-32/git_2.0.1-r0_core2-32.ipk 

# install node.js 10.2.x  (intepreter plus dev)
opkg install http://repo.opkg.net/edison/repo/core2-32/nodejs-npm_0.12.7-r1_core2-32.ipk     
opkg install http://repo.opkg.net/edison/repo/core2-32/nodejs-dev_0.12.7-r1_core2-32.ipk  

Then I cloned mraa from github:
git clone --recursive https://github.com/intel-iot-devkit/mraa.git

And finally I built mraa:
cd mraa/
mkdir build
cd build
cmake ..
make

Compile ends without errors, but when I try to use mraa from a nodejs program (like "Onboard blink" XDK sample) I got the that the mraa version is v0.7.2, ie the version installed in the Edison OS2.1, instead of v.0.9.x that I cloned from github.
 
It seems I missed something to enable node.js programs to use the mraa version that I just compiled.
Can you please help me? 
Thanks

arfoll, 2016-01-03:
Read docs/compiling.md, you need to set the prefix to /usr and use `make install`.

massca, 2016-01-03:
I'm sorry I miss this obvious step. Now it works as expected.
Thanks!

closed 2016-01-03 by massca