eclipse-mraa/mraa #396: Installation error Intel Edison Node v0.12.7 and npm 2.11.3

Nishanksingla, 2016-01-04:
I am having node version 0.12.7 and npm version 2.11.3 in intel edison and while installing "mraa", I am getting following error in the end:

gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.10.17-poky-edison+
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/node_modules/mraa
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
npm ERR! Linux 3.10.17-poky-edison+
npm ERR! argv "node" "/usr/bin/npm" "install" "mraa"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! mraa@0.9.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the mraa@0.9.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/root/npm-debug.log

Can you please help me in installing "mraa" ? 
I am unable to understand about what's going wrong ?
Please help me in installing "mraa" .

arfoll, 2016-01-04:
What edison image are you using and how did you install node.js 0.12? My guess is you're missing the -dev package. If you're on ublinux you'll have to recompile node.js from scratch, if you're on Poky/IOTDK then this should work, can you upload `/home/root/npm-debug.log` to dpaste.com or similar pastebin?

Nishanksingla, 2016-01-04:
I am using latest yocto image and I am getting "159.devkit"  version name when I run  "configure_edison --version" command. 
I have downloaded node 0.12 from this opkg repo "http://repo.opkg.net/edison/repo/core2-32/"
Here is the npm-debug.log file.
[npm-debug.log.txt](https://github.com/intel-iot-devkit/mraa/files/77372/npm-debug.log.txt)

arfoll, 2016-01-04:
Urgh there is indeed an issue with node 0.12.x when compiling from NPM. This does not happen when you compile from source please use that method for node 0.12.x, I'm guessing it's to do with the version conversion from node.js that was modified in 0.9.x, node.js has a weird versioning structure, it works on node 5 or node 0.10 but not 0.12 :/

Nishanksingla, 2016-01-04:
OK 
which method you are talking about in this sentence 
"this does not happen when you compile from source please use that method for node 0.12.x,"
Can you tell me steps for this method ?

arfoll, 2016-01-04:
See docs/building.MD. you can build on the Edison platform from git quite
easily, all the tools are already installed

Nishanksingla, 2016-01-06:
Thanks I will try this method.
If I uninstall current version of node and npm, then which version of node I have to install to get mraa library.

arfoll, 2016-01-06:
Mraa will build for your version of node, the only combination that doesn't work is installing from NPM with node 0.12.x. I'm not entirely sure why yet but looking into it.

Nishanksingla, 2016-01-06:
OK 
Tell me the version of npm and node for it works. I will install those version. 
The document "docs/building.MD" you told me to refer is not so clear, I am not getting the steps. 
The document is written for apt-get and in edison opkg is used.

arfoll, 2016-01-06:
Follow the 'basic steps' in the documentation, those will work on opkg. You don't need to install any extra dependencies, they're all there.

```
mkdir build
cd build
cmake ..
make
```

in src/javascript/ you'll have your 'mraa.node' module.

Using this method, any node version will work.

Nishanksingla, 2016-01-06:
OK 
Just one thing in which directory I have make build directory, in cloned repo directory or in src/javascript/ directory

Nishanksingla, 2016-01-06:
OK 
I have done the following steps
1.) cloned this repo
2.) cd to the repo
3.) mkdir build
4.)cd build
5.)cmake ..
6.)make

Now, I have got src/javascript folder inside build directory and I have mraa.node module also.
now Which command to run to install "mraa" node module ?

arfoll, 2016-01-07:
just run `make install`, or pop what's in build/src/javascript wherever your node modules get loaded from.

alext-mkrs, 2016-01-13:
Just so you know, I'm going to take a deeper look into this one, as it seems to be quite strange that it builds fine standalone and fails in npm. And as long as we have official image shipping with older node (and some components like xdk-agent and so on depend on the version), it will still be around for a while, so I think it's worth investing at least some time to try to fix it.

alext-mkrs, 2016-01-13:
Hmm, looks like I've missed some of the commit notifications, @arfoll, doesn't your commit c2681dc78c56ec2e9df88982bf05bf8c0fa18b04 by chance address this very problem we have here?

Nishanksingla, 2016-01-13:
To get mraa, I have to reflash my edison to get default version of node and npm. 
Now mraa is installed but now I am getting a strange problem
For the first time when I run PWM program to control servo motor it runs good for few time after that motor got burnt.(never happen this thing before with arduino code and python mraa code)

After this when I connected new servo and as soon as I run the program, Edison gets crashed and start rebooting.
Do you have any idea whats happening ? Why intel is getting crashed and reboot?
Can you provide me any solution for this ?

alext-mkrs, 2016-01-13:
Well, that's quite a generic description, which has not enough information to troubleshoot this and this honestly doesn't look like a library problem - such issues are much better suited for Intel Community forums at https://communities.intel.com/community/tech/edison.

Nishanksingla, 2016-01-13:
Ok thanks for the answer.
one more thing
Have you ever tried to run 2 servo using mraa pwm and intel edison ?

alext-mkrs, 2016-01-13:
I personally haven't

alext-mkrs, 2016-01-16:
And just to confirm for those who look for a solution - commit c2681dc78c56ec2e9df88982bf05bf8c0fa18b04 by @arfoll indeed fixes this problem - great job!

Until this is released to npm the workaround is to

1. download the npm tar.gz file manually (using wget, the full URL you can find by running `npm info mraa`)
1. unpack it using tar into some directory
1. manually make the edit introduced by the abovementioned commit in `binding.gyp` file
1. install the module using npm from that directory by running `npm install <path/to/dir>`

closed 2016-01-16 by alext-mkrs