eclipse-mraa/mraa #398: Problems to compile mraa on Edison running node 0.12.7

massca, 2016-01-05:
Hi, I was able to build on Edison (OS 2.1) mraa with node 0.10.x following the steps in the build doc. 
Unfortunately when I try the same steps with node 0.12.7 (installed from alexT repo) I got following errors:
root@max01:~/mraa/build# make 
[ 21%] Built target mraa
[ 46%] Built target _python-mraa
[ 47%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
/home/root/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:778:13: error: 'FunctionCallbackInfo' in namespace 'v8' does not name a template type
 typedef v8::FunctionCallbackInfo<v8::Value> SwigV8Arguments;
...
(lots of errors like before)
...
/home/root/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:9740:18: note: in expansion of macro 'SWIGV8_SYMBOL_NEW'
 exports_obj->Set(SWIGV8_SYMBOL_NEW("Uart"), _exports_Uart_obj);
                  ^
src/javascript/CMakeFiles/mraajs.dir/build.make:60: recipe for target 'src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o' failed
make[2]: *** [src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:231: recipe for target 'src/javascript/CMakeFiles/mraajs.dir/all' failed
make[1]: *** [src/javascript/CMakeFiles/mraajs.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2

arfoll, 2016-01-05:
Make sure to wipe the build directory so that cmake doesn't reuse cached values for nodejs versioning decisions. This worked on edison with a node 0.12.7 I compiled from source so no reason it shouldn't work here.

alext-mkrs, 2016-01-05:
+1, I was building it fine as well.

massca, 2016-01-07:
It works, I forgot to upgrade nodejs-dev. Thanks!

closed 2016-01-07 by massca