eclipse-mraa/mraa #403: Segmentation fault while using interrupt

sonofaforester, 2016-01-08:
My program receives an interrupt and then reads from the spi bus. While running, usually after 30 seconds or so, I'll get a segmentation fault. I'm using mraa v0.9.0 and node v0.12.7.

Using segfault-handler, I get the following:

PID 719 received SIGSEGV for address: 0x6960001c
/kcftech/socket/bin/node_modules/segfault-handler/build/Release/segfault-handler.node(+0x1288)[0xb77df288]
linux-gate.so.1(__kernel_rt_sigreturn+0x0)[0xb77e240c]
node(_ZN2v88Function4CallENS_6HandleINS_5ValueEEEiPS3_+0x13)[0x8191ab3]
/usr/lib/node_modules/mraa/mraa.node(_ZN4mraa4Gpio5v8isrEP9uv_work_si+0x118)[0xb6f97f21]
node(uv__work_done+0x86)[0x8557cb6]
node[0x85597d6]
node[0x8559885]
node(uv__io_poll+0x393)[0x8566273]
node(uv_run+0x112)[0x855a132]
node(_ZN4node5StartEiPPc+0x18f)[0x84ee3bf]
node(main+0x1b)[0x817295b]
/lib/libc.so.6(__libc_start_main+0xd9)[0x4e585b59]
node[0x8172c1c]

arfoll, 2016-01-08:
Do you get the segfault when the interrupt is fired or randomly? How often does the interrupt fire? Just trying to get data so I can reproduce.

Also you may want to quickly try using master, ff6b304fd2225ed7147ae3e7d964fd37aa934e8f may fix some memory allocations on 0.12.x node.js.

alext-mkrs, 2016-01-13:
Also, what do you see in the system log, `journalctl -xn 50 --no-pager` after it crashes?

sonofaforester, 2016-01-13:
Thanks for looking at this. Let me get back to you, I don't have the hardware to test it on this week.

To answer the earlier question, pulled master branch on 1/7/2016 and had the same problem. The interrupt fires about every 100ms, can't say if the segfault is on the interrupt or random but it only happens when receiving interrupts. When I get my hardware back, I'll test.

alext-mkrs, 2016-01-13:
Thanks. BTW if you're able to post your code (or better yet a reproducer - as short as possible) along with your HW description, that should help troubleshooting a lot.

alext-mkrs, 2016-01-25:
@sonofaforester so how about logs and ideally a reproducer for this one?

sonofaforester, 2016-01-29:
Here's a reproducer. I'm using Edison attached to the Sparkfun RaspPi block. I've got a square wave running at 100Hz on GP46. After about a minute or so the segmentation fault will occur.

var m = require('mraa');
console.log(m.getVersion());
var x = new m.Gpio(32); //GPIO Pin 8

var spi = new m.Spi(0);
spi.frequency(1000000);

function char(x) {
   return parseInt(x,16);
}

function a() {
var buf = new Buffer(4);
buf[0] = char('0xf4');
buf[1] = char('0x2e');
buf[2] = char('0x3e');
buf[3] =char('0x4e');
var buf2 = spi.write(buf);
}

x.isr(m.EDGE_BOTH, a);

setInterval(function(){ console.log("keep alive");
}, 10000);

Log you asked for is here:

-- Logs begin at Sat 2000-01-01 00:00:11 UTC, end at Fri 2016-01-29 15:15:20 UTC. --
Jan 29 15:15:13 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:13 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:13 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:14 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:14 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:14 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:14 edison xdk-daemon[15937]: Current daemon started
Jan 29 15:15:14 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:14 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:15 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:15 edison xdk-daemon[15937]: /opt/xdk-daemon/current/node_modules/mdns/lib/dns_sd.js:35
Jan 29 15:15:15 edison xdk-daemon[15937]: throw ex;
Jan 29 15:15:15 edison xdk-daemon[15937]: ^
Jan 29 15:15:15 edison xdk-daemon[15937]: Error: Module did not self-register.
Jan 29 15:15:15 edison xdk-daemon[15937]: at Error (native)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Module.load (module.js:355:32)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Function.Module._load (module.js:310:12)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Module.require (module.js:365:17)
Jan 29 15:15:15 edison xdk-daemon[15937]: at require (module.js:384:17)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Object.<anonymous> (/opt/xdk-daemon/current/node_modules/mdns/lib/dns_sd.js:24:20)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Module._compile (module.js:460:26)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Object.Module._extensions..js (module.js:478:10)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Module.load (module.js:355:32)
Jan 29 15:15:15 edison xdk-daemon[15937]: at Function.Module._load (module.js:310:12)
Jan 29 15:15:15 edison xdk-daemon[15937]: ps process exited with code 1
Jan 29 15:15:15 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:15 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:15 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:15 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:16 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:16 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:16 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:16 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:17 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:17 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:17 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:17 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:17 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:18 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:18 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:18 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:18 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:18 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:19 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:19 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:19 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:19 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:19 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:20 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Jan 29 15:15:20 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory

arfoll, 2016-02-04:
this just shows the xdk-daemon crashing and arduino doing it's `thing`. Have you tried running that without the XDK stuff?

sonofaforester, 2016-02-04:
So I removed the xdk-daemon, re-ran the test and had similar results. The difference being the log doesn't show the xdk stuff

-- Logs begin at Sat 2000-01-01 00:00:11 UTC, end at Thu 2016-02-04 21:26:03 UTC. --
Feb 04 21:25:52 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:53 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:53 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:53 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:53 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:53 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:54 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:54 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:54 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:54 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:54 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:55 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:55 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:55 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:55 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:55 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:56 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:56 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:56 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:56 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:57 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:57 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:57 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:57 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:57 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:58 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:58 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:58 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:58 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:58 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:59 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:59 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:59 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:59 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:25:59 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:00 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:00 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:00 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:00 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:00 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:01 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:01 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:01 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:01 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:02 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:02 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:02 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:02 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:02 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory
Feb 04 21:26:03 edison launcher.sh[182]: /opt/edison/launcher.sh: line 17: /dev/ttyGS0: No such file or directory

sonofaforester, 2016-02-19:
Any updates or more info that I can provide?

alext-mkrs, 2016-02-28:
I'm looking into this right now.

alext-mkrs, 2016-02-28:
Which expansion board are you using, @sonofaforester? You use `var x = new m.Gpio(32); //GPIO Pin 46` as GPIO initializer, but I don't see any pin 32 listed in the [pin muxing guide](http://www.emutexlabs.com/project/215-intel-edison-gpio-pin-multiplexing-guide)

sonofaforester, 2016-02-28:
It's the Sparkfun RaspiPi board.
https://www.sparkfun.com/products/13044

lrntplc, 2016-02-28:
FWIW, I'm having the same issue with both Nodejs 0.10 and Nodejs 0.12.7, on a Minibreakout board. The libmraa version is 0.9.2. My setup is a little different though. I am using 2 gpio pins (31 and 32) and I have 2 different ISRs, registered on falling edge. When the ISRs are called, I communicate over I2C with the boards generating the interrupts, in order to read the state, clear the interrupt, etc. After doing this for some time (I'd say, between 30-50 interrupts), I get the crash @sonofaforester  listed in the issue. The 'journalctl -f' shows:

Feb 28 20:43:57 edison libmraa[1788]: gpio: Failed to open edge for writing
Feb 28 20:43:57 edison libmraa[1788]: gpio: Failed to open edge for writing

Then, I tried to do everything in a timer, so that the ISR finishes as soon as possible. The node still crashes (after, approx, the same number of interrupts as in the previous case) but the only stack trace I get is:

PID 1870 received SIGSEGV for address: 0x0
/home/root/ecb/node_modules/segfault-handler/build/Release/segfault-handler.node(+0x1288)[0xb77a6288]
linux-gate.so.1(__kernel_rt_sigreturn+0x0)[0xb77a940c]

The 'journalctl -f' shows nothing in this case... :/

Apparently, this issue is similar to https://github.com/intel-iot-devkit/mraa/issues/208

alext-mkrs, 2016-02-28:
Yeah, generally what I'm observing is:

1. This is not related to SPI or anything specific going on in the ISR itself - it's just the fact that ISR is run.
1. This is only Node.js, C or Python are not affected
1. With simple `console.log()` as the ISR Node 0.10.38 crashes in ~1 minute, Node 0.12.7 in ~3 minutes and Node 4.3.0 didn't crash in ~15 minutes. Adding a fix for #387, which I've found along the road causes delays 0.12.7 crash to ~10 minutes.
1. If the ISR function is empty, the segfault doesn't happen even on node 0.10.38

And yes, it looks similar to #208.

I'm looking further into this.

alext-mkrs, 2016-02-29:
Just as a working note - here's how it looks in the gdb.
```
Program received signal SIGSEGV, Segmentation fault.
v8::Function::Call (this=0x88212d8, recv=..., argc=1, argv=0xbfffc4e0) at ../deps/v8/src/api.cc:4008
4008	  ON_BAILOUT(isolate, "v8::Function::Call()", return Local<v8::Value>());
(gdb) bt
#0  v8::Function::Call (this=0x88212d8, recv=..., argc=1, argv=0xbfffc4e0) at ../deps/v8/src/api.cc:4008
#1  0xb7561d42 in mraa::Gpio::v8isr (req=0xb6900468, status=0) at /home/root/mraa/api/mraa/gpio.hpp:147
#2  0x08557cb6 in uv__work_done (handle=0x87f86e0 <default_loop_struct+96>) at ../deps/uv/src/threadpool.c:236
#3  0x085597d6 in uv__async_event (loop=0x87f8680 <default_loop_struct>, w=0x87f876c <default_loop_struct+236>, nevents=1) at ../deps/uv/src/unix/async.c:92
#4  0x08559885 in uv__async_io (loop=0x87f8680 <default_loop_struct>, w=0x87f8770 <default_loop_struct+240>, events=1) at ../deps/uv/src/unix/async.c:132
#5  0x08566273 in uv__io_poll (loop=0x87f8680 <default_loop_struct>, timeout=1577) at ../deps/uv/src/unix/linux-core.c:345
#6  0x0855a132 in uv_run (loop=0x87f8680 <default_loop_struct>, mode=UV_RUN_ONCE) at ../deps/uv/src/unix/core.c:328
#7  0x084ee3bf in node::Start (argc=2, argv=<optimized out>) at ../src/node.cc:3722
#8  0x0817295b in main (argc=2, argv=0xbffffcf4) at ../src/node_main.cc:65
(gdb) thread apply all backtrace

Thread 1 (LWP 10640):
#0  v8::Function::Call (this=0x88212d8, recv=..., argc=1, argv=0xbfffc4e0) at ../deps/v8/src/api.cc:4008
#1  0xb7561d42 in mraa::Gpio::v8isr (req=0xb6900468, status=0) at /home/root/mraa/api/mraa/gpio.hpp:147
#2  0x08557cb6 in uv__work_done (handle=0x87f86e0 <default_loop_struct+96>) at ../deps/uv/src/threadpool.c:236
#3  0x085597d6 in uv__async_event (loop=0x87f8680 <default_loop_struct>, w=0x87f876c <default_loop_struct+236>, nevents=1) at ../deps/uv/src/unix/async.c:92
#4  0x08559885 in uv__async_io (loop=0x87f8680 <default_loop_struct>, w=0x87f8770 <default_loop_struct+240>, events=1) at ../deps/uv/src/unix/async.c:132
#5  0x08566273 in uv__io_poll (loop=0x87f8680 <default_loop_struct>, timeout=1577) at ../deps/uv/src/unix/linux-core.c:345
#6  0x0855a132 in uv_run (loop=0x87f8680 <default_loop_struct>, mode=UV_RUN_ONCE) at ../deps/uv/src/unix/core.c:328
#7  0x084ee3bf in node::Start (argc=2, argv=<optimized out>) at ../src/node.cc:3722
#8  0x0817295b in main (argc=2, argv=0xbffffcf4) at ../src/node_main.cc:65
```

Looks like at some point our
```
v8::Local<v8::Function> f = v8::Local<v8::Function>::New(v8::Isolate::GetCurrent(), This->m_v8isr);
```
does not succeed and we are calling something empty or undefined.

alext-mkrs, 2016-03-13:
JFYI, I'm still looking into this, Haven't bee able to spend too much quality time, but I'm progressing. 

Looks like it's even deeper than that and it currently looks like our GPIO context gets destroyed somewhere in the process at some point, which ends our `mraa_gpio_interrupt_handler()` loop (at least that's what I observe on Node 4.3.0 I was testing with today). It may be the same on 0.12.7, though I had no time to compare today.

Stay tuned :smiley:

alext-mkrs, 2016-03-20:
At the end of the day looks like the reason is more or less trivial. What we're seeing here is that our GPIO context is getting destroyed, just like I wrote previously. And it's the same on 0.12.7 and 4.3.0, just pops up at sligthly different stages of the mraa GPIO interrupt handing chain.

The actual reason is that in the example script you're using as a base for your program (and I used as a reproducer), we are creating a GPIO context and then just start a neverending `setInterval()`. But as long as that function doesn't really use our GPIO context object directly - Node decides it's time garbage-collect it at some point, wreaking havoc.

So the immediate solution is to add something that refers to our GPIO context object to `setInterval()`, e.g.
```javascript
var m = require('mraa');

console.log(m.getVersion());

var x = new m.Gpio(2);

function a() {
  console.log("trigger!");
}
 
x.dir(m.DIR_IN);
x.isr(m.EDGE_BOTH, a);

setInterval(function(){
  console.log("keep alive");
  if (!x) { console.log("our GPIO context got destroyed"); }
}, 10000);
```

It runs rock-solid for me both on 0.12.7 and 4.3.0 (mraa 0.9.6) with an Arduino 101 board toggling GPIO every 100ms (as in your original post) as an interrupt generator. Without the fix it bails out after ~7 minutes.

@sonofaforester, please give it a try and if you don't see any other problems in your test, I'll submit a patch for that example.

I'll also think if there's anything we can do on the library level to prevent this from happening, as generally speaking we *are* using that object, so why Node thinks otherwise I'm not sure. @arfoll (or anyone else), any ideas are welcome.

sonofaforester, 2016-03-23:
Thanks for working on this! I'm sorry it was sort of a red herring.

alext-mkrs, 2016-03-28:
Oh, no worries, it was a fun ride, and it's just that I didn't have enough time to finish it in one seating :smiley:

So I take this as it works fine for you, right?

commit 00a0abd0 (2016-04-11): isr.js: updated JS ISR example to avoid GPIO context being destroyed

Closes #403.
closed 2016-04-11 by arfoll