eclipse-mraa/mraa #406: Error to run NODE.JS application in intel edison

neuberfran, 2016-01-18:
Hi guys,

I put ubilinux in my intel edison and I blink led with c++ and python.

Now I have problems to blink led using node.js:

root@neuberfran:/usr/share/mraa/examples/javascript# node -v
v5.4.0
root@neuberfran:/usr/share/mraa/examples/javascript# npm -v
3.3.12
root@neuberfran:/usr/share/mraa/examples/javascript# node Blink-IO.js
module.js:423
return process.dlopen(module, path._makeLong(filename));
^

Error: Module did not self-register.
at Error (native)
at Object.Module._extensions..node (module.js:423:18)
at Module.load (module.js:344:32)
at Function.Module._load (module.js:301:12)
at Module.require (module.js:354:17)
at require (internal/module.js:12:17)
at Object.<anonymous> (/usr/share/mraa/examples/javascript/Blink-IO.js:25:9)
at Module._compile (module.js:398:26)
at Object.Module._extensions..js (module.js:405:10)
at Module.load (module.js:344:32)

To install node.js I’m using this tutorial

blog.wia.io/installing-node-js-v4-0-0-on-a-raspberry-pi/

But I changed nodejs.org/dist/v4.0.0/node-v4.0.0-linux-armv6l.tar.gz
To nodejs.org/dist/v4.0.0/node-v4.0.0-linux-x86.tar.gz

Because intel Edison is a x86 32 bits

https://www.youtube.com/watch?v=z3O26CC9nAc

I'm trying transport this tutorial (above) to intel edison. The first part: http://localhost:3000 worked. Than, If I try hello world here, will worked

About this site: http://stackoverflow.com/questions/28486891/uncaught-error-module-did-not-self-register/30233557 
I tried every things:
npm rebuild
rm -rf node_modules
npm i

But don't solved

http://www.emutexlabs.com/forum/general-questions/96-error-to-run-node-js-application-in-intel-edison#162

arfoll, 2016-01-18:
you'll have to compile nodejs from source, don't use a prebuild binary. Get the source tarball and rebuild. Then do an npm install mraa and clear the existing stuff from node_modules.

neuberfran, 2016-01-18:
Please I need more detail step by step I have to do even
If possible with commands

arfoll, 2016-01-18:
You should be able to do something like this:

```
wget https://nodejs.org/dist/v4.2.4/node-v4.2.4.tar.gz
tar xf node-v4.2.4.tar.gz
cd node-v4.2.4/
./configure --prefix=/usr
make
make install
```

And then:
```
npm install mraa
```

neuberfran, 2016-01-18:
I think the correct is node-v4.2.4-linux-x86.tar.gz 

The instalation interrupetd here in make command

[image]

arfoll, 2016-01-18:
Are you sure you didn't just run out of disk space? I see no error there,
please send text not screenshots

neuberfran, 2016-01-19:
root@neuberfran:~# cat /proc/cpuinfo | grep -E '^(b|pr|v|mo|$)'; uname -a; echo;                                                                                                                      free; echo; df; echo
processor       : 0
vendor_id       : GenuineIntel
model           : 74
model name      : Genuine Intel(R) CPU   4000  @  500MHz
bogomips        : 998.40

processor       : 1
vendor_id       : GenuineIntel
model           : 74
model name      : Genuine Intel(R) CPU   4000  @  500MHz
bogomips        : 998.40

Linux neuberfran.dyndns.ws 3.10.17-yocto-standard-r2 #7 SMP PREEMPT Thu Feb 26 0                                                                                                                     9:57:06 UTC 2015 i686 GNU/Linux

             total       used       free     shared    buffers     cached
Mem:        983516      83348     900168          0      14780      33260
-/+ buffers/cache:      35308     948208
Swap:            0          0          0

Filesystem      1K-blocks    Used Available Use% Mounted on
rootfs            1441648 1172656    173968  88% /
/dev/root         1441648 1172656    173968  88% /
devtmpfs           491448       0    491448   0% /dev
tmpfs               98352     300     98052   1% /run
tmpfs                5120       0      5120   0% /run/lock
tmpfs              196700       0    196700   0% /run/shm
tmpfs              491756       0    491756   0% /tmp
/dev/mmcblk0p7      32686    5352     27334  17% /boot
/dev/mmcblk0p10   1337936    2060   1319492   1% /home

neuberfran, 2016-01-19:
[image]

arfoll, 2016-01-19:
I still don't see an error. Please stop with the screeenshots, paste logs in dpaste.com or a similar pastebin!

neuberfran, 2016-01-19:
I downloaded wget https://nodejs.org/dist/v4.2.4/node-v4.2.4-linux-x86.tar.gz and before tar I command cd node-v4.2.4-linux-x86
But I can't do command ./configure --prefix=/usr Because ./configure does not exist in this package

arfoll, 2016-01-19:
You where wrong I gave the correct tarball. You're downlading a pre compiled tarball. ```https://nodejs.org/dist/v4.2.4/node-v4.2.4.tar.gz```

neuberfran, 2016-01-19:
Ok. I'm trying this now and mistakes(fix) are repeating
where I find the logs files to put in pastebin ?

neuberfran, 2016-01-19:
below I will put the link for the end of the installation (make) and after (make install) and node -v and npm -v

http://dpaste.com/3BP8N0K

arfoll, 2016-01-19:
Looks like I made a mistake, you need gcc 4.8 and above for node 4.x so it won't build on edison. My bad. You may have to downgrade to 0.12.x

alext-mkrs, 2016-01-19:
I have just tried - on Yocto image it compiles from sources just fine. Took about 1-1.5 hrs with "make -j2", but works nicely:

```bash
root@edison1:~/nodejs/node-v4.2.4# ./node -e "console.log('Hello from node.js ' + process.version)"
Hello from node.js v4.2.4
```
The error @neuberfran has looks like an incorrect gcc version indeed - however they should have added a check for that, so that's a bit strange. One more reason not to use ubilinux on Edison to me :smiley: with all due respect to Emutex folks.

@neuberfan, check what gcc you've got, on Yocto it's
```bash
root@edison1:~/nodejs/node-v4.2.4# gcc --version
gcc (GCC) 4.9.1
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@edison1:~/nodejs/node-v4.2.4#
```

neuberfran, 2016-01-19:
root@neuberfran:~# gcc --version
gcc (Debian 4.7.2-5) 4.7.2
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

arfoll, 2016-01-19:
So you need to update your edison image :)

Btw @alext do you think you could push node 4.x to your repo instead of
node 0.12.x?

alext-mkrs, 2016-01-19:
I've been thinking about this for a while now, the biggest problem is that there are some node.js apps in the image (like xdk-daemon and somesuch), which sometimes are sensitive to Node.js versions, so I'd need to find all of them and somehow test.

But in general I'd like to do that, yes. Especially if we make 4.x the only officially supported branch for mraa.

neuberfran, 2016-01-19:
I'm not using edison image (yocto) in this intel edison. I'm using ubilinux. Because I need debian here also

neuberfran, 2016-01-20:
I need help to upgrade gcc to 4.9

root@neuberfran:~/node-v4.2.4# cat /proc/version
Linux version 3.10.17-yocto-standard-r2 (root@p32mac) (gcc version 4.8.2 (GCC) ) #7 SMP PREEMPT Thu Feb 26 09:57:06 UTC 2015

root@neuberfran:~/node-v4.2.4# gcc --version
gcc (Debian 4.7.2-5) 4.7.2
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

arfoll, 2016-01-20:
Sorry but no one on here can help you easily upgrade your image to use gcc 4.9, it's not an easy task to do. I'd recommend not using ubilinux because we don't officially support it because it uses a mismatch of components leading to unreliability and isn't fully recompiled from source.

I would recommend rethinking why you want/need debian, but this is a debate best left to the community forum, not for here.

closed 2016-01-20 by arfoll

neuberfran, 2016-01-20:
Don't close issue

arfoll, 2016-01-20:
Sorry but this is not the right place for generic upgrade help to gcc 4.9 or really to help with node compilation in general :) Since we've ascertained that there is no issue with mraa I consider this 'closed'.

neuberfran, 2016-01-21:
I'll be waiting for the release ubilinux jessie  scheduled for January this year