eclipse-mraa/mraa #407: Error when build ipk on edison.

flymaxty, 2016-01-19:
Hi all,
I'm trying to build ipk on edison, but got error on below,
```bash
[ 96%] Built target Pwm3-cycle
[ 97%] Built target Spi-pot
[ 98%] Built target Uart
[100%] Built target blink-io-cpp
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: mraa
CPack: - Install project: mraa
CPack: Create package
Result '1'
Output ''
Error  'find: unrecognized: -uid
BusyBox v1.22.1 (2015-06-19 07:36:41 CEST) multi-call binary.

Usage: find [-HL] [PATH]... [OPTIONS] [ACTIONS]

tar: unrecognized option '--owner=0'
BusyBox v1.22.1 (2015-06-19 07:36:41 CEST) multi-call binary.

Usage: tar -[cxtZzJjahmvO] [-X FILE] [-T FILE] [-f TARFILE] [-C DIR] [FILE]...

'
Makefile:103: recipe for target 'package' failed
make: *** [package] Segmentation fault

```

Is there anything I forgot to install ?

arfoll, 2016-01-19:
You can't build the ipk using busybox, you'd need find from coreutils. To be honest building ipks that way is not really the 'correct' way you should be using Yocto/Bitbake.

But why are you doing this instead of just using `make install`? Having an ipk doesn't provide you with much unless you're distributing it and if you're doing that you really should be looking into bitbake.

alext-mkrs, 2016-01-19:
And if you *really* that keen on building anipk on the board, start with installing tar and find (coreutils) from my repo (or the default IoTDevKit's one, I'm not sure if it has those packages though).

And then if and when you face other problems, repeat - but by that time you probably would want to follow @arfoll's advice and go the Yocto way :smiley:

arfoll, 2016-01-19:
After tar and find you'll also want opkg-utils though might be on ur board
already

alext-mkrs, 2016-01-19:
I think we've established that this way of using our build scripts is not a recommended one and anyway provided instructions on using it this way - so I'm closing this one.

closed 2016-01-19 by alext-mkrs

flymaxty, 2016-01-20:
@arfoll @alext-mkrs Thanks for your reply!
Yes, I'm trying to distribute a specific version, but have less experience on yocto/bitbake, that's why I'm trying to build package on the board.
I will try them(use bitbake and coreutils) both in next few days.
Thanks for your help again!

alext-mkrs, 2016-01-20:
No problem

flymaxty, 2016-01-22:
@arfoll @alext-mkrs ,sorry to disturb you again. I have some problem when I use bitbake to compile mraa package.
The compile is ok, and get 4 packages(mraa,mraa-dev,mraa-dbg,mraa-doc).
and I remove the offical mraa (use `opkg remove mraa`) on my board, and installed mraa-_0.9.0-r0_core2-32.ipk(has no `include` files), but when I try to install others, I got errors:
```bash
root@edison:~/libs# opkg install mraa-dev_0.9.0-r0_core2-32.ipk
Multiple replacers for mraa-dev, using first one (mraa).
Multiple replacers for mraa-dev, using first one (mraa).
Multiple replacers for mraa-dev, using first one (mraa).
Multiple replacers for mraa-dev, using first one (mraa).
Multiple replacers for mraa-dev, using first one (mraa).
```

And I found that offical mraa ipk has all things in one package, but I got 4 packages frome bitbake, is there something I miss?
the bb file comes from here:
http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-iot-middleware/tree/recipes-devtools/mraa?h=dizzy
I repleace `src` section to use my repo.

Thanks again!  :)

arfoll, 2016-01-22:
This is normal, you get 4 pkgs which is better, the pkgs we generate on the repository contain the headers (mraa-dev), the binary libs (mraa) and I think the docs (mraa-doc). We don't even ship the debug symbols but you can install them with (mraa-dbg).

My guess is that your opkg DB has gotten a little confused since mraa (from repo) 'replaces' mraa-dev since it includes it. Make sure the package is really gone from opkg with ```opkg list | grep mraa```