eclipse-mraa/mraa #408: gcc compile error when building Node.JS binding on Fedora 22

whbruce, 2016-01-19:
I'm getting a gcc error when building Node.JS binding on Fedora 22
cmake completes successfully and relevant output is as follows:
```
INFO - libmraa Version v0.9.0-18-g7e30224
INFO - Target arch is x86_64
-- The CXX compiler identification is GNU 5.1.1
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found SWIG: /usr/lib64/ccache/swig (found version "3.0.7") 
-- Found Node: /usr/bin/node  
INFO - Node version is 0.10.36
INFO - Node using v8 3.14.5.10
DEBUG - Padded V8 version to match SWIG format
```

But when I build I see the following error
```
cd /home/hbruce/Projects/mraa/build/fc-js/src/javascript && /usr/lib64/ccache/c++   -Dmraajs_EXPORTS -g -fPIC -I/home
/hbruce/Projects/mraa/api -I/home/hbruce/Projects/mraa/api/mraa -I/home/hbruce/Projects/mraa/include -I/usr/include/s
rc -I/usr/include/node -I/usr/include/deps/v8/include -I/usr/include/deps/uv/include -I/home/hbruce/Projects/mraa/src
/javascript/..     -DBUILDING_NODE_EXTENSION -DSWIGJAVASCRIPT=TRUE -o CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx
.o -c /home/hbruce/Projects/mraa/build/fc-js/src/javascript/mraajsJAVASCRIPT_wrap.cxx

In file included from /home/hbruce/Projects/mraa/build/fc-js/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:0:
/usr/include/node/node.h:150:46: error: ‘uv_statbuf_t’ does not name a type
 v8::Local<v8::Object> BuildStatsObject(const uv_statbuf_t* s);
                                              ^
In file included from /home/hbruce/Projects/mraa/build/fc-js/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:0:
/usr/include/node/node.h:252:16: error: variable or field ‘SetErrno’ declared void
 NODE_EXTERN void SetErrno(uv_err_t err);
                ^
/usr/include/node/node.h:252:16: error: ‘uv_err_t’ was not declared in this scope
```

/usr/include/deps does not exist. I'm not sure if that is to be expected.

Some googling turned up redhat bug [nodejs-devel should depend on compat-libuv010-devel](https://bugzilla.redhat.com/show_bug.cgi?id=1215729), but I have already applied the package updates and it has not fixed the issue. 

Any pointers would be appreciated. 
 
Henry

arfoll, 2016-01-19:
What versions of node-gyp and nodejs are you running? Can you paste the info from `rpm -qi` or similar?

alext-mkrs, 2016-01-19:
I have fedora 22 as my mraa dev system for some time and this piece got broken at some point. Tbh I don't recall all of the details of my investigation as it was a while ago, but I believe what I've found out was that the compat- library installed headers into a place where our build process doesn't see them and what gets picked is a standard set from a normal libuv package, which is too new for Fedora's node.

whbruce, 2016-01-19:
@alext-mkrs, great clue on compat-libuv010. I looked at where include files were installed (/usr/include/compat-libuv010) and added path before {NODE_INCLUDE_DIRS} and build worked.

I'll work on cleaner fix to cmake, do some testing and issue a PR.

alext-mkrs, 2016-01-19:
Great, glad to help. I've never got to getting it fixed, just switched to compiling mraa with cross toolchain only :smiley: (prior to that I was running host builds as well). It's very nice that finally it gets sorted out.

arfoll, 2016-01-19:
> compiling mraa with cross toolchain only [image: :smiley:] (prior to that

Just adding this to the reasons I don't like fedora :)

Bar overriding this NODE_INCLUDE_DIRS var I don't see what we can do to
stop distros that ship a node.js compiled with another version of libuv...
imho that's just broken distro pkgs.

alext-mkrs, 2016-02-06:
This was implemented by PR #414, closing this one.

closed 2016-02-06 by alext-mkrs