eclipse-mraa/mraa #411: Unable to build from source on Edison using Node 4.2.3

deadprogram, 2016-01-21:
Pulled latest mraa from git.

The cmake file appears to run OK:

```
root@cylon13:~/mraa/build# cmake .. -DBUILDSWIGJAVA=OFF -DBUILDSWIGPYTHON=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr
INFO - libmraa Version v0.9.0-27-g8b68f30
INFO - Target arch is i586
INFO - Adding support for platform 
INFO - Adding support for all platforms
INFO - Adding x86 platforms
INFO - Node version is 4.2.3
INFO - Node using v8 4.5.103.35

INFO - Using V8 version > 3 so requiring C++11 compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/mraa/build
```

However, running `make` results in the following errors:

```
root@cylon13:~/mraa/build# make
[ 28%] Built target mraa
[ 30%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
In file included from /usr/local/include/node/node.h:42:0,
                 from /home/root/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:156:
/usr/local/include/node/v8.h:336:1: error: expected unqualified-id before 'using'
 using Handle = Local<T>;
 ^
/usr/local/include/node/v8.h: In constructor 'v8::MaybeLocal<T>::MaybeLocal()':
/usr/local/include/node/v8.h:353:33: error: 'nullptr' was not declared in this scope
   V8_INLINE MaybeLocal() : val_(nullptr) {}
                                 ^
/usr/local/include/node/v8.h: In member function 'bool v8::MaybeLocal<T>::IsEmpty() const':
/usr/local/include/node/v8.h:360:51: error: 'nullptr' was not declared in this scope
   V8_INLINE bool IsEmpty() const { return val_ == nullptr; }
                                                   ^
/usr/local/include/node/v8.h: In member function 'bool v8::MaybeLocal<T>::ToLocal(v8::Local<S>*) const':
/usr/local/include/node/v8.h:364:29: error: 'nullptr' was not declared in this scope
     out->val_ = IsEmpty() ? nullptr : this->val_;
                             ^
/usr/local/include/node/v8.h: In member function 'bool v8::WeakCallbackInfo<T>::IsFirstPass() const':
/usr/local/include/node/v8.h:430:50: error: 'nullptr' was not declared in this scope
   bool IsFirstPass() const { return callback_ != nullptr; }
```

Here is a gist with the full output:

https://gist.github.com/deadprogram/6c5ce16ce3702fa734b7

Thanks!

arfoll, 2016-01-22:
That's really weird, because I just compiled that exact version of node from git and then built mraa and it works... Can you check /usr/include/node/v8-version.h and make sure that the V8 version matches what cmake found?

xbolshe, 2016-01-23:
It is strange, but I have the same issue:

DEBUG: Executing shell function do_compile
-- The C compiler identification is GNU 4.9.1
-- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc
-- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
INFO - libmraa Version v0.9.0-29-g078218f
INFO - Target arch is i586
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.8", minimum required is "2.7") 
INFO - Package arch is i586
INFO - Adding support for platform 
INFO - Adding support for all platforms
INFO - Adding x86 platforms
-- The CXX compiler identification is GNU 4.9.1
-- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++
-- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found SWIG: /usr/local/bin/swig (found version "3.0.8") 
-- Found PythonLibs: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/python2.7 (found suitable version "2.7.3", minimum required is "2.7") 
-- Found Node: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/node  
INFO - Node version is 5.4.1
INFO - Node using v8 4.6.85.31

DEBUG - Padded V8 version to match SWIG format
INFO - Using V8 version > 3 so requiring C++11 compiler
-- Configuring done
WARNING: Target "_python-mraa" requests linking to directory "/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/python2.7".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.9.0+gitAUTOINC+078218f680-r0/git/build
Scanning dependencies of target mraa
[  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
[  2%] Building C object src/CMakeFiles/mraa.dir/gpio/gpio.c.o
[  3%] Building C object src/CMakeFiles/mraa.dir/i2c/i2c.c.o
[  5%] Building C object src/CMakeFiles/mraa.dir/pwm/pwm.c.o
[  6%] Building C object src/CMakeFiles/mraa.dir/spi/spi.c.o
[  7%] Building C object src/CMakeFiles/mraa.dir/aio/aio.c.o
[  8%] Building C object src/CMakeFiles/mraa.dir/uart/uart.c.o
[ 10%] Building C object src/CMakeFiles/mraa.dir/iio/iio.c.o
[ 11%] Building C object src/CMakeFiles/mraa.dir/x86/x86.c.o
[ 12%] Building C object src/CMakeFiles/mraa.dir/x86/intel_galileo_rev_d.c.o
[ 14%] Building C object src/CMakeFiles/mraa.dir/x86/intel_galileo_rev_g.c.o
[ 15%] Building C object src/CMakeFiles/mraa.dir/x86/intel_edison_fab_c.c.o
[ 16%] Building C object src/CMakeFiles/mraa.dir/x86/intel_de3815.c.o
[ 17%] Building C object src/CMakeFiles/mraa.dir/x86/intel_nuc5.c.o
[ 19%] Building C object src/CMakeFiles/mraa.dir/x86/intel_minnow_byt_compatible.c.o
[ 20%] Building C object src/CMakeFiles/mraa.dir/x86/intel_sofia_3gr.c.o
[ 21%] Building C object src/CMakeFiles/mraa.dir/version.c.o
Linking C shared library libmraa.so
[ 21%] Built target mraa
[ 23%] Swig source
Scanning dependencies of target _python-mraa
[ 24%] Building CXX object src/python/CMakeFiles/_python-mraa.dir/mraaPYTHON_wrap.cxx.o
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.9.0+gitAUTOINC+078218f680-r0/git/build/src/python/mraaPYTHON_wrap.cxx:11:0: warning: "SWIGPYTHON" redefined
 #define SWIGPYTHON
 ^
<command-line>:0:0: note: this is the location of the previous definition
[ 25%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/mraa.c.o
[ 26%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/gpio/gpio.c.o
[... 15 lines cut ...]
Linking CXX shared module _mraa.so
[ 46%] Built target _python-mraa
[ 47%] Swig source
Scanning dependencies of target mraajs
[ 48%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
In file included from /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/node.h:42:0,
                 from /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.9.0+gitAUTOINC+078218f680-r0/git/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:169:
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h:336:1: error: expected unqualified-id before 'using'
 using Handle = Local<T>;
 ^
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h: In constructor 'v8::MaybeLocal<T>::MaybeLocal()':
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h:353:33: error: 'nullptr' was not declared in this scope
   V8_INLINE MaybeLocal() : val_(nullptr) {}
                                 ^
[... cut ...]
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h:364:29: error: 'nullptr' was not declared in this scope
     out->val_ = IsEmpty() ? nullptr : this->val_;
                             ^
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h: In member function 'bool v8::WeakCallbackInfo<T>::IsFirstPass() const':
/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/include/node/v8.h:430:50: error: 'nullptr' was not declared in this scope
   bool IsFirstPass() const { return callback_ != nullptr; }

xbolshe, 2016-01-23:
The C++11 flag is dropped somewhere. There is no C++11 flag in mraajs.dir/flags.make

xbolshe, 2016-01-23:
It seems the issue is related with an old CMAKE version 2.8.12 in my case where 
```
set_property (TARGET mraajs PROPERTY CXX_STANDARD 11)
```
does not work.

xbolshe, 2016-01-23:
My proposal to fix: https://github.com/intel-iot-devkit/mraa/pull/416

It looks like when CMAKE version is less than 3.1:

```
| DEBUG: Executing shell function do_compile
| -- The C compiler identification is GNU 4.9.1
| -- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc
| -- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Looking for include file pthread.h
| -- Looking for include file pthread.h - found
| -- Looking for pthread_create
| -- Looking for pthread_create - not found
| -- Looking for pthread_create in pthreads
| -- Looking for pthread_create in pthreads - not found
| -- Looking for pthread_create in pthread
| -- Looking for pthread_create in pthread - found
| -- Found Threads: TRUE
| INFO - libmraa Version v0.9.0-29-g078218f
| INFO - cmake Version 2.8.12
| INFO - Target arch is i586
| -- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.8", minimum required is "2.7")
| INFO - Package arch is i586
| INFO - Adding support for platform
| INFO - Adding support for all platforms
| INFO - Adding x86 platforms
| -- The CXX compiler identification is GNU 4.9.1
| -- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++
| -- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Found SWIG: /usr/local/bin/swig (found version "3.0.8")
| -- Found PythonLibs: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/python2.7 (found suitable version "2.7.3", minimum required is "2.7")
| -- Found Node: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/node
| INFO - Node version is 5.4.1
| INFO - Node using v8 4.6.85.31
| 
| DEBUG - Padded V8 version to match SWIG format
| INFO - Using V8 version > 3 so requiring C++11 compiler
| CMake Error at src/javascript/CMakeLists.txt:41 (message):
|    FATAL ERROR: Need to use CMAKE version 3.1+, but it is 2.8.12
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.9.0+gitAUTOINC+078218f680-r0/git/build/CMakeFiles/CMakeOutput.log".
| See also "/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.9.0+gitAUTOINC+078218f680-r0/git/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
```

arfoll, 2016-01-24:
Oh interesting, great catch, makes sense I had an updated cmake on my board.

Rather than fail is rather set cflags in a dirty way if we're on an older
cmake, that way we won't break on Edison for the near future.

xbolshe, 2016-01-24:
I have made an update: change fail to set flags.

commit 2e5e3eae (2016-01-24): mraajs: Check for CMAKE version > 3.1 or use workaround

Older cmake versions don't support the CXX_STANDARD property. Remove doubled
CMAKE_CXX_FLAGS in mraajs COMPILE_FLAGS

Closes #411
closed 2016-01-24 by arfoll