eclipse-mraa/mraa #417: mraa_uart_write() - Only string objects? Send values?

gregorykelleher, 2016-01-24:
Hi, I have a question. I want to be able to send hex values as binary data over UART using the MRAA library. 

In the function documentation I found: **mraa_uart_write()** which takes the parameters: **mraa_uart_context, const char *, size_t**

However, I don't want to send a string object; I want to send a hex value like I would with Arduino using **Serial.write(0x7c)** for example.

How can I write to the serial port without sending a const char *?

thank you

arfoll, 2016-01-24:
Isn't a hex value like this essentially just a char? I mean yes there's no
call to send a single char down that way but surely sending a pointer isn't
much different? On the c++ interface but I'm not convinced of the
usefulness.

gregorykelleher, 2016-01-24:
Sorry, bit of novice migrating from Arduino to C++. 

I should be using **mraa.hpp**, so in that case it's **int write (const char *data, int length)** which allows me to write bytes in a String object to a device.

But how might I send a hex value as a single byte, as I'm familiar doing in Arduino as **Serial.write(0x12)**?

arfoll, 2016-01-25:
No problem, that's fair enough, so it works the same in C++ & C really, you can't really send anything but char's on a uart, 0x12 or 0x7c is a 'char'.

See the example in examples/uart.c, but essentially to send a single byte you need to do:

```
#include "mraa.h"

int
main(int argc, char** argv)
{
    mraa_uart_context uart = mraa_uart_init(0);

    char x = 0x12;                                                                                                                                                                                                                                                                                        
    mraa_uart_write(uart, &x, 1);
}
```

So essentially you pass a pointer with one char to mraa_uart_write. You can't write less than a char on a uart, that's just the way it works, since the kernel treats it as a char device.

Hope that helps!

gregorykelleher, 2016-01-25:
Cheers, that worked a treat. Simple question after all. Thanks again!

closed 2016-01-25 by arfoll