eclipse-mraa/mraa #419: FTDI Interrupts do not work with Java and most likely Python

petreeftime, 2016-02-02:
At https://github.com/intel-iot-devkit/mraa/blob/master/src/gpio/gpio.c#L272 the FTDI GPIO interrupt handler is called, but setup for Java is called a few lines after this: https://github.com/intel-iot-devkit/mraa/blob/master/src/gpio/gpio.c#L295, additionally, Python is handling its' ISRs here: https://github.com/intel-iot-devkit/mraa/blob/master/src/gpio/gpio.c#L321, code which is again not called by the FTDI interrupt function handler.

The code should be refactored such that bindings can work with FTDI. Perhaps each language could expose a per-thread setup, isr call and teardown function if they require it.

arfoll, 2016-02-02:
@whbruce this is a bit of a problem, didn't realise this didn't work from SWIG :/ How can we fix this?

whbruce, 2016-02-02:
Sorry didn't test with SWIG bindings. As per @petreeftime's comments language bindings are not correctly handled and refactoring is required. I think I can fix this by making gpio interrupt adv_funcs more granular so ft4222 thread is handled the same way as gpio "poll" thread.  I'll get to this right away.

alext-mkrs, 2016-02-13:
Fixed with PR #420, closing.

closed 2016-02-13 by alext-mkrs