eclipse-mraa/mraa #426: Failed at the mraa@0.9.1 install script.

rahul2704, 2016-02-12:
hey there i keep getting this error when installing mraa
I want to use blynk library with mraa still installed
im using this `npm install -g mraa blynk-library`
i contacted blynk devs, they say the problem is with mraa lib. i can use blynk lib without mraa but i want both installed

arfoll, 2016-02-12:
What version of node?

rahul2704, 2016-02-12:
on the board?? or mac?
v5.3.0

rahul2704, 2016-02-12:
on edison running `node --version` gives
v0.10.38

arfoll, 2016-02-12:
where are you trying to install mraa? Because it won't work on your mac

arfoll, 2016-02-12:
mraa 0.9.2 builds for me on 0.10.38 and 5.40. You're going to have to be more specific/provide logs or explain why the blynk guys think it's my fault ;-)

rahul2704, 2016-02-12:
i have ssh into edison
im installing into edison after running /dev/cu.usbserial-XXX.. etc
in fact i use bloop to screen directly into edison
i can install blynk only, which erases mraa that came in with yocto firmware in devkit then i can't use example projects in xdk iot as it says mraa missing

rahul2704, 2016-02-12:
`yp ERR! cwd /usr/lib/node_modules/mraa
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 

npm ERR! mraa@0.9.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@0.9.1 install script.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.10.17-poky-edison+
npm ERR! command "node" "/usr/bin/npm" "install" "-g" "mraa" "blynk-library"
npm ERR! cwd /home/root
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0
root@Edison_R:~#

rahul2704, 2016-02-12:
i am new to node.js, i am sorry! edison interface is not easy at all, especially with linux!! is there any user friendly way of installing packages without such hassle?

arfoll, 2016-02-12:
ok so to start with that's not a mraa issue ;-) 

Update mraa using opkg (the xdk should even have examples how this works). The readme.md at front pretty much explains how to do that very easily and the XDK should provide a way to do it as well.

rahul2704, 2016-02-12:
oh you mean the echo... command??

arfoll, 2016-02-12:
Yes

rahul2704, 2016-02-12:
```
root@edison:~# echo "src mraa-upm http://iotdk.intel.com/repos/2.0/intelgalactic" > /etc/opkg/mraa-upm.conf
root@edison:~# opkg update
Downloading http://iotdk.intel.com/repos/2.0/intelgalactic/Packages.
Updated list of available packages in /var/lib/opkg/iotkit.
Downloading http://iotdk.intel.com/repos/2.0/intelgalactic/Packages.
Updated list of available packages in /var/lib/opkg/mraa-upm.
root@edison:~# opkg install mraa
Upgrading mraa from 0.8.0 to 0.9.0 on root.
Downloading http://iotdk.intel.com/repos/2.0/intelgalactic/mraa_0.9.0_i586.ipk.
Removing obsolete file /usr/lib/libmraa.so.0.8.0.
Removing obsolete file /usr/share/mraa/examples/python/uart.py.
Configuring mraa.
```
done?
now i can run npm install -g mraa blynk-library?

arfoll, 2016-02-12:
remove mraa from that npm line and yes.

rahul2704, 2016-02-12:
won't this erase existing mraa?

arfoll, 2016-02-12:
It shouldn't since blynk-js doesn't depend on mraa so npm won't go and
fetch it.

rahul2704, 2016-02-12:
```
root@edison:~# npm install -g blynk-library
/usr/bin/blynk-ctrl.js -> /usr/lib/node_modules/blynk-library/bin/blynk-ctrl.js
/usr/bin/blynk.js -> /usr/lib/node_modules/blynk-library/bin/blynk.js
blynk-library@0.0.28 /usr/lib/node_modules/blynk-library
```

how to I check which libraries are in the edison? any software providing GUI? maybe cyberduck?
thanks for your patience :)

arfoll, 2016-02-12:
easiest way is to call mraa.getVersion() from node.js.

rahul2704, 2016-02-12:
ERROR:     throw err;                                                                                                                                 
  ERROR:         ^                                                                                                                                    
ERROR: Error: Cannot find module 'mraa'                                                                                                               
    at Function.Module._resolveFilename (module.js:338:15)                                                                                            
    at Function.Module._load (module.js:280:25)                                                                                                       
    at Module.require (module.js:364:17)                                                                                                              
    at require (module.js:380:17)                                                                                                                     
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:19:12)                                                                                   
    at Module._compile (module.js:456:26)                                                                                                             
    at Object.Module._extensions..js (module.js:474:10)                                                                                               
    at Module.load (module.js:356:32)                                                                                                                 
    at Function.Module._load (module.js:312:12)                                                                                                       
    at Function.Module.runMain (module.js:497:10)

rahul2704, 2016-02-12:
same as before! installing as you told me does this again! i believe installing blynk-lib overwrites mraa?
any walkarounds? as i need both libraries? how to get mraa back?? using echo cmd as before
?

arfoll, 2016-02-12:
Ah urgh they run a script to install the lib afterwards...

in any case, I dont get it, I did the following:
```
npm install -g mraa
npm install blynk-library
```

And it worked fine for me.

arfoll, 2016-02-17:
Closing due to lack of activity and 'works for me'

closed 2016-02-17 by arfoll