eclipse-mraa/mraa #430: Errors when reading ADC values

BTWS, 2016-02-16:
I have a similar issue to #117.

I'm using a Intel Galileo Gen 2. It works a few times and then it stops working.

[image]

[image]

This is the code i'm using:

// eglibc needed: https://software.intel.com/en-us/iot/hardware/galileo/downloads
import mraa  # https://github.com/intel-iot-devkit/mraa
import math

print mraa.getVersion()

RESOLUTION = 10
NUMOFSENSORS = 6
REFERENCEVOLTAGE = 5.0 # Can be changed to 3.3V if needed

// aPins = [mraa.Aio(0),mraa.Aio(1),mraa.Aio(2),mraa.Aio(3),mraa.Aio(4),mraa.Aio(5)]
aVal = [None] * NUMOFSENSORS

aPins = []
for i in range(NUMOFSENSORS):
    aPins.append(mraa.Aio(i))

def setAllResolution(aPinArray, res):
    for a in range(len(aPins)):
        aPins[a].setBit(res) 

def getAllSensorValues():
    for i in range(NUMOFSENSORS):
        try:
            aVal[i] = getVoltage(aPins[i])
        except Exception, e:
            print e
            print ("Are you sure you have an ADC?")

setAllResolution(aPins,10)
getAllSensorValues()

for p in aVal:
    print ("%.5f" % p)  # Print only 5 decimals after comma

alext-mkrs, 2016-02-17:
And getVoltage() is?

BTWS, 2016-02-21:
I'm sorry. I didn't upload the whole file.

# -*- coding: utf-8 -*-
# eglibc needed: https://software.intel.com/en-us/iot/hardware/galileo/downloads
import mraa  # https://github.com/intel-iot-devkit/mraa
import math
import os
import os.path
import time

print mraa.getVersion()

# Analog read example for Intel Galileo Gen 2

RESOLUTION = 12
NUMOFSENSORS = 6
REFERENCEVOLTAGE = 5.0 # Can be changed to 3.3V if needed
PULLUPS = [49,51,53,55,57,59]

# --Special thanks to--
# http://www.emutexlabs.com/component/content/article?id=203:getting-started-with-intel-galileo-gen-2
# -->4.3 Pin configuration options for Galileo Gen 2
# https://communities.intel.com/message/249663
# http://wiki.ros.org/IntelGalileo/IntelGalileoGPIO

# aPins = [mraa.Aio(0),mraa.Aio(1),mraa.Aio(2),mraa.Aio(3),mraa.Aio(4),mraa.Aio(5)]
aVal = [None] * NUMOFSENSORS

aPins = []
for i in range(NUMOFSENSORS):
    print i
    aPins.append(mraa.Aio(i))

def init():
    # Use GPIO table from right to left
    for i in range (NUMOFSENSORS):
        pinExport(PULLUPS[i])
        initializePin(i)
    setAllResolution(aPins)
    
def initializePin(aPin):
    pinDir(str(PULLUPS[aPin]), 'in');
    aPins[aPin].read()  # #For some reason it does not update until you read, so lets read
    
def getAnalogFloatValue(aPin):  # Gives float between 0.0 and 1.0
    val = aPin.readFloat()
    return val
    
def getRawValue(aPin):
    val = aPin.read()   # value between 0-1023 or 0-4095 depending on 10 or 12 bit
    return val
    
def setResolution(aPin, res):
    aPin.setBit(res)    # 10 or 12

def setAllResolution(aPinArray, res = RESOLUTION):
    for a in range(len(aPins)):
        aPins[a].setBit(res)  

def getResolution(aPin, res):
    return aPin.getBit(res)
    
def getVoltage(aPin, refVolt = REFERENCEVOLTAGE):
    val = aPin.readFloat()
    val = val * refVolt
    return val

def getAllSensorValues():
    for i in range(NUMOFSENSORS):
        try:
            aVal[i] = getVoltage(aPins[i])
            # aVal[i] = getRawValue(aPins[i])
        except Exception, e:
            print e
            print ("Are you sure you have an ADC?")
            
def readSensorRaw(sensPIN):
    path = "/sys/bus/iio/devices/iio:device0/in_voltage%s_raw" % sensPIN
    File = open(path, 'r')
    return (int(File.read()))
    
def readSensorsRawPro(number):
    sensordata = list()
    for i in range(number):
        sensordata.append(readSensorRaw(i))
    return sensordata
    
#Reads sensordata and return list
def readSensorsRaw():
    sensordata = list()
    File0 = open('/sys/bus/iio/devices/iio:device0/in_voltage0_raw', 'r')	#Back left
    File1 = open('/sys/bus/iio/devices/iio:device0/in_voltage1_raw', 'r')	#Front left
    File2 = open('/sys/bus/iio/devices/iio:device0/in_voltage2_raw', 'r')	#Back right
    File3 = open('/sys/bus/iio/devices/iio:device0/in_voltage3_raw', 'r')	#Front right
    sensordata.append(int(File0.read()))
    sensordata.append(int(File1.read()))
    sensordata.append(int(File2.read()))
    sensordata.append(int(File3.read()))
    return sensordata
   
def pinSetup(pinexp, pindir, pinval):
    if os.path.exists("/sys/class/gpio/gpio%s" % pinexp) == False:
        strexport = ("echo \"%s\" > /sys/class/gpio/export" % pinexp)
        os.system(strexport)  # write a GPIO number to the export file, which reserves it for use
        print strexport
    else:
        print "INFO: GPIO %s already exists, skipping export" % pinexp
    strdir = ("echo \"%s\" > /sys/class/gpio/gpio%s/direction" % (pindir, pinexp))
    strval = ("echo \"%s\" > /sys/class/gpio/gpio%s/value" % (pinval, pinexp))
    os.system(strdir)  # sets GPIO as an output/input('out'/'in')
    os.system(strval)  # sets GPIO high/low('1'/'0'), enable dir_out
    print strdir
    print strval

def pinExport(pinexp):
    if os.path.exists("/sys/class/gpio/gpio%s" % pinexp) == False:
        strexport = ("echo \"%s\" > /sys/class/gpio/export" % pinexp)
        os.system(strexport)  # write a GPIO number to the export file, which reserves it for use
        print strexport
    else:
        print "INFO: GPIO %s already exists, skipping export" % pinexp

def pinHLDir(pinexp, pindir, pinval):
    strdir = ("echo \"%s\" > /sys/class/gpio/gpio%s/direction" % (pindir, pinexp))
    strval = ("echo \"%s\" > /sys/class/gpio/gpio%s/value" % (pinval, pinexp))
    # first %s is replaced with first parameter, second %s is replaced with second parameter
    # \" = "
    # print strdir
    # print strval
    os.system(strdir)  # sets GPIO as an output/input('out'/'in')
    os.system(strval)  # sets GPIO high/low('1'/'0'), enable dir_out   

def pinDir(pinexp, pindir):
    strdir = ("echo \"%s\" > /sys/class/gpio/gpio%s/direction" % (pindir, pinexp))
    os.system(strdir)  # sets GPIO as an output/input('out'/'in')
    print strdir

def pinHL(pinexp, pinval):
    strval = ("echo \"%s\" > /sys/class/gpio/gpio%s/value" % (pinval, pinexp))
    os.system(strval)  # sets GPIO high/low('1'/'0'), enable dir_out
    print strval

init()

# print readSensorRaw(0)
sensdata = list()
sensdata = readSensorsRawPro(6)

for s in sensdata:
    print s
    
getAllSensorValues()

for p in aVal:
    print ("%.5f" % p)  # Print only 5 decimals after comma

This code works, but when I connect 5V/GND/3.3V directly to A4,A5 errors appear. Doing the same with pin A0,A1,A2,A3 works perfectly.

[image]

In this screenshot I connected GND to A0 and the second time GND to A5.

arfoll, 2016-02-21:
Wow - ok, remove all that initlisation stuff, this is what mraa does for you... You don't need anything else. if someone isn't working with:

```
a0 = mraa.Aio(0)
a0.read()
```

Then file a bug. Likely the multiple muxing is confusing stuff, admitedly the driver crashing is not a good thing (tm) :) But you should not be doing any pin setup behind mraa's back. If you really want to that's why we have mraa's 'raw' GPIO api..

BTWS, 2016-02-22:
I wrote some simple code to test. Connecting Ground/5V/3.3V directly to A4 or A5 gives errors.

```python
# eglibc needed: https://software.intel.com/en-us/iot/hardware/galileo/downloads
import mraa  # https://github.com/intel-iot-devkit/mraa
import math

print mraa.getVersion()

RESOLUTION = 12
NUMOFSENSORS = 6
REFERENCEVOLTAGE = 5.0 # Can be changed to 3.3V if needed

# aPins = [mraa.Aio(0),mraa.Aio(1),mraa.Aio(2),mraa.Aio(3),mraa.Aio(4),mraa.Aio(5)]
aVal = [None] * NUMOFSENSORS

aPins = list()
for i in range(NUMOFSENSORS):
    aPins.append(mraa.Aio(i))

    
def getAnalogFloatValue(aPin):
    val = aPin.readFloat()
    return val
    
def setResolution(aPin, res):
    aPins[aPin].setBit(res)    #   10 or 12

def setAllResolution(aPinArray, res = RESOLUTION):
    for a in range(len(aPins)):
        aPins[a].setBit(res) 

def getVoltage(aPin, refVolt = REFERENCEVOLTAGE):
    val = aPin.readFloat()
    val = val * refVolt
    return val

def getAllSensorValues():
    for i in range(NUMOFSENSORS):
        try:
            aVal[i] = getVoltage(aPins[i])
        except Exception, e:
            print e
            print ("Are you sure you have an ADC?")

setAllResolution(aPins, 12)
getAllSensorValues()

for p in aVal:
    print ("%.5f" % p)
```

In this image I connected Ground to A3. The second time to A4.

[image]

BTWS, 2016-02-23:
Is it possible to implement this in the mraa library (see link)?

https://communities.intel.com/message/375502#375502 

> Actually a hardware implementation of A4/A5 and A0/A1/A2/A3 is diffedent on Galileo Gen1 board.
> A4 and A5 require to setup gpio29 (no need for A0/A1/A2/A3).
 
> If gpio29 = LOW and A4/A5 are connected to 5V/3.3V/GND, I2C bus is blocked.
> As a result there is no response from all devices connected to this I2C bus.
> (for example, from CY8C9540A chip which is responsible for gpio16-gpio55).
 
> Also I recommend to tune gpio20.
 
> Same logic is present on Gen2 board where need to tune gpio60.

> **Conclusion**:
> 1. need to perform a correct setting of gpio29 and gpio20 on Gen1 board.
> 2. need to perform a correct setting of gpio60, gpio78 and gpio79 on Gen2 board.
>
> xbolshe

BTWS, 2016-03-07:
What are the correct settings of the pins A4 and A5 to use it as an ADC and not I2C?

I used these settings and it didn't work.
> # A4
> 78 = (value = LOW)
> 60 = (direction = 'out, value = HIGH)
>  
> # A5
> 79 = (value = LOW)
> 60 = (direction = 'out, value = HIGH)

[image]
[source](http://www.emutexlabs.com/component/content/article?id=203:getting-started-with-intel-galileo-gen-2)

xbolshe, 2016-03-11:
I guess the problem is related with https://github.com/intel-iot-devkit/mraa/issues/390
Also check https://github.com/intel-iot-devkit/mraa/issues/117

BR,
xbolshe

xbolshe, 2016-03-11:
@BTWS 
Try to setup gpio57 and 59 to INPUT direction. It will disable resistors. I guess it will fix a time-out issue.

BR,
xbolshe

xbolshe, 2016-03-12:
Here is my proposal to fix:
https://github.com/intel-iot-devkit/mraa/pull/459

BR,
xbolshe

alext-mkrs, 2016-03-28:
The fix was commited with 3397c95c0a15fa42d15ed0b7a4b04ff0f6d17276 and 95c259f6b2420f04078ea83ec5c9a9a9e52f3a77, closing this one.

closed 2016-03-28 by alext-mkrs