eclipse-mraa/mraa #432: Blink example fails on Edison

uchobby, 2016-02-17:
Using the following code in node...
blink.js......
var mraa = require("mraa");

mraa.init();
console.log('MRAA Versions: ' + mraa.getVersion()); //write the mraa version to the Intel XDK console
console.log('MRAA Platform Name:'+mraa.getPlatformName());
var myOnboardLed = new mraa.Gpio(4); //LED hooked up to digital pin 13 (or built in pin on Galileo Gen1 & Gen2 plus Edison Arduino Breakout board)
myOnboardLed.dir(mraa.DIR_OUT); //set the gpio direction to output
console.log("Pin:"+mraa.getPinName(4));
console.log("  Dir:"+myOnboardLed.readDir());
console.log("  R:"+myOnboardLed.read());

var ledState = true; //Boolean to hold the state of Led

periodicActivity(); //call the periodicActivity function

function periodicActivity() {
    console.log("Blink");
    myOnboardLed.write(ledState?1:0); //if ledState is true then write a '1' (high) otherwise write a '0' (low)
    ledState = !ledState; //invert the ledState
    setTimeout(periodicActivity,1000); //call the indicated function after 1 second (1000 milliseconds)
}

Run command...
root@edison:~/projects/sockettest# node blink.js

Console output...
MRAA Versions: v0.9.3
MRAA Platform Name:Intel Edison
Pin:J17-14
  Dir:0
  R:0
Blink
Blink
Blink
and on and on...

Using and O-Scope I can tell that pin 13 on the Arduino header is not set for output.

I have tried other pins, many valid arduino pin numbers cause an errors. For example the error below is output if I try to use pin 4.  

root@edison:~/projects/sockettest# node blink.js
/home/root/projects/sockettest/blink.js:10
(or built in pin on Galileo Gen1 & Gen2 plus Edison Arduino Breakout board)
    ^^^^^
SyntaxError: Unexpected identifier
    at exports.runInThisContext (vm.js:73:16)
    at Module._compile (module.js:443:25)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3

arfoll, 2016-02-17:
I'm confused - why are you asking for pin4 but want pin13 to flash? Just open pin13 ```new mraa.Gpio(13)```.

dfowler7437, 2016-02-17:
Tried pin4 in case pin13 was bad. Expect LED to flash on 13. Used an O-Scope to monitor pins.  Why would pin4 create an error?  Why would 13 not flash the LED?

I'm thinking that mraa is not detecting the Arduino carrier board correctly.  I did see a mention of some tristate issue with an insane suggestion that you should not reset the board with the reset button.  Will test this...

arfoll, 2016-02-17:
oooh yes you should not use the reset button (agreed it's insane), just pull the power. Check mraa.getPlatformVersion() should return 'arduino'

uchobby, 2016-02-17:
Just tested and that was the issue!  OMG so LAME!! we cant use reset? Really?  And this is not in bold H1 on the getting started with mraa pages?   

I think the bug report he is a failure of the documentation to inform users about this issue.

The fix is and H1 tag in the readme.rd with something like "Beware poor hardware design makes reset button unusable with mraa"

Yes I'm frustrated, hours..  I will never get back!

arfoll, 2016-02-17:
I'm real sorry for your lost time, it's not on the front page because mraa is for all platforms not just for edison. Though I've added something in the edison.md page

My personal experience of embedded devices is to never use the soft reset buttons and always do a full cold reboot as those are always the more reliable option.

closed 2016-02-17 by arfoll

dfowler7437, 2016-02-17:
Reset buttons are by no means soft resets. They should, by design, be equivalent to power up reset. The button is typically called a hard reset. It's usually better to press the button as a power cycle could be brutal to the components in a system. Power cycling is temperature cycling...

I recommend that you do use a reset button when a hard reset is required.

arfoll, 2016-02-17:
What I meant is that its implemented as a warm reboot which changes a few
things. I'm not saying its correct or good, just that's the way it is. I'm
not the hardware designer of Intel boards or eve a hardware engineer so
complaining here won't do much ;-) we've already conveyed our displeasure
but feel free to join the Intel forum and complain more if you wish...

'Could be brutal'. Sure hardware guys have told me I'm a brute but I've
never fried a board that way so until then...

WoodyWoodsta, 2016-08-08:
Probably should have posted https://github.com/intel-iot-devkit/mraa/issues/49#issuecomment-238210319 here, as this is a more recent issue. Apologies! Any help would be greatly appreciated!

commit 6b379400 (2016-02-17): edison.md: Add warning on using reset button

Closes #432