eclipse-mraa/mraa #434: pip support?

miminashi, 2016-02-18:
Is there a pip support for Python binding?

For node.js, I can install mraa by:

```
npm install mraa
```

I want to install mraa by following command:

```
pip install mraa
```

arfoll, 2016-02-18:
Yeah it would be nice, but not really on my priority list I'll be honest, if someone has an idea on how to make it work nicely I'm happy to add it to our infra so we push it to the right place, but I'm guessing we'd need to do something similar to what we do for node.js which is quite a bit of work.

Psychokiller1888, 2019-06-23:
3.5 years later it is still not available? It's a shame, because mraa is good, but total waste of time to install in virtual environement

ax-meyer, 2021-08-31:
Yep, would be really nice to have. I am trying to use mraa in a docker container and installation through pip would make my life a lot easier...