eclipse-mraa/mraa #436: analog input on Edison/Arduino breakout is broken with latest MRAA master (9.3)

jontrulson, 2016-02-20:
Hi, I've recently started on a new analog driver, and started by updating to the latest MRAA (9.3, latest commit: 6b3794003f36f880d5b9d90c6e6bf7dab6bcea5b ).  

The driver works fine on the G2.  On edison, analog reads always read 0.  Reading the sysfs files directly also always report 0 (before and after an analog context has been initialized), and of course analog drivers no longer function.

There was a similar problem like this in the past that I reported that was caused by the removal of some SPI initialization code (needed due to the fact that the ADC is accessed via SPI).

This code now seems to reside in mraa_intel_edison_spi_init_pre(), and only seems to be called when initializing an SPI context.

If I modify my example code to just initialize a dummy SPI context before calling the drivers ctor(), all works fine again.

ie:

mraa::Spi dummy(0);

// create analog contexts, do analog reads, etc....

So... this needs to be fixed, otherwise analog is non-functional on edison/arduino unless you also initialize an unnecessary SPI context.   I don't think this is the intended behavior :)

arfoll, 2016-02-20:
Argh - so I tried to fix this in the adc branch - can you take a look and verify as I'm away from an edison board all weekend :/ I was assured that this wasn't a problem by Constantin, I'm mostly annoyed at myself for having let this one slip yet again :/

jontrulson, 2016-02-20:
Yes, this seems to have fixed the problem.  Analog reads seem to be working fine now.  Thanks!

arfoll, 2016-02-22:
0.9.4 released with the fix.

closed 2016-02-22 by arfoll