eclipse-mraa/mraa #440: GPIO output toggle on start

pdumais, 2016-02-25:
I have Node.js application that controls 2 GPIO for controlling relays.
Whenever I start my application, my relays seem to toggle. I can here them click. I've isolated the problem down to just including MRAA in my application. Ideed, if I remove all code and just do a "var mraa = require("mraa");", I here my relays click.

Is this a know issue/limitation?

libmraa.so.0.8.0

tingleby, 2016-02-25:
Is this Intel Edison?

pdumais, 2016-02-25:
Yes, I'm sorry. 
Using Linux version 3.10.17-poky-edison+ (sys_dswci@tlsndgbuild004) (gcc version 4.9.1 (GCC) ) #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015

alext-mkrs, 2016-02-25:
First of all, I'd strongly suggest you to update mraa, the latest is 0.9.4.

Second thing, what expansion board is this and which exact GPIOs are we talking about?

pdumais, 2016-02-26:
I'm using the arduino board. It seems to be doing this with any GPIO. But for my particular case 15,16,17,18, 19 (which are the A1-A5 on arduino). I've also had the same proble with arduino pin D3,D4,D5. 

The problem seems to be that if the app sets the output high then exists, upon re-executing the app, the GPIO seems to get pulled low again when executing the require() instruction. 

So since my app only pulls the GPIO high, this gives a toggle effect.

alext-mkrs, 2016-02-28:
Okay, that sounds to me like the need to add a cleanup section to your program. Library initialization brings subsystems into a known, specific state and that's I would say a feature, not a bug.

pdumais, 2016-02-29:
I understand. This is limiting though since my app really is just a script to toggle a relay. So I want the relay to stay in that state once the app exits, and I want to stay the same once I re-enter it.

Also, what if a an application would crash for any reason? I would  like to resume my app and keep the GPIO state as it was before.

While I do expect a library to initialize cleanly in a known state, I expect it not to change the status of my GPIOs 

Maybe I suggest that instead of having this cleanup done in the library init, that it should be done explicitly though something like mraa_reset() or something like that? I'm sure I won't be the only user that would find this usefull.

Thanks.

pdumais, 2016-03-02:
Also, I worry about what will happen if one script is running, setting a GPIO high, while a second script gets invoked and also require mraa. In that case, my GPIOs would toggle.

pdumais, 2016-03-27:
Do those concerns sound valid? Does the team agree that this should get fixed?

arfoll, 2016-03-27:
We're working on fixing this via imraa which currently nearly does
everything needed. Basically we'd use imraa to initialise the board once
and then piggyback off the already initialised board to avoid remuxing
pins. Sadly this isn't quite finished, hopefully in a few weeks this will
work well.

pdumais, 2016-03-27:
Thank you, I appreciate the feedback. Any chance you could link this issue to the work currently being done?

arfoll, 2016-03-27:
There isn't currently an issue to describe it, I'll make one later tonight,
firmata branch merged imraa
https://github.com/intel-iot-devkit/mraa/tree/master/imraa and includes the
io setup part but doesn't yet include the skipping in mraa_init when a
platform is already initialised. We ran out of time to implement this so
will be done in next sprint unless someone beets me to it :)

closed 2017-10-30 by pdumais