eclipse-mraa/mraa #442: Java/Edison; Writing to the /dev/ttyMFD1 does not work BUT reading does

youcangetme, 2016-02-26:
I am having trouble reading serial data from a Java capped mraa app running on an Edison. The Java app reads just fine but when it transmits data the listener never receives it. I have used another Arduino to mimic the opposite end and the listener does in fact receive serial data when sent.

```java

/**
 * Attempting to read Serial TTL output from Edison
 */

import mraa.Result;
import mraa.Uart;
import mraa.UartParity;

public class SerialTester {

	static Uart uart;

	static {
		System.loadLibrary("mraajava");
	}

	public static void main(String[] args) throws Exception {
		setup();
		while (true) {
			String s = readSerial();
			if (s.length() > 0) {
				System.out.println("--> " + s);
				uart.writeStr("Anybody listening? Hello?");
			}
		}
	}

	public static String readSerial() {
		StringBuilder tmp = new StringBuilder();
		while (uart.dataAvailable()) {
			tmp.append(uart.readStr(1));
		}
		return tmp.toString();
	}

	public static void setup() {
		uart = new Uart("/dev/ttyMFD1");

		if (uart.setBaudRate(9600) != Result.SUCCESS) {
			System.err.println("Error setting baud rate");
			System.exit(1);
		}
		if (uart.setMode(8, UartParity.UART_PARITY_NONE, 1) != Result.SUCCESS) {
			System.err.println("Error setting mode");
			System.exit(1);
		}
		if (uart.setFlowcontrol(false, false) != Result.SUCCESS) {
			System.err.println("Error setting flow control");
			System.exit(1);
		}
		
		uart.writeStr("Hello World!");
	}
}

```

arfoll, 2016-02-26:
Use Uart(0) instead of giving device, that'll set the pinmux correctly

youcangetme, 2016-02-26:
Works correctly now, thank you for your fast response!

closed 2016-02-27 by alext-mkrs