eclipse-mraa/mraa #448: compile mraa on BeagleBone

Pillar1989, 2016-03-01:
Hi, 
   I try to follow the compile docment build mraa for beaglebone black. But it popup error.
nodejs:v0.12.10
swig:v3.0.8
root@beaglebone:~/mraa_and_upm/mraa/build# make
Scanning dependencies of target mraa
[  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
[  2%] Building C object src/CMakeFiles/mraa.dir/gpio/gpio.c.o
[  3%] Building C object src/CMakeFiles/mraa.dir/i2c/i2c.c.o
[  4%] Building C object src/CMakeFiles/mraa.dir/pwm/pwm.c.o
[  5%] Building C object src/CMakeFiles/mraa.dir/spi/spi.c.o
[  6%] Building C object src/CMakeFiles/mraa.dir/aio/aio.c.o
[  7%] Building C object src/CMakeFiles/mraa.dir/uart/uart.c.o
[  8%] Building C object src/CMakeFiles/mraa.dir/iio/iio.c.o
[  9%] Building C object src/CMakeFiles/mraa.dir/arm/arm.c.o
[ 10%] Building C object src/CMakeFiles/mraa.dir/arm/96boards.c.o
[ 11%] Building C object src/CMakeFiles/mraa.dir/arm/raspberry_pi.c.o
[ 12%] Building C object src/CMakeFiles/mraa.dir/arm/beaglebone.c.o
[ 13%] Building C object src/CMakeFiles/mraa.dir/arm/banana.c.o
[ 14%] Building C object src/CMakeFiles/mraa.dir/version.c.o
[ 15%] Linking C shared library libmraa.so
[ 15%] Built target mraa
[ 16%] Swig source
Scanning dependencies of target _python-mraa
[ 17%] Building CXX object src/python/CMakeFiles/_python-mraa.dir/mraaPYTHON_wrap.cxx.o
/root/mraa_and_upm/mraa/build/src/python/mraaPYTHON_wrap.cxx:11:0: warning: "SWIGPYTHON" redefined
 #define SWIGPYTHON
 ^
<command-line>:0:0: note: this is the location of the previous definition
[ 18%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/mraa.c.o
[ 20%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/gpio/gpio.c.o
[ 21%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/i2c/i2c.c.o
[ 22%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/pwm/pwm.c.o
[ 23%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/spi/spi.c.o
[ 24%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/aio/aio.c.o
[ 25%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/uart/uart.c.o
[ 26%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/iio/iio.c.o
[ 27%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/arm/arm.c.o
[ 28%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/arm/96boards.c.o
[ 29%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/arm/raspberry_pi.c.o
[ 30%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/arm/beaglebone.c.o
[ 31%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/arm/banana.c.o
[ 32%] Building C object src/python/CMakeFiles/_python-mraa.dir/__/version.c.o
[ 33%] Linking CXX shared module _mraa.so
[ 33%] Built target _python-mraa
[ 34%] Swig source
Scanning dependencies of target mraajs
[ 35%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
in namespace 'v8' does not name a template type
 typedef v8::FunctionCallbackInfo<v8::Value> SwigV8Arguments;
             ^
in namespace 'v8' does not name a template type
 typedef v8::PropertyCallbackInfo<v8::Value> SwigV8PropertyCallbackInfo;
             ^
 char*)':
 no member named 'ThrowException'
 #define SWIGV8_THROW_EXCEPTION(err) v8::Isolate::GetCurrent()->ThrowException(err)
                                                                ^
IGV8_THROW_EXCEPTION'
   SWIGV8_THROW_EXCEPTION(v8::Exception::Error(SWIGV8_STRING_NEW(msg)));
   ^
member of 'v8::String'
 #define SWIGV8_STRING_NEW(str) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), str)
                                ^
SWIGV8_THROW_EXCEPTION'
 #define SWIGV8_THROW_EXCEPTION(err) v8::Isolate::GetCurrent()->ThrowException(err)
                                                                               ^
WIGV8_STRING_NEW'
   SWIGV8_THROW_EXCEPTION(v8::Exception::Error(SWIGV8_STRING_NEW(msg)));
                                               ^
loadErrorHandler::error(int, const char*)':
member of 'v8::String'
 #define SWIGV8_STRING_NEW(str) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), str)
                                ^
WIGV8_STRING_NEW'
     err = v8::Exception::Error(SWIGV8_STRING_NEW(msg));
                                ^
 no member named 'ThrowException'
 #define SWIGV8_THROW_EXCEPTION(err) v8::Isolate::GetCurrent()->ThrowException(err)
                                                                ^
IGV8_THROW_EXCEPTION'
         SWIGV8_THROW_EXCEPTION(err);
         ^
Proxy()':
 no member named 'AdjustAmountOfExternalAllocatedMemory'
 #define SWIGV8_ADJUST_MEMORY(size) v8::Isolate::GetCurrent()->AdjustAmountOfExternalAllocatedMemory(size)
                                                               ^
IGV8_ADJUST_MEMORY'
     SWIGV8_ADJUST_MEMORY(SWIGV8_AVG_OBJ_SIZE);
     ^
Proxy()':
8::Object>' has no member named 'Reset'
     handle.Reset();
            ^
 no member named 'AdjustAmountOfExternalAllocatedMemory'
 #define SWIGV8_ADJUST_MEMORY(size) v8::Isolate::GetCurrent()->AdjustAmountOfExternalAllocatedMemory(size)
                                                               ^
IGV8_ADJUST_MEMORY'
     SWIGV8_ADJUST_MEMORY(-SWIGV8_AVG_OBJ_SIZE);
     ^
/root/mraa_and_upm/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: At global scope:
amespace 'v8' does not name a template type
   void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
                           ^
efore '<' token
   void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
                                           ^
ncePtr(v8::Handle<v8::Object>, void**, swig_type_info*, int)':
r call to 'v8::HandleScope::HandleScope(v8::Isolate*)'
 #define SWIGV8_HANDLESCOPE() v8::HandleScope scope(v8::Isolate::GetCurrent());
                                                                             ^
IGV8_HANDLESCOPE'
   SWIGV8_HANDLESCOPE();
   ^
/root/mraa_and_upm/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:793:77: note: candidates are:
 #define SWIGV8_HANDLESCOPE() v8::HandleScope scope(v8::Isolate::GetCurrent());
                                                                             ^
IGV8_HANDLESCOPE'
   SWIGV8_HANDLESCOPE();
   ^
In file included from /usr/include/nodejs/src/node.h:62:0,
                 from /root/mraa_and_upm/mraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:156:
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
   HandleScope(const HandleScope&);
   ^
' to 'const v8::HandleScope&'
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
   HandleScope();
.....

How can I fixed it ?

tingleby, 2016-03-01:
@arfoll Should be no problem with node 0.12 right?

arfoll, 2016-03-01:
can you wipe the build dir and paste the output of cmake's configuration again? Likely the SWIG_V8_VERSION is parsed wrong on that particular version of node.js, moving to 4.x will mean less issues that should 'just work'

Pillar1989, 2016-03-02:
#398 Thanks, I need update node-dev.

closed 2016-03-02 by Pillar1989