eclipse-mraa/mraa #449: compiling issue on upm with mraa/firmata branch

longwei, 2016-03-01:
I run into this issue when compiling upm with mraa/firmata branch.
mraa/firmate itself build ok, but when I built upm. I got a multiple definition regarding common.hpp.

```
Linking CXX shared module _pyupm_stepmotor.so
CMakeFiles/_pyupm_stepmotor.dir/stepmotor.cxx.o: In function `mraa::addSubplatform(mraa::Platform, std::string)':
/opt/poky-edison/1.7.2/sysroots/core2-32-poky-linux/usr/include/mraa/common.hpp:301: multiple definition of `mraa::addSubplatform(mraa::Platform, std::string)'
CMakeFiles/_pyupm_stepmotor.dir/pyupm_stepmotorPYTHON_wrap.cxx.o:/opt/poky-edison/1.7.2/sysroots/core2-32-poky-linux/usr/include/mraa/common.hpp:301: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/stepmotor/_pyupm_stepmotor.so] Error 1
make[1]: *** [src/stepmotor/CMakeFiles/_pyupm_stepmotor.dir/all] Error 2
make: *** [all] Error 2
```

I don't have this build error if mraa is built from master branch.

arfoll, 2016-03-01:
@Propanu seems we broke upm :) haven't looked at why can you take a look?

Propanu, 2016-03-01:
Yep, new `addSubplatform()` function is not inlined. Will submit a patch shortly!

Propanu, 2016-03-01:
#451 should fix it.

arfoll, 2016-03-01:
Fix merged

closed 2016-03-01 by arfoll