eclipse-mraa/mraa #452: MRAA on Beaglebone - Jessie (Debian 8.3)

mkellner, 2016-03-02:
I see in the beaglebone sources, a define:
    #define SYSFS_DEVICES_CAPEMGR_SLOTS "/sys/devices/bone_capemgr.*/slots”
On my system running the old Debian 7.9 (Wheezy), this directory exists.

With a newer kernel and version of Debian 8.3 (Jessie), there is no longer a /sys/devices/bone_capemgr.*

It is now /sys/devices/platform/bone_capemgr/slots
The ubuntu releases have this directory as well.

Has anyone built mraa for a more recent OS release than Debian 7.9?

Pillar1989, 2016-03-02:
Hi
root@beaglebone:~/mraa/build/examples# ./mraa-gpio version
Version v0.9.5 on Beaglebone Black Rev. B
root@beaglebone:~/mraa/build/examples# uname -a
Linux beaglebone 4.1.18-ti-rt-r49 #1 SMP PREEMPT RT Thu Feb 25 21:59:17 UTC 2016 armv7l GNU/Linux
root@beaglebone:~/mraa/build/examples#

mkellner, 2016-03-02:
I see that it builds for you. It builds for me too.

Are you able to connect to pins and perform IO?
I'm trying to connect to AIN1 (P9_40, mraa pin 86). I've tried to mraa_aio_init(1); or mraa_aio_init(86) and both indicate that the pin can't do Analog.

When I try the analogin_a0 example, I get this in the syslog:
Dec 11 18:20:15 arm libmraa[2433]: libmraa initialised for platform 'Beaglebone Black Rev. C' of type 6
Dec 11 18:20:15 arm libmraa[2433]: aio: pin uncapable of aio

DiegoPaezA, 2016-04-22:
I have the same problem but my uname -a give:
Linux beaglebone 3.8.13-bone70 #1 SMP Fri Jan 23 02:15:42 UTC 2015 armv7l GNU/Linux
mraa-gpio version
Version v0.10.0 on Beaglebone Black Rev. C

did you find a solution ?

alext-mkrs, 2016-04-25:
I guess the main problem with this one is that no one out of current mraa developers has access to Beaglebone to make sure fix covers all/most variations. The fix itself looks simple at the first glance. @arfoll, or do we have a BB somewhere?

tingleby, 2016-04-25:
Let me see if I can dig mine out, It got buried when I moved.

Pillar1989, 2016-04-25:
Maybe you can try this. https://github.com/Pillar1989/mraa
We have been tested Beaglebone Jessie image.

hcostelha, 2017-04-28:
Is there nay possibility of getting this fixed? I am unable to read an analog input from a BeagleBone Black. It either says "pin XX no capable of aio" or "pin XX beyond platform definition".