eclipse-mraa/mraa #456: Edison SPI can't work on frequency more then 10 kHz?

ivictbor, 2016-03-06:
Hello. This issue is also posted by me at https://communities.intel.com/message/378084#378084 but I also will leave it fere becouse may be it is mraa issue.

I started using Edison with latest firmware **Release 2.1 Yocto* complete image**. But after first tries to make SPI exchanges on **1MHz **I noticed freezing and errors in **dmesg **and found issue on communities https://communities.intel.com/thread/78149?start=0&tstart=0 .
Now I followed recommendation in that topic and downgraded to **Release 2.0 Yocto* complete image**. (edison-image-ww25.5-15)
So now I on:
```
cat /proc/version  
Linux version 3.10.17-poky-edison+ (sys_dswci@tlsndgbuild004) (gcc version 4.9.1 (GCC) ) #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015  
cat /etc/*release  
EDISON-3.0  
Copyright Intel 2015  
configure_edison --version  
159  
```
Is this output correct for **edison-image-ww25.5-15**? Or it is not downgraded correctly? (I copied zip to Edison Flash Drive and issued reboot ota via ssh)
I wrote next simple test python2 script (needs **opkg install mraa**):
```
import mraa  
from time import sleep, time  
  
BULK_SIZE = 512  
SPI_FREQ = 100e3 # 100KHz  
  
print("Starting SPI test using bulks {} at {}".format(BULK_SIZE, SPI_FREQ))  
  
spi = mraa.Spi(1)  
spi.frequency(int(SPI_FREQ))  
spi.lsbmode(True)  
spi.mode(mraa.SPI_MODE3)  
  
start = time()  
tot_bytes = 0  
while True:  
    sleep(0.01)  
    txbuf = bytearray([0xFF] * BULK_SIZE)  
    rxbuf = spi.write(txbuf)  
    tot_bytes += len(txbuf)  
    print("{}: Rate: {} B/s".format(time(), tot_bytes / (time() - start)))  
```
After working one or two minutes it hanged up. I can't kill it with **Ctrl+C**, or suspend with **Ctrl+Z**. I can only exit from SSH with **~.**, reconnect to Edison and see this script in ps.
I can't kill it by **killall -9 python** or by pid with **kill**. Only reboot helps. After two minutes dmesg shows:
```
[  360.610725] INFO: task python2:345 blocked for more than 120 seconds.  
[  360.610813] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.  
[  360.610889] python2        D f5051d8c  6040  345    336 0x00000004  
[  360.610929]  f5051dd4 00000086 f6c81bd0 f5051d8c c12726e3 c1d04000 f6c9e000 f73fdc00  
[  360.610986]  00000046 f6c83d30 f73fdc00 f5d95900 f5d34344 f6c0e000 f5051d98 c12727bf  
[  360.611040]  f5cd8400 f5051da0 c125ad6e f5051dbc c125c4e3 f5cd8400 00000005 f5d81600  
[  360.611095] Call Trace:  
[  360.611135]  [<c12726e3>] ? try_to_wake_up+0x193/0x250  
[  360.611164]  [<c12727bf>] ? wake_up_process+0x1f/0x40  
[  360.611191]  [<c125ad6e>] ? wake_up_worker+0x1e/0x20  
[  360.611216]  [<c125c4e3>] ? insert_work+0x53/0x90  
[  360.611246]  [<c18df1e7>] ? _raw_spin_unlock+0x17/0x40  
[  360.611272]  [<c125ce8f>] ? __queue_work+0x10f/0x340  
[  360.611299]  [<c18de0e3>] schedule+0x23/0x60  
[  360.611324]  [<c18db745>] schedule_timeout+0x165/0x2a0  
[  360.611355]  [<c1270ddb>] ? get_parent_ip+0xb/0x40  
[  360.611380]  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0  
[  360.611408]  [<c18dd19b>] wait_for_completion+0xab/0xe0  
[  360.611433]  [<c1272800>] ? wake_up_state+0x20/0x20  
[  360.611463]  [<c15c1050>] spidev_sync+0x80/0xb0  
[  360.611493]  [<c15c19df>] spidev_ioctl+0x65f/0x6f0  
[  360.611518]  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0  
[  360.611549]  [<c15c1130>] ? spidev_write+0xb0/0xb0  
[  360.611577]  [<c15c1380>] ? spidev_probe+0x190/0x190  
[  360.611604]  [<c1336686>] do_vfs_ioctl+0x306/0x550  
[  360.611632]  [<c1480cba>] ? inode_has_perm.isra.32.constprop.65+0x3a/0x50  
[  360.611657]  [<c1480d57>] ? file_has_perm+0x87/0x90  
[  360.611686]  [<c148113b>] ? selinux_file_ioctl+0x4b/0xe0  
[... 76 lines cut ...]
[  720.558600]  [<c18df1e7>] ? _raw_spin_unlock+0x17/0x40  
[  720.558626]  [<c125ce8f>] ? __queue_work+0x10f/0x340  
[  720.558653]  [<c18de0e3>] schedule+0x23/0x60  
[  720.558678]  [<c18db745>] schedule_timeout+0x165/0x2a0  
[  720.558710]  [<c1270ddb>] ? get_parent_ip+0xb/0x40  
[  720.558735]  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0  
[  720.558762]  [<c18dd19b>] wait_for_completion+0xab/0xe0  
[  720.558787]  [<c1272800>] ? wake_up_state+0x20/0x20  
[  720.558819]  [<c15c1050>] spidev_sync+0x80/0xb0  
[  720.558849]  [<c15c19df>] spidev_ioctl+0x65f/0x6f0  
[  720.558874]  [<c18e2c75>] ? sub_preempt_count+0x95/0xf0  
[  720.558905]  [<c15c1130>] ? spidev_write+0xb0/0xb0  
[  720.558933]  [<c15c1380>] ? spidev_probe+0x190/0x190  
[  720.558960]  [<c1336686>] do_vfs_ioctl+0x306/0x550  
[  720.558988]  [<c1480cba>] ? inode_has_perm.isra.32.constprop.65+0x3a/0x50  
[  720.559013]  [<c1480d57>] ? file_has_perm+0x87/0x90  
[  720.559042]  [<c148113b>] ? selinux_file_ioctl+0x4b/0xe0  
[  720.559068]  [<c1336930>] SyS_ioctl+0x60/0x90  
[  720.559097]  [<c18df928>] syscall_call+0x7/0xb  
[  720.559125]  [<c18d0000>] ? init_intel_cacheinfo+0x1b3/0x37e  
```
If I change SPI_FREQ to 10e3 it work for long time, if to 20e3 it may be hang after several minutes.
What can be wrong? Do I need downgrade to some release earlier then 2.0? Where can I found it? Can I help to community to fix this? What I need for this?
My project needs SPI at several MHz, is it possible with Intel Edison at all? On any Image, any linux - I only need make it work. I bought It because document
"Hardware Guide, September 2014 Revision 002" in section "4.4 SPI interface" says - 25 MHz Master mode, 16.67 MHz slave mode. Is it misspelling in datasheet and it should be "25 KHz Master mode"?

xbolshe, 2016-03-07:
Hi,

as I know this problem is out of MRAA. It is related with a SPI driver of edison-image-ww25.5-15.
You may find more information here:

[https://communities.intel.com/thread/78149?start=15&tstart=0](https://communities.intel.com/thread/78149?start=15&tstart=0)

If you will downgrade the Linux image, the problem may be fixed.

BR,
xbolshe

ivictbor, 2016-03-07:
@xbolshe as I wrote in issue I already downgraded it

xbolshe, 2016-03-07:
@ivictbor I have tested edison-image-ww25.5-15 and edison-image-ww18-15.zip for SPI bus. Both of them do not work in my case. So now I use edison-image-ww05-15.zip. Here is a link to download:

[https://downloadcenter.intel.com/download/24909/Intel-Edison-OS-Image-release-2-0](https://downloadcenter.intel.com/download/24909/Intel-Edison-OS-Image-release-2-0)

BR,
xbolshe

ivictbor, 2016-03-07:
@xbolshe Thanks very much for help! I will try it. It is very sad that Intel released several releases one after other with the same critical bug and they all can be downloaded at the official Intel site... I had a better opinion about Intel

ivictbor, 2016-03-13:
Downgrading to edison-image-ww05-15.zip solved issue

closed 2016-03-13 by ivictbor