eclipse-mraa/mraa #460: Firmata: use cases, guidance?

rwaldron, 2016-03-12:
Hello everyone! I'm trying to get a better understanding for how the Firmata implementation is expected to be used. I've been working with the protocol for a little over 4 years and I'm currently the active maintainer of [Firmata.js](https://github.com/jgautier/firmata/), which is a Node.js module for communicating with microcontrollers running  [StandardFirmata](https://github.com/firmata/arduino/tree/master/examples/StandardFirmata) (StandardFirmataPlus is actually the recommended variant these days). 

The most common use cases: 

1. Arduino Uno & Mega come with StandardFirmata preloaded, so out of the box the board can communicate with programs using Firmata.js
1. With the Arduino IDE, a user uploads StandardFirmata (StandardFirmata, or any of the variants: \*ChipKIT, \*Ethernet, \*EthernetPlus, \*Plus, \*WiFi)  StandardFirmata to their Arduino compatible board, then the board can communicate with programs using Firmata.js

I'm trying to understand how mraa's Firmata implementation fits into those well known use cases. What boards does this run on? Digging through the source I've seen mentions of the Curie, so presumably the Arduino 101? Does this run by default, ie. can I connect the Arduino 101 to a host computer and write, for example, Node.js programs that interact with the board via Firmata protocol? What is the process for getting started there? 

Or is it the other way around? Based on the code in [firmata.c](https://github.com/intel-iot-devkit/mraa/blob/firmata/src/firmata/firmata.c) (and implied by the [example host program](https://github.com/intel-iot-devkit/mraa/blob/firmata/examples/firmata_curie_imu.c)), it looks like the implementation is actually meant to be a _host_ that would connect to a microcontroller running one of the variants mentioned above? 

Sorry for the mountain of questions, but any information would be greatly appreciated! Thanks in advance :)

arfoll, 2016-03-14:
Hi Rick, the firmata stuff is a bit WIP so that's why there's no docs etc :) But let me try answer your questions.

1. Initially this is targeted at the the Arduino 101, realistically there's nothing board specific about what we're doing so it'll likely work on all although testing will likely be limited.
2. We actually want it to be easier than uploading anything with the IDE. We're creating something called 'imraa' which will flash users boards as soon as they're plugged in to the device running imraa with the correct version of Firmata
3. We're making an extensible Firmata plugin for the 101 IMU and we'll provide UPM modules for talking to it.
4. We're investigating how to use the FirmataBLE stuff, this is still an investigation at this stage. So far we're only supporting the std/stdplus variants but we're open to supporting more.
5. Firmata boards will be treated as 'subplatforms' so once you add the subplatform offset you'll be on the firmata board, for example arduino GPIO13 == 512+13. I2c bus 0 becomes 0 + 512 etc... We plan on supporting multiple firmata connected boards but not in the first instuance. The only requirement is to add the board via mraa_add_subplatform(MRAA_GENERIC_FIRMATA, "/dev/ttyACM0"). We're also planning on making mraa_init() detect arduino 101s automatically (and possibly others) so that this call is not even required.

If you want to try it out we can keep this thread running in case you run into any issues!

rwaldron, 2016-03-14:
> We actually want it to be easier than uploading anything with the IDE. We're creating something called 'imraa' which will flash users boards as soon as they're plugged in to the device running imraa with the correct version of Firmata

How is the correct version determined?

arfoll, 2016-03-14:
Correct version is currently simply flashing with what we have set in /etc/imraa.conf which will point to a binary sketch in /usr/share/mraa/. For now we're using a StandardFirmata sketch but we'll move to a sketch based on ConfigurableFirmata with the IMU plugin v.shortly.

rwaldron, 2016-03-14:
> Firmata boards will be treated as 'subplatforms' so once you add the subplatform offset you'll be on the firmata board, for example arduino GPIO13 == 512+13. I2c bus 0 becomes 0 + 512 etc... 

Can you give me a concrete example?

arfoll, 2016-03-14:
see examples/i2c_firmata.c in firmata branch, be essentially here's an example that reads the test byte from a bmp085 sensor over i2c.

```
int
main(int argc, char** argv)
{
    mraa_init();
    mraa_add_subplatform(MRAA_GENERIC_FIRMATA, "/dev/ttyACM0");

    mraa_i2c_context i2c;
    i2c = mraa_i2c_init(0 + 512);
    mraa_i2c_address(i2c, 0x77);
    int res = mraa_i2c_read_byte_data(i2c, 0xd0);
    printf("res is 0x%x\n", res);
}
```

rwaldron, 2016-03-14:
Thanks, I think I was having trouble understanding what subplatform meant in this context. 

Re: the Firmata IMU plugin, I think there needs to be a "non-standard command" byte defined in the Firmata protocol, which you can use to safely implement any additional commands/responses. 

cc'ing @soundanalogous

arfoll, 2016-03-14:
Indeed, this is what we have, i'll publish it this week.

soundanalogous, 2016-03-14:
> Re: the Firmata IMU plugin, I think there needs to be a "non-standard command" byte defined in the Firmata protocol, which you can use to safely implement any additional commands/responses.

Is this regarding the Arduino 101 on-board IMU?

arfoll, 2016-03-14:
@soundanalogous yes the byte we're using is
```#define CURIE_IMU (0x11)```

soundanalogous, 2016-03-14:
FYI I'm considering the use of 14bit IDs for devices: https://github.com/firmata/ConfigurableFirmata/issues/16#issuecomment-184387110

Hansen0314, 2020-10-10:
Hi @arfoll :
 can MRAA  support [StandardFirmata Plus](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino) for now? I got some trouble when I init Arduino UART by using MRAA  firmata.
[image]