eclipse-mraa/mraa #463: Firmata: I2C read/continuous read have inconsistent signatures

rwaldron, 2016-03-14:
Firmata allows _optionally_ specifying a register to read from in both read and continuous read modes. [Presently, the following is true](https://github.com/intel-iot-devkit/mraa/blob/firmata/src/firmata/firmata_mraa.c#L119-L178): 

- `mraa_firmata_send_i2c_read_req` doesn't accept a `register`
- `mraa_firmata_send_i2c_read_cont_req` does accept a `register`, but cannot be called without `register`

I believe this can be easily corrected with overloaded functions, if that's acceptable?

arfoll, 2016-03-15:
Yeah sure, the two functiosn are very much similar, no issues with that

rwaldron, 2016-03-21:
I'm not sure how to resolve this. I tried using the approach I explained above, but the current compiler and configuration disallow that form of function overload definition. Any ideas how I should proceed to get around that? 

My second approach was an attempt to use stdarg, but that was quickly abandoned.

arfoll, 2016-03-21:
Ok so I got the wrong end of the stick. Continuous read is not at all what I'd assumed it was, it's very much a firmata thing (https://github.com/firmata/protocol/blob/master/i2c.md). So for now I'll change it to a standard read all the time.

If we want to do something like that it'll have to be done via firmata sysex calls, or something similar To be fair I need to work out how to get those APIs to something non C, currently they're C only.

rwaldron, 2016-03-21:
> So for now I'll change it to a standard read all the time.

Ok, but there is still the issue that `mraa_firmata_send_i2c_read_req` requires a register (it's called `command` here), but a register is optional. 

Also, just to illuminate on continuous reads: I2C continuous read is actually the exact same semantics as digital and analog reporting, ie: "send a request once, which results in a read and report on every sample interval until told to stop" (https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino#L800-L833).