eclipse-mraa/mraa #468: Incorrect version of swig detected after cmake command

jancassio, 2016-03-17:
Hello,

I'm trying to compile mraa to use with nodejs but I can't because (I guess) the `cmake`routine found a incorrect version of `swig` installed in my system.

I'm 100% sure I'm using swig 3.0.8 in my Ubuntu image, for some reason the version found is 3.0.2 and the required is 3.0.5.

Below the full output:

```
$ cmake -DBUILDSWIGNODE=ON ..
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
INFO - libmraa Version v0.6.1-50-g3184965
INFO - Target arch is x86_64
INFO - Adding x86 platforms
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.10", minimum required is "2.7") 
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.10", minimum required is "2.7") 
WARN - SWIG is 3.0.2. Please upgrade to 3.0.5+ to build nodejs addon
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jancassio/Works/labs/node/mraa/src/mraa/build
```

Also, I tried to do the same step using branch `tocompile` without success.

Thanks,

xbolshe, 2016-03-17:
So, /usr/bin/swig3.0 has a version 3.0.2. 
May you execute "/usr/bin/swig3.0 -version" and post an output here?

And where is swig 3.0.8 installed?

BR,
xbolshe

jancassio, 2016-03-17:
I installed swig from the source, the location is:
`/usr/local/bin/swig`

At `/usr/bin` I found swig2.0 and swig3.0 (shame of me).

Can I change the swig path at make process? Or is better to move my latest swig to /usr/bin ?

xbolshe, 2016-03-17:
You may use any as you wish: setup SWIG_EXECUTABLE or move a file to /use/bin, change a prefix to /usr/bin during swig compilation.

BR,
xbolshe

jancassio, 2016-03-17:
I'll keep the latest one and remove others, seems fair enough. 
Thank you!

closed 2016-03-17 by jancassio