eclipse-mraa/mraa #469: Error: Illegal arguments for construction of _exports_Uart

arane4114, 2016-03-17:
Im working with the javascript [uart example](https://github.com/intel-iot-devkit/mraa/blob/master/examples/javascript/uart.js) its giving me the following error when I attempt to run it with node version v4.4.0 on an Edison. Ive run it as a normal user and as root.

> MRAA Version: v0.9.6
/home/abi/tma-comms/mraa_test.js:4
u = new m.Uart(1)
Error: Illegal arguments for construction of _exports_Uart
    at Error (native)
    at Object.<anonymous> (/home/abi/tma-comms/mraa_test.js:4:5)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3
root@ubilinux:/home/abi/tma-comms#

xbolshe, 2016-03-17:
MRAA defines only one UART for Edison - Uart(0).

BR,
xbolshe

wouterds, 2016-03-18:
I have the exact same issue on Raspberry Pi (1) with Linux Jessie (it's occurring in a vendor lib that I'm using).

arfoll, 2016-03-20:
Does this fix the issue?

arane4114, 2016-03-23:
I uninstalled and completed from source. The issue persists with u = new m.Uart(0), but works with u = new m.Uart('/dev/ttyMFD1'). Im getting garbage from the reads, but I suspect thats an issue with my settings/electrical not so much MRAA.

arane4114, 2016-03-23:
The example provided in the node js docs for setting the mode of UART won't compile. Specifically it states that UART_PARITY_NONE isn't defined.

alext-mkrs, 2016-03-28:
Looks like you're mixing several things in, so let's clarify that. First of all, `m.Uart(0)` (that @xbolshe  suggested) is the same thing as `m.Uart('/dev/ttyMFD1')`, which you've later tried. Using index is more future-proof as device names may change. So just use 0 and you should be good.

Now the second piece - what exactly won't compile, which example? Earlier you've tried the JS example, which doesn't need compilation and doesn't contain `UART_PARITY_NONE` references (there's just 0 used directly).

arane4114, 2016-03-28:
So Ive gotten everything working on my side. Here is what I did:
Initially I installed mraa simply via npm. The example that I linked to in the original location would not run and would produce the error code notes in the original post.
I then uninstalled the copy installed via nam, compiled and installed the pre-reqs to mraa and then mraa itself. That allowed me to open m.Uart('/dev/ttyMFD1'). While setting the parity bit as described in the  [set mode documentation](http://iotdk.intel.com/docs/master/mraa/node/classes/uart.html) to `UART_PARITY_NONE ` node complained that it wasn't defined. I dug through the source and saw that `UART_PARITY_NONE ` had the value 0. Replacing that in the setMode call worked and I can now talk and listen to my device on UART.
Sorry if I wasn't clear with my earlier posts.

alext-mkrs, 2016-03-28:
I see, thanks.

So generally, you should have better used port index (0) instead of the file name (/dev/ttyMFD1), as the latter may change in the future. Initially you have used port index 1, which is not supposed to work as max for Edison is 0.

As far as the `UART_PARITY_NONE` is concerned - in Node.js it would look like `m.UART_PARITY_NONE`, if you used `m` as the variable name when `require`-ing mraa module. 

I.e.
```javascript
m = require('mraa');

console.log('UART_PARITY_NONE is ' + m.UART_PARITY_NONE);
```

Using plain 0 would also work, as you've already discovered, but is also not recommended due to slight chance of this changing as well.

I'm glad you have everything workin now, so I'm closing this one.

closed 2016-03-28 by alext-mkrs

vsivanagulu, 2019-08-19:
I'm also got same issue how to resolve