eclipse-mraa/mraa #470: Error handling

wda2945, 2016-03-20:
Error handling in mraa needs work! Look at this:

`uint8_t
mraa_i2c_read_byte(mraa_i2c_context dev)
{
    if (IS_FUNC_DEFINED(dev, i2c_read_byte_replace))
        return dev->advance_func->i2c_read_byte_replace(dev);
    i2c_smbus_data_t d;
    if (mraa_i2c_smbus_access(dev->fh, I2C_SMBUS_READ, I2C_NOCMD, I2C_SMBUS_BYTE, &d) < 0) {
        syslog(LOG_ERR, "i2c: Failed to write");
        return 0;
    }
    return 0x0FF & d.byte;
}
`

If any error occurs, the call returns 0. If that's a valid register return value, which it may be, there's no way  in the code to know an error's occurred. Someone has to read the system log, looking for messages! How is that going to work? Note that it says "I2C: Failed to write" for a read. Actually, mraa logs that kind of message (with no more details) for various errors.

The solution is obvious: have a return type of int and return -1 for error!
This is just one of many examples.

arfoll, 2016-03-20:
Agreed, some of the types used try to be as simple as possible and sometimes end up being a bit confusing. The plan is to introduce functions that take a pointer for these causes. Use mraa_i2c_read_byte_data if you'd like an error message.

The reason the log always says that we failed to write is that in Linux using i2c-dev we don't know much more than that. It means we failed to write what we expected to the bus (to get an answer). Obviously that could be a bus/device error but as far we're concerned it's a failed write (regardless if you where reading or not).

wda2945, 2016-03-21:
Hi Brendan,
Thanks for the response. 

> 
> Agreed, some of the types used try to be as simple as possible and sometimes end up being a bit confusing. The plan is to introduce functions that take a pointer for these causes. Use mraa_i2c_read_byte_data if you'd like an error message.
It's not just "a bit confusing" it's bad practice. Out-of-band error signaling is fundamental to good programming practice. The libc low-level io primitives, for example, like read and write used a signed type (ssize_t) so they can provide an out of band error return. It's virtually a standard. And who wants to perform an io operation and not get an error message? Or worse, who wants to read syslog periodically to ensure everything has worked!
> The reason the log always says that we failed to write is that in Linux using i2c-dev we don't know much more than that. It means we failed to write what we expected to the bus (to get an answer). [...]
The purpose of an error message is to help a programmer debug his/her application. Your syslog message is the equivalent of "-1". "An anonymous i2c operation has failed." You could tell the poor programmer what i2c address he/she was accessing, which register in said device, and which mraa call it happened in. I need to know whether I was trying to read or write and I don't care if it's always a write "as far as you're concerned"!

Usually, syslog contains extra details about a reported error, to aid in debugging. The error should be reported (back to the caller) so the programmer knows to go hunt for more info, and there should be more info actually there. As you're a low-level io wrapper,  in this case, it would be better if you just returned -1 and left the caller to write a more informative and contextual message to syslog, if appropriate.

Try putting yourself in the shoes of your users and consider how you'd use what you're providing to debug a program.

End of lecture. Have a nice day! 😳

arfoll, 2016-03-21:
I get your point and thanks for reporting on this - after the FIrmata stuff has quietened down a bit I'll queue these changes up. We're actually planning a minor API break for 1.0 so I'll get these fixed. In the meantime, feel free to submit patches in the mraa-ng branch. I'll try get syslog output to be a little more helpful.

More suggestions are more than welcome.

alext-mkrs, 2016-05-09:
Same question as for #488 - do you think we've covered all we wanted to in those commits?

arfoll, 2016-05-09:
Yup

closed 2016-05-09 by arfoll