eclipse-mraa/mraa #474: Edison Miniboard config is missing a uart

wda2945, 2016-03-22:
The edison miniboard provides two uarts at its expansion pins but only one is configured in intel_edison_fab_c.c

It needs something like the following:

    b->uart_dev_count = 2;
....
    b->uart_dev[1].rx = 44;
    b->uart_dev[1].tx = 4;
    b->uart_dev[1].device_path = "/dev/ttyMFD2";

arfoll, 2016-03-22:
possibly, I think there was a reason for us not having that as a default option, for now just use the mraa_uart_init_raw() option. I'll check tommorow on a board why it's not enabled in the config.

alext-mkrs, 2016-03-28:
@arfoll, have you had a chance to check this out or anything I could help with?

arfoll, 2016-03-28:
Yes, basically a getty is running on ttyMFD2 so we don't have it in the platform configuration by default since you'd have to disable the tty first and it's pre-muxed so raw mode works ok. I've added something to docs/edison.md and I think that's bascally all we can do :/

alext-mkrs, 2016-03-28:
I see, indeed there's commit 8b02c419f8e70b1c904ae1b90a921f35f1a4d260, which adds that. I agree that's as much as we can do given the default setup.

Let me close this one then.

closed 2016-03-28 by alext-mkrs