eclipse-mraa/mraa #478: MRAA_ERROR_PLATFORM_ALREADY_INITIALISED

xbolshe, 2016-03-24:
DEBUG: Executing shell function do_compile
| -- The C compiler identification is GNU 4.9.1
| -- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc
| -- Check for working C compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Looking for include file pthread.h
| -- Looking for include file pthread.h - found
| -- Looking for pthread_create
| -- Looking for pthread_create - not found
| -- Looking for pthread_create in pthreads
| -- Looking for pthread_create in pthreads - not found
| -- Looking for pthread_create in pthread
| -- Looking for pthread_create in pthread - found
| -- Found Threads: TRUE
| -- INFO - libmraa Version v0.10.0
| -- INFO - cmake Version 3.3.1
| -- INFO - Target arch is i586
| -- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.8", minimum required is "2.7")
| -- INFO - Package arch is i586
| -- INFO - Adding support for platform
| -- INFO - Adding support for all platforms
| INFO - Adding x86 platforms
| -- The CXX compiler identification is GNU 4.9.1
| -- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++
| -- Check for working CXX compiler: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Found SWIG: /usr/local/bin/swig (found version "3.0.8")
| -- Found PythonLibs: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/python2.7 (found suitable version "2.7.3", minimum required is "2.7")
| -- Found Nodejs: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/scripts/node
| INFO - Node version is 5.9.0
| INFO - Node using v8 4.6.85.31
| -- INFO - Padded V8 version to match SWIG format
| -- INFO - swig Version 3.0.8
| -- INFO - CXX compiler Version GNU 4.9.1
| -- INFO - Using V8 version > 3 so requiring C++11 compiler
| -- INFO - install NODE modules to /usr/lib/node_modules/mraa/
| -- Configuring done
| WARNING: Target "_python-mraa" requests linking to directory "/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/sysroots/quark/usr/lib/python2.7".  Targets may link only to libraries.  CMake is dropping the item.
| -- Generating done
| -- Build files have been written to: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/build
| Scanning dependencies of target mraa
| [  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
| /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/src/mraa.c: In function 'mraa_result_print':
| /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/src/mraa.c:464:9: error: duplicate case value
|          case MRAA_ERROR_PLATFORM_ALREADY_INITIALISED:
|          ^
| /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/src/mraa.c:428:9: error: previously used here
|          case MRAA_SUCCESS:
|          ^
| src/CMakeFiles/mraa.dir/build.make:62: recipe for target 'src/CMakeFiles/mraa.dir/mraa.c.o' failed
| make[2]: *** [src/CMakeFiles/mraa.dir/mraa.c.o] Error 1
| CMakeFiles/Makefile2:93: recipe for target 'src/CMakeFiles/mraa.dir/all' failed
| make[1]: *** [src/CMakeFiles/mraa.dir/all] Error 2
| Makefile:160: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/temp/log.do_compile.3162)

/**
 * MRAA return codes
 */
typedef enum {
    MRAA_SUCCESS = 0,                             /**< Expected response */
    MRAA_ERROR_FEATURE_NOT_IMPLEMENTED = 1,       /**< Feature TODO */
    MRAA_ERROR_FEATURE_NOT_SUPPORTED = 2,         /**< Feature not supported by HW */
    MRAA_ERROR_INVALID_VERBOSITY_LEVEL = 3,       /**< Verbosity level wrong */
    MRAA_ERROR_INVALID_PARAMETER = 4,             /**< Parameter invalid */
    MRAA_ERROR_INVALID_HANDLE = 5,                /**< Handle invalid */
    MRAA_ERROR_NO_RESOURCES = 6,                  /**< No resource of that type avail */
    MRAA_ERROR_INVALID_RESOURCE = 7,              /**< Resource invalid */
    MRAA_ERROR_INVALID_QUEUE_TYPE = 8,            /**< Queue type incorrect */
    MRAA_ERROR_NO_DATA_AVAILABLE = 9,             /**< No data available */
    MRAA_ERROR_INVALID_PLATFORM = 10,             /**< Platform not recognised */
    MRAA_ERROR_PLATFORM_NOT_INITIALISED = 11,     /**< Board information not initialised */
    MRAA_ERROR_PLATFORM_ALREADY_INITIALISED = 0,  /**< Board is already initialised, same as MRAA_SUCESS */

    MRAA_ERROR_UNSPECIFIED = 99 /**< Unknown Error */
} mraa_result_t;

xbolshe, 2016-03-24:
MRAA_SUCCESS and MRAA_ERROR_PLATFORM_ALREADY_INITIALISED have the same value

arfoll, 2016-03-24:
Yeah this is on purpose so mraa_init() stops replying MRAA_ERROR_PLATFORM_ALREADY_INITIALISED which was a pretty stupid error code. Maybe I'm stupid but is this not legit?

I'm surrpised this broke your build, it works on travis/arch/debian/fedora/ubuntu/clang :/ And passed static analysis!

I guess if not we can move MRAA_ERROR_PLATFORM_ALREADY_INITIALISED to be a #define.

xbolshe, 2016-03-24:
C99
http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf

(The use of enumerators with = may produce enumeration constants with values that duplicate other values in the same enumeration.)  The enumerators of an enumeration are also known as its members.

arfoll, 2016-03-24:
what about 02ba5f993fef7cb8c23563feaa55adc0fcf968a8?

xbolshe, 2016-03-24:
The problem is not related with enum, it is related with CASE:

[image]

arfoll, 2016-03-24:
this case statement got removed with 51c60a0ac4c52d0d1032fd17d88636642bd0ea85 in the 0.10.0 release though.

xbolshe, 2016-03-24:
| -- Generating done
| -- Build files have been written to: /sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/build
| Scanning dependencies of target mraa
| [  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
| [  2%] Building C object src/CMakeFiles/mraa.dir/gpio/gpio.c.o
| [  4%] Building C object src/CMakeFiles/mraa.dir/i2c/i2c.c.o
| [  5%] Building C object src/CMakeFiles/mraa.dir/pwm/pwm.c.o
| [  6%] Building C object src/CMakeFiles/mraa.dir/spi/spi.c.o
| [  8%] Building C object src/CMakeFiles/mraa.dir/aio/aio.c.o
| [  9%] Building C object src/CMakeFiles/mraa.dir/uart/uart.c.o
| [ 10%] Building C object src/CMakeFiles/mraa.dir/iio/iio.c.o
| [ 12%] Building C object src/CMakeFiles/mraa.dir/x86/x86.c.o
| [ 13%] Building C object src/CMakeFiles/mraa.dir/x86/intel_galileo_rev_d.c.o
| [ 15%] Building C object src/CMakeFiles/mraa.dir/x86/intel_galileo_rev_g.c.o
| [ 16%] Building C object src/CMakeFiles/mraa.dir/x86/intel_edison_fab_c.c.o

Also have a problem with mraajsJAVASCRIPT_wrap.cxx.o. Now I will check and open new bug if it is.

xbolshe, 2016-03-24:
"this case statement got removed with 51c60a0 in the 0.10.0 release though."

Sure. But it is downloaded with this case from github. It's strange...

arfoll, 2016-03-24:
I can't find it in the release tarball from git or my local clone, you sure you've got a clean checkout? 52f9c48bbc seems to be good but I see you're using AUTOINC so maybe something is not quite right? Maybe a bitbake/devtool cleanall mraa is in order?

arfoll, 2016-03-24:
this is what you should have
```bash
brendan@silver ~/git/mraa (git)-[master] % md5sum src/mraa.c
1a4002c73235d86a9b0f4590c48e244d  src/mraa.c
```

xbolshe, 2016-03-24:
user@user-VirtualBox:/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git/src$ md5sum mraa.c 
2df3d5f5b6b0e22bf000a9a6725f5b5b  mraa.c

52f9c48bbc is a part of https://github.com/intel-iot-devkit/mraa/commit/52f9c48bbc4c63c6c4ec6285760a4e5c67fad4dd

xbolshe, 2016-03-24:
I have done clean/fetch of mraa several times.

xbolshe, 2016-03-24:
user@user-VirtualBox:/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git$ **git show**
commit 52f9c48bbc4c63c6c4ec6285760a4e5c67fad4dd
Author: Brendan Le Foll <brendan.le.foll@intel.com>
Date:   Thu Mar 24 17:01:01 2016 +0000

    mraa: Update to v0.10.0
    
    Signed-off-by: Brendan Le Foll <brendan.le.foll@intel.com>

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b150d2b..6691600 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -25,7 +25,7 @@ include (GetGitRevisionDescription)
 git_describe (VERSION "--tags")
 if ("x_${VERSION}" STREQUAL "x_GIT-NOTFOUND" OR "x_${VERSION}" STREQUAL "x_HEAD-HASH-NOTFOUND" OR "x_${VERSION}" STREQUAL "x_-128-NOTFOUND")
   message (WARNING " - Install git to compile a production libmraa!")
-  set (VERSION "v0.9.6-dirty")
+  set (VERSION "v0.10.0-dirty")
 endif ()
 
 message (STATUS "INFO - libmraa Version ${VERSION}")
diff --git a/docs/changelog.md b/docs/changelog.md
index 828d8ff..ac014cb 100644
commit 52f9c48bbc4c63c6c4ec6285760a4e5c67fad4dd
Author: Brendan Le Foll <brendan.le.foll@intel.com>
Date:   Thu Mar 24 17:01:01 2016 +0000

    mraa: Update to v0.10.0
    
    Signed-off-by: Brendan Le Foll <brendan.le.foll@intel.com>

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b150d2b..6691600 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -25,7 +25,7 @@ include (GetGitRevisionDescription)
 git_describe (VERSION "--tags")
 if ("x_${VERSION}" STREQUAL "x_GIT-NOTFOUND" OR "x_${VERSION}" STREQUAL "x_HEAD-HASH
-NOTFOUND" OR "x_${VERSION}" STREQUAL "x_-128-NOTFOUND")
   message (WARNING " - Install git to compile a production libmraa!")
-  set (VERSION "v0.9.6-dirty")
+  set (VERSION "v0.10.0-dirty")
 endif ()

arfoll, 2016-03-24:
if your md5sum doesn't match 1a4002c73235d86a9b0f4590c48e244d then yo don't have the right source files. However I have no idea what's wrong or how that's possible. In the git checkout can you do a `git describe --tags` and `git rev-parse HEAD`?

arfoll, 2016-03-24:
@tingleby any ideas what's possibly up here?

xbolshe, 2016-03-24:
user@user-VirtualBox:/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git$ git describe --tags
v0.10.0

user@user-VirtualBox:/sde/r1.2.0/iot_1.2.0_kernel_3.19.8/build/tmp/work/i586-poky-linux/libmraa/0.10.0+gitAUTOINC+52f9c48bbc-r0/git$ git rev-parse HEAD
52f9c48bbc4c63c6c4ec6285760a4e5c67fad4dd

arfoll, 2016-03-24:
in build/downloads/git2 can you check the right stuff is in there?

arfoll, 2016-03-24:
are you sure you don't have a patch in SRC_URI?

xbolshe, 2016-03-24:
Ohh.. I have a patch.

arfoll, 2016-03-24:
phew! I thought I was going crazy &| git/github was breaking!

xbolshe, 2016-03-24:
I confirm that a compilation is correct. That was my fault.

closed 2016-03-24 by xbolshe

arfoll, 2016-03-24:
No problem, tbh the C99 thing is still very valid, what do you think of the changes in the branch c99?

xbolshe, 2016-03-24:
May be it is better to add mraa_deinit() (or imraa_deinit - why not?) to imraa_init() and do not use MRAA_ERROR_PLATFORM_ALREADY_INITIALISED at imraa_init(). In this case imraa_xxx performs a real init/deinit, and mraa_xxxx is just a wrapper for speed up. I guess that MRAA_ERROR_PLATFORM_ALREADY_INITIALISED with the same value as MRAA_SUCCESS is useless.And it is better do not use __STDC_VERSION__ for compatibility reasons.

xbolshe, 2016-03-24:
Like this:

      if (plat != NULL) {
            mraa_deinit();
     }

arfoll, 2016-03-24:
So {imraa,mraa}_init() returning MRAA_ERROR_PLATFORM_ALREADY_INITIALISED was a merge mistake on my part that shouldn't have gotten in. imraa or mraa should return MRAA_SUCCESS reguardless of whether a platform was actually found, it just means structures initialised without error. imraa_init()'s plat != NULL should return an error and mraa_init()'s should return MRAA_SUCCESS.

imraa_init() is meant to be called from imraa only it's not part of mraa's public API, so the check really is redundant but there just to make sure imraa doesn't do something stupid. mraa_init/mraa_deinit are really the ones that could/should be called by users.

So I'm preparing mraa's 1.0, let's completely remove MRAA_ERROR_PLATFORM_ALREADY_INITIALISED in that release. I'll use mraa-ng to stage changes like this.