eclipse-mraa/mraa #479: Add Support for the Artik Board

thoughtentity, 2016-03-27:
Any plans on adding the [Artik](https://www.artik.io) as one of the supported boards for the mraa library? The Artik team recently released a [commercially available](https://www.artik.io/blog/2016/whats-new-in-artik) version of their board and have [documented](https://developer.artik.io/documentation/developer-guide/kernel-gpio.html) GPIO interactions with the board through both sysfs and through the kernel.

I can help with hardware testing the library with the board as I got an Artik 5 board on hand if needed.

arfoll, 2016-03-27:
There are no plans AFAIK but if you have a board and are willing to do a
bit of research I imagine it wouldn't be hard. I'm happy to help out and
give you a hand with this if you'd like when I get back from holiday in 2
weeks.

thoughtentity, 2016-03-27:
I can try taking a crack at this. Will probably start work on this sometime in April. Open to taking a pull request on this once the work is done?

arfoll, 2016-03-27:
Yes definitely. I'm more than happy to merge something like this.

BinduRao2018, 2018-10-12:
Hi,
I am not going to use Artik board shortly and would like to use mraa to drive my Grove PI sensors. Is there any information on the above mentioned activity please?

BinduRao2018, 2018-10-16:
what is the latest plan to support for ARTIK platform? Please comment.

Propanu, 2018-10-17:
Hi @BinduRao2018, adding support for Artik boards is not on the roadmap for the MRAA team and I doubt any of the current project maintainers have the board, but we would be happy to review and merge any community submitted implementations.

BinduRao2018, 2018-10-18:
Ok, Thanks @Propanu for the clarification. I am in the process of analyzing pin mapping for Eagleye 530s as there is a requirement for us to support on Artik board. I will update/ might seek clarification in this regard on a separate thread.