eclipse-mraa/mraa #481: Build is failing for 0.10.0 using npm

thiagogcm, 2016-03-28:
CC(target) Release/obj.target/mraa/src/mraa.o
../src/mraa.c:46:25: fatal error: json-c/json.h: No such file or directory
 #include <json-c/json.h>
                         ^
compilation terminated.
mraa.target.mk:128: recipe for target 'Release/obj.target/mraa/src/mraa.o' failed
make: *** [Release/obj.target/mraa/src/mraa.o] Error 1
make: Leaving directory '/home/root/node_modules/mraa/build'

arfoll, 2016-03-28:
You need libudev-dev and libjson-c-dev on your system (or similar named dev
libs). This is an unfortunate consequence of not being able to pass params
easily to npm so we build everything.

alext-mkrs, 2016-03-28:
I think this was documented in commit 518c780b15a77887aa158ed94d570f72294e51a9, though I'm not sure why is this showing up in the NPM build - does it build with imraa enabled by default?

alext-mkrs, 2016-03-28:
Looks like I'm facing this as well (on Edison, to be clear). I have just installed the 0.10.0 from opkg repo and as long as I have Node.js 0.12.7, I'm reinstalling the node module from npm. I have installed `json-c-dev` package (otherwise I faced the same error @thiagogcm reported).

With that package installed npm builds the module just fine, however upon trying to use it I get the following:
```
root@edison1:~/mraa-json-npm# node
> m=require('mraa');
Error: /home/root/mraa-json-npm/node_modules/mraa/build/Release/mraa.node: undefined symbol: json_tokener_parse
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at repl:1:3
    at REPLServer.defaultEval (repl.js:132:27)
    at bound (domain.js:254:14)
    at REPLServer.runBound [as eval] (domain.js:267:12)
    at REPLServer.<anonymous> (repl.js:279:12)
>
```

Looking further at the `mraa.node` dependencies I see no `libjson{-c}` dependency listed. What's even more interesting, I don't see dependency on `libmraa` either, which should be there according to recent changes (and that's present in `mraa.node` shipped with the IPK):
```bash
# This is what NPM has built for me
root@edison1:~/mraa-json-npm# ldd /home/root/mraa-json-npm/node_modules/mraa/build/Release/mraa.node
	linux-gate.so.1 (0xb77ba000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xb7661000)
	libm.so.6 => /lib/libm.so.6 (0xb7610000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb75fc000)
	libpthread.so.0 => /lib/libpthread.so.0 (0xb75e1000)
	libc.so.6 => /lib/libc.so.6 (0xb7479000)
	/lib/ld-linux.so.2 (0x4e541000)

# This is what mraa 0.10.0 IPK has
root@edison1:~/mraa-json-npm# ldd /usr/lib/node_modules/mraa/mraa.node
ldd: warning: you do not have execution permission for `/usr/lib/node_modules/mraa/mraa.node'
	linux-gate.so.1 (0xb77a8000)
	libmraa.so.0 => /usr/lib/libmraa.so.0 (0xb7739000)
	libpthread.so.0 => /lib/libpthread.so.0 (0xb771e000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xb7631000)
	libm.so.6 => /lib/libm.so.6 (0xb75e0000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb75cb000)
	libc.so.6 => /lib/libc.so.6 (0xb7464000)
	/lib/ld-linux.so.2 (0x4e541000)
```

Moreover, `libmraa.so` in the IPK is not linked to `libjson-c.so` either. Looks like the IPK's version is built with `-DIMRAA=OFF`.

So looks like we have three things here (in addition to a need to have `json-c-dev` and `systemd-dev` packages, on Edison `udev` is coming out of `systemd` recipe):

1. `libmraa.so` in the IPK is [seemingly] not built with `-DIMRAA=ON`;
1. When built through NPM, `mraa.node` is not linked to `libmraa.so`, contrary to `mraa.node` within the IPK;
1. When built through NPM, `mraa.node` is not linked to `libjson-c.so` either, so effectively the NPM module as of 0.10.0 release would build, but will not work due to that unresolved symbol.

I'm looking into what could be done here - looks like we need to adjust `binding.gyp`, as the fastest way - but a clarification on whether an NPM-built `mraa.node` should be linked to `libmraa.so` (new approach) or not (old approach) is IMHO in order. @arfoll, @whbruce, comments are welcome.

alext-mkrs, 2016-03-28:
So in essence looks like that recent change with `mraa.node` now dynamically linking to `libmraa.so` was not propagated to `binding.gyp`. That can be solved with adding a
```json
'ldflags': [ '-lmraa' ],
```
section and IMHO that would be the most logical solution so that both IPK and NPM `mraa.node` versions use the same approach.

However due to the fact that IPK version of the `libmraa.so` is apparently built with `-DIMRAA=OFF -DFIRMATA=OFF`, while in `binding.gyp` both are enabled, just the above is not enough - we'd need to build IPK with imraa and firmata enabled too. I would vote for this approach.

@arfoll, what do you think?

arfoll, 2016-03-28:
Arg yes so I made a mistake. To build with IMRAA we don't just have to have the headers we obviously need the linker flags....

So the idea is that npm mraa builds libmraa.so for itself and doesn't link against it to reduce deps/issues with old mraa from ipks lying around.

@tingleby can you help?

alext-mkrs, 2016-03-28:
I see, that's another option I've checked out (`'ldflags': [ '-ljson-c' ]` instead), which seemed to work fine as well, at least on the "require->getVersion()" level.

Let me test it a bit more and then submit a PR. If Tom has any corrections to that, I'm all ears.

tingleby, 2016-03-28:
I think for now, npm will just not have imraa included. We will have a small update for this so it gets out onto NPM safely. @arfoll right?

arfoll, 2016-03-28:
@alext-mkrs  that would work but really I'm worried about node people getting confused (just as this thread started) having another -dev, @tingleby is going to do a quick 0.10.1 release removing imraa support from npm that will be easiest/safest and when I get back from vacation I'll re-evaluate :)

@tingleby yup that's the plan

alext-mkrs, 2016-03-28:
Makes sense to me! :smiley:

tingleby, 2016-03-28:
OK so faa8d4d3d41fac62a963304bcbfb916e87ebd2dd should fix this. I also tagged a patch release. (v0.10.1)
I've tested it and I managed to pull from NPM and ran the module fine.

closed 2016-03-28 by tingleby

alext-mkrs, 2016-03-28:
Works fine for me as well now, so @thiagogcm, give it a try.

thiagogcm, 2016-04-02:
Sorry for the late reply. That solved the problem. Thanks!!

alext-mkrs, 2016-04-04:
No worries, thanks for confirmation