eclipse-mraa/mraa #485: Yocto build fails with JavaScript enabled

whbruce, 2016-03-28:
I am adding mraa to a Yocto build using recipes from intel-iot-middleware layer. This has been working for me (with a bbapend to support builds later than 0.9.6), but with version 0.10.1 I get the following error in do_package_qa
`
ERROR: QA Issue: package mraa contains bad RPATH /path/to/build/tmp/work/core2-32-poky-linux/mraa/0.10.1-r1/build/src in file /path/to/build/tmp/work/core2-32-poky-linux/mraa/0.10.1-r1/packages-split/mraa/usr/lib/node_modules/mraa/mraa.node [rpaths]
`

I can see why this is happening by looking at the log.do_compile. You can see that I am targeting Edison, but I would expect the same behavior with any target.
`/path/to/build/tmp/sysroots/x86_64-linux/usr/bin/i586-poky-linux/i586-poky-linux-g++  -fPIC  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse  --sysroot=/path/to/build/tmp/sysroots/edison  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -shared  -o mraa.node CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o ../libmraa.so.0.10.1 -lpthread -ldl -Wl,-rpath,/path/to/build/tmp/work/core2-32-poky-linux/mraa/0.10.1-r1/build/src 
`

Swig and/or cmake is somehow causing insertion of an RPATH. I don't want to skip QA tests. Does anyone know how to fix this cleanly?

xbolshe, 2016-03-28:
I use -DCMAKE_SKIP_RPATH=TRUE to fix this problem.

xbolshe, 2016-03-28:
https://cmake.org/cmake/help/v3.0/variable/CMAKE_SKIP_RPATH.html

Propanu, 2016-03-28:
Opened #486 to fix this.

whbruce, 2016-03-28:
Thanks @xbolshe. @Propanu's PR uses set_target_properties to set CMAKE_SKIP_RPATH=TRUE. I did a manual fix and it resolved build failure for me.

Propanu, 2016-03-31:
Turns out skipping RPATHs entirely will break linking for the Python bindings. Will have to find a better solution.

petreeftime, 2016-04-06:
Python package is installed in /usr/lib/python2.7/site-packages/ instead of /usr/lib64/python2.7/site-packages/ for me, might be related to this issue.

xbolshe, 2016-04-06:
@Propanu It is strange. I have no problem with Node nor Python with Galileo and Edison boards.
What recipes do you use? What kind of linking problem do you have? What board do you use?

Propanu, 2016-04-26:
@xbolshe with RPATHs off and a custom install dir that is not on LD's path, you'll get a linker error when you try to load MRAA from Node or Python. This is also caused by the fact we went from static to dynamic with our loading our bindings to avoid multiple init calls. What's interesting though is that RPATHs are supposed to be off by default but Yocto builds clearly say otherwise. It does help to have an RPATH added to the binaries, when installing in a non-system folder, but instead of adding the build source path (which is temporary for Yocto builds) we want to add the install path and location of the .so. Thus I think one of the other options [in here](https://cmake.org/Wiki/CMake_RPATH_handling#Default_RPATH_settings) will do the trick.

sandeep-mvd, 2016-11-18:
I see the same error when trying to update mraa to 1.5.1
I am working on edison as well.
@Propanu The issue you opened #486 is closed. In 1.5.1 mraa both the patches are present. But still I see this error.
Is there any working workaround?

arfoll, 2016-11-18:
Try applying this patch 7cade840122cb1a45d7aa178a7ad80d4174ae0bf

arfoll, 2016-11-18:
new patch - d6f53f92cedcc6d5b240bedebe8f2766c0f06679

sandeep-mvd, 2016-11-22:
Thank you @arfoll 
It solves the problem.

alext-mkrs, 2016-11-23:
This one seems to be resolved now, @arfoll, @whbruce, do you think it can be closed?

arfoll, 2016-11-23:
not yet, patch is in a branch, I need to merge it into master.

arfoll, 2016-11-23:
Merged, thanks!

closed 2016-11-23 by arfoll