eclipse-mraa/mraa #488: Size checking in pwm.c

wda2945, 2016-03-30:
mraa_pwm_read_period() and mraa_pwm_read_duty() gave me "ppm: Error in reading..." syslogs and a -1 reply.

The error was EFAULT ("Bad Address") [I added strerror() to the message.]

For some reason, the preceding lseek(SEEK_END) in both cases returned 4096! **Without checking**, this was passed as length to read() together with a buffer of size 64 bytes! Happily it returned an EFAULT instead of stamping over the heap!

Changing the read() length to MAX_SIZE fixed it.

```
off_t size = lseek(period_f, 0, SEEK_END);
lseek(period_f, 0, SEEK_SET);
ssize_t rb = read(period_f, output, size + 1);
```

arfoll, 2016-03-30:
If you send a pull request I can merge it or just submit the change. What would you rather?

wda2945, 2016-03-30:
Well….

I’m working with my own branch so I can make whatever changes I fancy. 😏 And I can debug the library and fix things as I go.
For example, I changed the syslog messages, and made these functions non-static.

In this case, you need to decide whether to check size, or use MAX_SIZE. It might affect other platforms which I don’t care about.
So I’d rather report the bug and let you decide how to fix it.

If I started down the fixing-mraa-to-suit-me path, who knows where it would end?

Martin

arfoll, 2016-03-30:
That's fair enough, I'm just very interested in understanding what you think doesn't work - for example why should these functions be non static? afaik they're internal to mraa and not part of the public API. Are you finding them useful?

The syslog messages I've been thinking of overhauling for a while but tbh am stuck between a few ideas and not much time to do it - what do you think would be more useful?

Anyways I'll put mraa_pwm_read_duty on my list of functions to fix, tbh the pwm code is not nice, I'll put it on my list of stuff to fix.

wda2945, 2016-03-30:
Hi Brendan,

Maintaining software sucks. 
(If I appear opinionated, I should tell you that I’m a retired VP Engineering and wrote my first code in 1967! 😱)

The key problem I’ve had with mraa from the start is that it's a ‘details hiding' wrapper that I can’t use without reading the source code! 

I can’t use it as a shared library because I need to debug and experiment (printfs?) to figure out what’s going on. So I build it from source into my app.

My first attempt to use it last year failed because I had inadvertently put an MCP23008 I2C GPIO at its default slave address. libmraa crashed on loading due presumably to bad Miniboard identification. This was irritating because my app knew what platform it was running on!

My experience is that error handling takes most of the development work and this is the biggest problem area I see.

I first worked with i2c.I had mis-connected an ADC chip, but the read() failed with a MRAA_ERROR_INVALID_HANDLE return! (And a "Failed to write”).

I looked at the source and found that errors could give me:
	Null pointer		OK
	0 (in band) 		ugh!
	mraa_result_t		OK (Most often it’s MRAA_ERROR_INVALID_HANDLE)
	-1				?
	Many, but not all errors get an opaque syslog message. 
	Some get a syslog message but an OK return! (A bad i2c bus number gets your commands sent to where your peripherals aren’t!)

Back to your first question. When my pwm (servo) thread didn’t work, I started testing with the pwm example code. It works as supplied but pwm output quits when I increase the pwm period to more than a couple of mS. I used read_period and read_duty to see what was actually written to the device. read_duty is not static BTW, but read_period currently is. Somewhere in this, the read errors started, and I got diverted.

I suspect an OS problem (when the pwm output stops, the Edison crashes!) so I’m updating from 159.sdk to 159 to see if anything changes.

End of another lecture. I told you I was opinionated! 🤓

Martin

alext-mkrs, 2016-05-09:
As long as @wda2945's pull request with error checking enhancements was merged a while ago, do you think we can close this one, @arfoll, @wda2945?

arfoll, 2016-05-09:
Yes this one can be closed.

closed 2016-05-09 by arfoll