eclipse-mraa/mraa #492: Mraa ISR time-out in Python

BTWS, 2016-04-13:
Is similar functionality possible like [pulseIn on Arduino](https://www.arduino.cc/en/Reference/PulseIn)? 

`pulseIn(pin, value, timeout)`

I wan't to use an interrupt but with a **time-out** of a given amount, let's say 0.5 seconds.

Relevant:
[gpio-interrupt-isr](http://iotdk.intel.com/docs/master/mraa/python/example.html#gpio-interrupt-isr)
[hello_isr.py](https://github.com/intel-iot-devkit/mraa/blob/master/examples/python/hello_isr.py)

arfoll, 2016-04-13:
No there's nothing like this, the problem is that we're on Linux so you simply can't make assumptions about timings this way.

alext-mkrs, 2016-05-09:
As long as @arfoll has provided an answer, I'm closing this one

closed 2016-05-09 by alext-mkrs