eclipse-mraa/mraa #499: latest mraa for nodejs (0.10.1) on Edison behaves differently than preinstalled one (0.9.5) for Yocto 1.7.3

elvin-nnov, 2016-04-28:
I installed the latest firmware for Edison, it is Poky (Yocto Project Reference Distro) 1.7.3. It contains preinstalled binary mraa.node in the /usr/lib/node_modules/mraa/mraa.node. its version is "version": "v0.9.5-dirty".
Using XDK for IoT I created the sample "Analog read" and this combination works well.

1 var mraa = require('mraa'); //require mraa
2 var analogPin4 = new mraa.Aio(4); //setup access analog input Analog pin 4 (A4)
3 setInterval(function() {
4    var analogValue = analogPin4.read(); //read the value of the analog pin
5    console.log(analogValue); //write the value of the analog pin to the console    
6 }, 16);

then I logged to the Edison, went to the ~/.node_app_slot and installed mraa manually using "npm install mraa" command. The same will happen if we add mraa to the package.json of XDK for IoT project adding this part:
  "dependencies": {
    "mraa" : "latest"
  }
mraa version will be "version": "0.10.1

Thus, running the above sample we use just downloaded mraa, but not preinstalled one.

And in this case above sample starts work wrong. It ignores values from analog pin #4, instead of this line #4 will return value taken from A0. All the time.

And the question is: is there special version strategy which will make Yocto 1.7.3 working correctly only with mraa 0.9.5 or is it a bug in latest mraa version?

BTW, if I install manually 0.9.5 to the ~/.node_app_slot, everything works well

arfoll, 2016-04-28:
There's a bug in AIO 0.10.0+ releases fixed with b258867c376526bc9e88f931be9bfaf8429c7958. It'll be fixed with the incoming 1.0.0 release.

elvin-nnov, 2016-04-28:
@arfoll Thank you for the information!

closed 2016-04-28 by elvin-nnov