eclipse-mraa/mraa #501: Difficult to use NodeJS with mraa and upm

luigisaetta, 2016-04-29:
The version of NodeJS shipped with Edison is 0.10. This is not really a stable version and not usable for serious things.
The upgrade to a more recent and stable version (even 0.12.7) is not easy and then you enter in a rather complicated voyage to have mraa and upm working with Node.
The best solution would be definitively ship with the Image at least a Node4 (LTS).
In the meantime, you should provide an end-to-end documentation on how to create a Node environment where you can do serous developments and test.

arfoll, 2016-04-29:
You're preaching to the choir (and tbh the wrong place communities.intel.com
is the right place)

I believe @alext's repo has 0.12.x pkgs. Until then you'll have to compile
your own node version. Sorry!

luigisaetta, 2016-04-29:
Well, I have already posted on communities.intel.com, and there the suggestion was: "open an issue on github".

xbolshe, 2016-04-30:
Looking at https://communities.intel.com/thread/101617 need to differentiate the issues:
1. issue with old NodeJS version
2. issue with MRAA
3. issue with UPM
4. issue with MQTT

Here is a place to rise MRAA components' issues only.
Questions with NodeJS version issue in the Edison image need to address on communities.intel.com.
Information how to compile NodeJS you may find on https://nodejs.org

alext-mkrs, 2016-05-09:
@luigisaetta, do you want to pursue mraa-related pieces further or should we close this one?

luigisaetta, 2016-05-09:
OK, you can close.

closed 2016-05-09 by arfoll