eclipse-mraa/mraa #502: After upgrading to mraa 1.0 node fails to locate libmraa.so.0

ramonmata, 2016-05-01:
Hello,
I just upgraded my Intel Edison to the 3.0 Yocto release, so then upgraded all upgradeable opkg packages to the latest.

Then when trying the example for sensor BISS0001, node complains about a missing libmraa.so.0 file

**libmraa.so.0: cannot open shared object file: No such file or directory**

```
Intel XDK - IoT App Daemon v0.0.38 - commands: run, list, debug, status
Upload on board is not from current project - uploading complete project
No NPM modules found.
transferring /var/folders/m9/6wtq5xqx2rqgs46rq03_g_mh0000gn/T/xdk-9708nad0c8g.tar from project to board
x README.md
x icon.png
x main.js
x package.json
x xdk/project-info.json
Upload Complete

ERROR: module.js:356
ERROR:   Module._extensions[extension](this, filename);
   ERROR:                             ^
ERROR: Error: libmraa.so.0: cannot open shared object file: No such file or directory
```

I was able to solve this problem by creating a symbolic link with that name in `/usr/lib` pointing to `libmraa.so.1` but perhaps this is a issue that needs to be solved

best regards!

arfoll, 2016-05-01:
We just need a upm release to fix. @propanu?

hellofunk, 2016-05-03:
I am having this identical problem. I'm a bit of a linux newbie, can you advise on the actual steps you took at the edison ssh prompt to resolve this?

ramonmata, 2016-05-03:
Try this:

```
cd /usr/lib
ln -s libmraa.so.1 libmraa.so.0
```

regards

arfoll, 2016-05-03:
Thanks @ramonmata. Note that there may be some issues due to the versioning/API change in some sensors though I expect the effects to be very limited.

alext-mkrs, 2016-05-09:
it's not very clear from the error message, so @arfoll, you think that's mraa Node binding problem or UPM problem, to make sure? In view of your ping for @propanu, I'd assume the latter, but maybe mraa releases are now not done by you?

arfoll, 2016-05-09:
Mraa releases are done by me, this API change was done on purpose since mraa's API changed slightly with 1.0.0. The next version of UPM will fix this since they will link to libmraa.so.1.

Propanu, 2016-05-09:
Yes, a UPM update will fix this issue, we're working on it and 0.7.0 will be out soon.

alext-mkrs, 2016-05-09:
Understood, thanks.

deadprogram, 2016-06-24:
Hi, @Propanu seems like this problem still exists, even after the upm 0.7.0 release.

deadprogram, 2016-06-24:
To elaborate somewhat: the npm package update path is still broken in a similar way to what is descibed in this issue. Trying out the opkg builds seem to indicate they require a new version of Node than is installed by default on Yocto 3.0.

ramonmata, 2016-06-24:
Well I have a distinct result

- I have installed the latest Yocto release 3.5 doing the manual install under Linux Mint, (_I had some trouble with the GUI in Windows and Linux Mint_)
- Then went to XDK and uploaded the same program

And works for me!

```
root@edison:~# uname -ar
Linux edison 3.10.98-poky-edison+ #1 SMP PREEMPT Mon Jun 6 14:32:08 PDT 2016 i686 GNU/Linux
```

deadprogram, 2016-06-24:
Yocto 3.5 has node 4.4 installed as default, which appears to be the new target version. I also confirm the default install working as expected.

Propanu, 2016-08-04:
Just to clarify:
 * IoTDK 3.0 comes with Node.js 0.10 and MRAA + UPM are built against it.
 * IoTDK 3.5 ships with Node.js 4.4 and MRAA + UPM will work with this newer version instead.

In either case, the latest packages in the `intelgalactic` repos (both 3.0 and 3.5) will not have this issue anymore. IMO this can be closed.

tingleby, 2016-08-04:
Let us know if this crops up again!
Seems to be closed. Closing

closed 2016-08-04 by tingleby