eclipse-mraa/mraa #520: Support to the old fashion GPIOs pin class

bmeneguele, 2016-06-02:
Hello everyone, I would like to ask you guys about a modification that I think would be a good idea. Instead of using hard coded file paths to GPIOs, would be useful some field into the mraa_gpio_conext_t structure to store the sysfs pin name format, because some boards or even old kernel version create other file paths to the gpio class.

For instance, right now I am trying to port libmraa to cubietruck, which has lots of users using the old 3.4.x kernel version and the GPIO sysfs class creates the paths following the format /sys/class/gpio/gpio<pin_number>_<pin_name>, where <pin_number> is the number specified at the .fex file passed by the bootloader to the kernel and <pin_name> is the name specified by the processor's datasheet like PB20, PA03, in lower case. 

Currently, libmraa supports and handles only /sys/class/gpio/gpio<pin_number> format, which is true just for the new kernel version with device trees passed by the bootloader to the kernel. 

Other classes won't be a problem because they are all mapped at /dev/{i2c, spi, ...}.
What do you guys think about this idea of a new field at mraa_gpio_context_t (opaque pointer to struct _gpio)?

arfoll, 2016-06-02:
So first off, yeah new platform! And thanks for getting in touch.

So we store SYSFS_CLASS_GPIO which it sounds like would be fine for cubiboard too (if needs be we could make that a platform thing rather than a hardcoded thing in gpio.c), why don't we simply store (as well as _gpio->pin) _gpio->gpio_path which would be like `gpio256` or in your case `gpio_PB20`. You'll still need a dev->pin equivalent for mraa_gpio_get_pin_raw you can just use -1. To keep compatibility I'd suggest that if _gpio->gpio_path is NULL then you use _gpio->pin just as before. Or if there is a valid _gpio->pin something could just go fill in all the _gpio->gpio_path on platform init.

Does that sound ok?

bmeneguele, 2016-06-02:
Yes, it sounds ok to me. I'll implement this new _gpio->gpio_path and the compatibility part and send a pull request before send all cubietruck specific stuff. As I can see, everything else will remain the same, SYSFS_CLASS_GPIO is the same too (I think and hope this is always the same to all platforms, at least to those existent so far).

But don't close this issue yet, any new question or problem I'll come back to ask =).

bmeneguele, 2016-06-03:
There it is, pull request #522 refers to this issue. I already have the initial support to Cubietruck ready, but for now I have tested just GPIOs on old kernel version, when I finish to test it on newer ones I'll open a new pull request.

Thank you very much @arfoll .

bmeneguele, 2016-06-06:
@alext-mkrs you said on PR #522 about the gpios test cases, which should be modified to handle the old pin sysfs path schema as I added. But I was looking at it and found a problem: the old kernel schema uses the pin name itself, like 'PA20, PB3 ,...' , and there isn't anyway to get this name through python api. 

One possible solution would be to create a getPath() method to get the 'gpio_path' attribute from the mraa_gpio_context_t instance inside GPIO class, which is private. I created this attribute to abstract both new and old pin paths and in this way could be used around all test cases. What do you think?

alext-mkrs, 2016-06-07:
I think that coul be helpful to create such a method, yes. Or rework getPin() to support new mode, however that would change its return type, so it would be a more invasive change and therefore not worth it.