eclipse-mraa/mraa #530: i2c intel Edison

ruscoff, 2016-06-23:
Hi all, 
  i'm using i2c bus 1 on intel edison miniboard version 1.0.0.

Log report very often this message:
i2c1: write_byte: Access error: Connection timed out

i can't understand why

Thanks

Marco

alext-mkrs, 2016-06-23:
Could you please describe your setup and application in more detail, to understand the context better? Like what are you communicating with, what does yor app do and do on.

ruscoff, 2016-06-24:
It's communicating with three ADC (MPC3427) over same i2c bus (bus 1). Level translation is made with TCA9406 from 1.8V to 5V.  However i see same behavior with NMOS BS138 as level translator like in intel arduino board. 

I can read from ADC but often it shows i2c write error.

This is the function that use i2c writeByte.

`
void ADCComm::writeConfiguration(uint8_t address, uint8_t data)
{ // I2C write in ADC configuration to start conversion and select channel  

	if(!commChannel)
		return;

	commChannel->address(address);
	commChannel->writeByte(data);

}`

It is call from thread with this pseudocode:

WriteConfiguration():
usleep(1000);
while(!isAcquisitionStart)
{
   ReadADCStatus();
    usleep(1000);
}
usleep(70000); //AcquisitionTime
While(!isAcquisitionFinished)
{
    ReadADCStatus();
    usleep(2000);
}
ReadADCResult();

ruscoff, 2016-06-24:
Maybe this image can help. I'm observing i2c scl (clock) with oscilloscope. 10x probe. These measures are take with no chip connected only pullup resistor (10Kohm).

When error start seems kernl going nuts, wrong clock frequency 4Khz:

[image]

And strange waiting times:
[image]
[image]

alext-mkrs, 2016-06-24:
Thanks. That looks like a HW thing though... You mentioned these were taken with only pullup connected, but the error was present - do you mean that you are able to reproduce this without any specific sensors connected? WHat would be the shortest reproducer program then?

ruscoff, 2016-06-24:
I think simple write and read over i2c. By the way if i power on and then power off the board the problem disappears and clock is exactly 100khz as expected.  At the moment i'm trying to find when error appear.  

I'm testing with no chip connected and seems frequency is ok also on high processor (60°C) temperature, log report:  "i2c1: write_byte: Access error: Remote I/O error" but i think because there aren't any answer on bus.

After many days maybe i found the solutions, setting SDA and SCL pullstrenght to 2k instead of 20k seems  resolve. Putting 20k shows clock frequency error, changing to 2k realign it.

ruscoff, 2016-06-24:
By the way is so strange because 10k is enough for i2c standard for very short trace and no chip connected.

alext-mkrs, 2016-06-26:
> After many days maybe i found the solutions, setting SDA and SCL pullstrenght to 2k instead of 20k seems resolve. Putting 20k shows clock frequency error, changing to 2k realign it.

So this indeed is a hardware thing then. Mraa, being a software library, cannot help (or break) this one, so such issues are better addressed in the Intel community forums for Edison.

closed 2016-06-26 by alext-mkrs

ruscoff, 2016-06-27:
I'm agree with you is hardware problem, but i hoped that you could help me to know what error strings tells

alext-mkrs, 2016-06-28:
You mean that "connection timed out" one? That comes from the Linux kernel/i2c device driver directly, as mraa generally just does an ioctl() call after preparing data to write/read.