eclipse-mraa/mraa #534: who generates version.c

ouki-wang, 2016-06-27:
my version.c is incorrect. I don't want to chang it manully because other library rely on it
the project base on  commit V1.1.0  (a54386c5d329d01c5b15c4cf353d848d78a62adc)
version.c
#include "version.h"

const char* gVERSION = "show";
const char* gVERSION_SHORT = "show.show.show";

Who can help me solve it。

ouki-wang, 2016-06-27:
cmake generates version.c

closed 2016-06-27 by ouki-wang

arfoll, 2016-06-27:
Indeed, just remove the git cmake module here: https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L30 and then change the version by setting VERSION here: https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L34. I'd highly recommend adding a tag like `-verydirty` to make sure you remember what you've done!