eclipse-mraa/mraa #544: onboard blink example fails

anupam19, 2016-07-18:
I tried the onboard blink example written in C. But I didn't see anything. I am using Edison with Arduino breakout board. But I am unable to lit anything. Even tried with other pins but same issue. With Arduino code it works. But not from linux. Library is installed properly. Tried many community suggestion but nothing worked so far. As far as I am getting hardware is correct as I am able to test it via Arduino code. Please help

Hbrinj, 2016-07-18:
Hi there, what version of mraa are you using? This can be found by running mraa and calling the getVersion function.

Hbrinj, 2016-07-18:
And can you also provide me with the code you're attempting to run.

alext-mkrs, 2016-07-18:
@Hbrinj it looks like he's referring to [blink_onboard example](https://github.com/intel-iot-devkit/mraa/blob/master/examples/blink_onboard.c).

@anupam19, please post the output of the program as well as your compilation command line. I've just tried it on my Edison with Release 3.0 image and latest mraa, worked fine - see compilation command and the example output below.

```bash
root@edison1:~# gcc -o blink_onboard `pkg-config --cflags --libs mraa` blink_onboard.c 
root@edison1:~# ./blink_onboard 
Welcome to libmraa
 Version: v1.1.2
 Running on Intel Edison
^C
root@edison1:~#
```

Hbrinj, 2016-07-18:
@alext-mkrs Ah, yes sorry i missed that, thanks for testing it on a board too, i haven't got mine at home.

anupam19, 2016-07-19:
It worked after a reboot. My edison was running for 3 days continuously. Although everything was working there while these 3 days. I will check it in next wk running edison 3 days continuously and see if the bug returns.

I used : ```cc blink_onboard.c -lmraa -o blink_onboard for compilation```
Output : 
```root@edison:~# ./blink_onboard 
Welcome to libmraa
 Version: v1.1.2
 Running on Intel Edison
^C
root@edison:~#```

alext-mkrs, 2016-07-19:
Check the journalctl output if it fails, there might be some clue.

alext-mkrs, 2016-07-19:
Actually, @anupam19  could you please post the output of the "journalctl -xn 200" command? Please just don't paste it directly, use pastebin.com or similar service.

alext-mkrs, 2016-07-24:
@anupam19, is this still actual?

anupam19, 2016-07-24:
@alext-mkrs 
I will do it tomorrow. I had put my edison for the same condition as I said earlier. I just want to see if the bug is a reproducible one or not. I will check it tomorrow and will post the result.

alext-mkrs, 2016-07-31:
@anupam19, so how is it going?

anupam19, 2016-07-31:
Sorry guys for the delay. After 1 wk of keeping it on, I just checked it today and the LED is working fine with mraa. However WiFi glitches remains same. ie. even after a day WiFi gets disconnected and never gets reconnected unless manually interface is resetted. However for mraa, till date working fine. As I said earlier, after the reboot it worked.

alext-mkrs, 2016-08-01:
Ok, thanks for the information. This confirms my understanding that mraa is unlikely to be the reason. As for the wi-fi, I suggest you to start a thread on that in Intel's Maker community forums,  where tech support folks will help you troubleshoot it.

closed 2016-08-01 by alext-mkrs

anupam19, 2016-09-16:
Again I faced the same problem. This time restart didn't work. However, I have the journalctl log with me 
```
Sep 16 07:09:30 edison libmraa[319]: libmraa version v1.3.0 initialised by user 'root' with EUID 0
Sep 16 07:09:30 edison libmraa[319]: gpio214: init: Failed to write to 'export': No such device
Sep 16 07:09:30 edison libmraa[319]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Sep 16 07:09:30 edison libmraa[319]: libmraa initialised for platform 'Intel Edison' of type 2
Sep 16 07:09:30 edison libmraa[331]: libmraa version v1.3.0 initialised by user 'root' with EUID 0
Sep 16 07:09:30 edison libmraa[331]: gpio214: init: Failed to write to 'export': No such device
Sep 16 07:09:30 edison libmraa[331]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Sep 16 07:09:30 edison libmraa[331]: libmraa initialised for platform 'Intel Edison' of type 2
Sep 16 07:09:30 edison libmraa[337]: libmraa version v1.3.0 initialised by user 'root' with EUID 0
Sep 16 07:09:30 edison libmraa[337]: gpio214: init: Failed to write to 'export': No such device
Sep 16 07:09:30 edison libmraa[337]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Sep 16 07:09:30 edison libmraa[337]: libmraa initialised for platform 'Intel Edison' of type 2
Sep 16 07:09:30 edison libmraa[337]: i2c_init: Selected bus 6
Sep 16 07:09:49 edison libmraa[397]: libmraa version v1.3.0 initialised by user 'root' with EUID 0
Sep 16 07:09:49 edison libmraa[397]: gpio214: init: Failed to write to 'export': No such device
Sep 16 07:09:49 edison libmraa[397]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Sep 16 07:09:49 edison libmraa[397]: libmraa initialised for platform 'Intel Edison' of type 2
Sep 16 07:16:25 edison libmraa[415]: libmraa version v1.3.0 initialised by user 'root' with EUID 0
Sep 16 07:16:25 edison libmraa[415]: gpio214: init: Failed to write to 'export': No such device
Sep 16 07:16:25 edison libmraa[415]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Sep 16 07:16:25 edison libmraa[415]: libmraa initialised for platform 'Intel Edison' of type 2
```

And the mraa version : 
```
root@edison:~# ./blink_onboard 
Welcome to libmraa
 Version: v1.3.0
 Running on Intel Edison
```

mraa info : 
```
root@edison:~# opkg info mraa
Package: mraa
Version: 1.3.0
Provides: mraa-dev, mraa-dbg, mraa-doc
Replaces: mraa-dev, mraa-dbg, mraa-doc, libmraa, libmraa-dev, libmraa-doc
Conflicts: mraa-dev, mraa-dbg, mraa-doc
Status: install user installed
Section: libs
Architecture: i586
Maintainer: Intel IoT-Devkit
MD5Sum: aae99c8fe4e0b3a24229ddf924f68f57
Size: 359230
Filename: mraa_1.3.0_i586.ipk
Source: https://github.com/intel-iot-devkit/mraa
Description: mraa built using CMake
Installed-Time: 1473934155
```

arfoll, 2016-09-16:
This is a hardware issue often caused by using the reset buttons on the arduino breakout. Simplest fix is to completely poweroff for a minute or two and never use the buttons :) If that doesn't work then maybe something else.

anupam19, 2016-09-16:
Yup, that works, weird though. But I guess, we need to accept some things the way they are :)