eclipse-mraa/mraa #545: Firmata AIO always reads 0

petreeftime, 2016-07-19:
I'm trying to read the values of a temperature sensor connected to A1 on a Arduino 101 over Firmata and it always returns 0. I'll print the Python code, but the issue exists for Java as well.

```
import mraa
import time

mraa.addSubplatform(mraa.GENERIC_FIRMATA, "/dev/ttyACM0")
a = mraa.Aio(512 + 1)

while True:
   print(a.read())
   time.sleep(1)
```

As far as I can tell from the code, pin 11 should be A1 for Firmata, but for some reason it doesn't work, it says:

```
>>> a = mraa.Aio(512 + 11)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1470, in __init__
    this = _mraa.new_Aio(pin)
ValueError: Invalid AIO pin specified - do you have an ADC?

>>> mraa.getPinName(512 + 11)
'A1'

>>> mraa.getPinName(512 + 1) 
'IO1'
```

Hbrinj, 2016-07-19:
Have you by chance gone through the IMRAA setup? https://github.com/intel-iot-devkit/mraa/blob/master/docs/imraa.md

petreeftime, 2016-07-19:
Yes, MRAA over Firmata works (I2C and  GPIO at least), AIO does not.

Hbrinj, 2016-07-19:
Hmm, i will test it out at on my board when i get the chance this evening. I will let you know.

Hbrinj, 2016-07-20:
I've replicated the issue, i'll begin diagnosing the issue :)

Hbrinj, 2016-07-27:
I've gotten AIO working again with the commit referenced.

arfoll, 2016-07-27:
merged with https://github.com/intel-iot-devkit/mraa/commit/230b6084ba5b3f8e3e2cc1aa972bdf5c22ca0f73. Thanks!

closed 2016-07-27 by arfoll