eclipse-mraa/mraa #546: make fails with 'linux/ioctl.h file not found' error

bso-intel, 2016-07-20:
Make failed with the following error:
[build] make
Scanning dependencies of target mraa
[  1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
In file included from /Users/bso/mraa/src/mraa.c:52:
In file included from /Users/bso/mraa/include/mraa_internal.h:35:
In file included from /Users/bso/mraa/include/mraa_internal_types.h:31:
In file included from /Users/bso/mraa/api/mraa/iio.h:28:
/Users/bso/mraa/api/mraa/iio_kernel_headers.h:113:10: fatal error: 'linux/ioctl.h' file not found
#include <linux/ioctl.h>
         ^
1 error generated.
make[2]: *** [src/CMakeFiles/mraa.dir/mraa.c.o] Error 1
make[1]: *** [src/CMakeFiles/mraa.dir/all] Error 2
make: *** [all] Error 2

Hbrinj, 2016-07-20:
Yes mraa currently doesn't build on Mac OS. It would be great if you could fix the issue and do a pull request :)

bhorn, 2016-07-20:
If you need the C++ API only:
In "ccmake .."

- Disable BUILDSWIGJAVA
- Disable BUILDSWIGNODE
- Disable BUILDSWIGPYTHON
- Disable ENABLEEXAMPLES

And try this patch hacked by myself: http://pastebin.com/t88bEmPi
(WARNING: This is very dirty and not recomended to use in production. It compiles but I haven't testet it further...)

alext-mkrs, 2016-07-21:
@bso-intel, what's your goal here, why do you want to compile it under MacOS?

bhorn, 2016-07-21:
Update: http://pastebin.com/HiQDLfLt
Now you can actually use it and compile the examples.

@alext-mkrs, maybe the same motivation like me: I want to develop and test on my Mac (with the MOCK-Board) while I don't want to crosscompile/deploy and need the Target for testing all the time.

alext-mkrs, 2016-07-24:
Well, I guess so as well, but it's always better to know for sure :smiley:

As far as MacOS is concerned, we indeed don't support it as of today and I haven't ever tried to run/compile mock for it as I don't have access to any Mac device. In general, as you can see from the code and your own hack, it is doable, but one would need to provide Linux some definition alternatives for MacOS.

Striving for as few changes to the library as possible when creating mock, I've intentionally left most of the includes in place (as on Windows MSYS2 provides them) and only cut out or provided local copies (for Windows compilation) of those that are not available. All in all, one would just need to try to compile mraa on MacOS with mock enabled and carefully go error by error providing alternatives or fixes. So @bhorn if you're up to making your initial hack into something production-ready, that's more than welcome. Alternatively we'd need to wait until someone from project contributors get access to a Mac :smiley:

bhorn, 2016-07-24:
@alext-mkrs well I do my best to support the Project. I could even do the ARTIK-Bindings, too. ARTIK would need some time because I have to do this in my spare time.
Anyway, I'll fork the git and hope to find some time later this day (GMT+2) to make something usefull out of my dirty hacks that will go with your code-design.

bhorn, 2016-07-24:
So I forked, got a decent codebase running without SWIG. For SWIG I need to compile V8 with clang. Done that, but how do I link MRAA against my own V8-libraries? I cant find anything where you try to link against it..?

(tbh.: I don't really know cmake. I can use it but don't write/modify it)

arfoll, 2016-07-24:
Check `cmake -LA ..` output from your build dir. There are node include
dirs you can override if pkgconfig didn't find things (which I guess on a
Mac there isn't). Note alot of prepared node pkgs don't include the actual
headers even if they say they're dev packages. So you may have to compile
it yourself.

I don't know os x enough to give you much morr advice off the top of my
head sorry.

bhorn, 2016-07-24:
Thing is it's dying at
>[ 27%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
>[ 28%] Linking CXX shared module mraa.node
>Undefined symbols for architecture x86_64:
>  "v8::HandleScope::CreateHandle(v8::internal::Isolate*, v8::internal::Object*)", referenced from:
>      SWIG_V8_NewPointerObj(void*, swig_type_info*, int) in mraajsJAVASCRIPT_wrap.cxx.o
>      mraa::Gpio::v8isr(uv_work_s*, int) in mraajsJAVASCRIPT_wrap.cxx.o
> [...]

So I think it finds the needed headers but can't link the libraries. 

Core problem:
V8/Node (ie installed by brew) for OSX got compiled with c++/g++. But the common build system for OSX is clang (OSX symlinks g++/c++ to clang). So in order to link MRAA with V8/Node one have to compile V8 and Node on his own. I've done this but can't find a single -lv8_{} or other lib-includes in any of the cmake and makefiles to tweak them to include my versions.

BinduRao2018, 2018-10-15:
Hi @bhorn,
I came across this thread, while looking for ARTIK adaptation on mraa. I see that you had mentioned about this a while ago? Do you have any leads on this please? I would like to use/build on this as I am using mrra on Raspberry PI and UP2 board already and would be starting on ARTIK board soon.

Thanks,
Bindu